openbmc/openbmc #2968: bitbake mboxd failed with compile errors

yongli3, 2018-03-03:
The error message as below:
| ../git/mboxd_lpc.c: In function '__init_lpc_dev':
| ../git/mboxd_lpc.c:53:9: error: variable 'map' has initializer but incomplete type
|   struct aspeed_lpc_ctrl_mapping map = {
|          ^~~~~~~~~~~~~~~~~~~~~~~
| ../git/mboxd_lpc.c:54:3: error: unknown field 'window_type' specified in initializer
|    .window_type = ASPEED_LPC_CTRL_WINDOW_MEMORY,

amboar, 2018-03-05:
@yongli3 usually this is due to your network setup. The header is downloaded from [gregkh's char-misc tree](https://git.kernel.org/pub/scm/linux/kernel/git/gregkh/char-misc.git/plain/include/uapi/linux/aspeed-lpc-ctrl.h?h=char-misc-next) if it's not already present in your build tree. Here's the `configure.ac` line: https://github.com/openbmc/phosphor-mboxd/blob/master/configure.ac#L20

If you have a proxy make sure you set all the appropriate environment variables to make sure wget can grab the file.

yongli3, 2018-03-07:
@amboar  thanks for your comment. I can find this aspeed-lpc-ctrl.h  in the git folder in yocto. It seems to me that the .bb file has problems, it creates a empty in the build folder.

amboar, 2018-03-14:
@yongli3 did you sort this out?

closed 2019-04-02 by amboar