openbmc/openbmc #2969: OP920:GUI: User password failed to change and no confirmation text / rejection text

cdeadmin, 2018-03-05:


gtmills, 2018-03-05:
Looks like there is no code underneath to actually do the password change. "Save change" on the Manage user account doesn't call anything. https://github.com/openbmc/phosphor-webui/blob/master/app/users/controllers/user-accounts-controller.html#L38

gtmills, 2018-03-05:
users/manage-accounts is the page

gtmills, 2018-03-05:
Command to change the password is 
```curl -c cjar -b cjar -k -H "Content-Type: application/json" -d "{\"data\": [\"abc123\"] }" -X POST  https://${bmc}/xyz/openbmc_project/user/root/action/SetPassword```

iftekharuli, 2018-03-14:
This is a duplicate of the know issue https://github.com/openbmc/openbmc/issues/2741 
We did not code this part due to the lack of the completion rest command when we handed off the GUI.

rfrandse, 2018-03-20:
<https://gerrit.openbmc-project.xyz/9610> Display error and success text
Resolves  #2969 GUI: User password failed to change and no confirmation text / rejection text

gtmills, 2018-03-20:
[image]
Successful password change

[image]

error during password change

closed 2018-03-23 by geissonator