openbmc/openbmc #2971: Startup Script

ThaiThu, 2018-03-06:
Hi all,

We are using evb platform, and we want to run some script at startup, the script is as simple as below:

#!/bin/sh

echo "Hello World !!!"

And the recipe is as follow:

DESCRIPTION = "sysinfo"
LICENSE = "GPLv2"
LIC_FILES_CHKSUM = "file://COPYING;md5=d32239bcb673463ab874e80d47fae504"

PR = "r0"
PV = "0010"

SRC_URI = "file://COPYING \
                file://hello.sh \
                "
DEPENDS = "${@bb.utils.contains('DISTRO_FEATURES', 'systemd', 'systemd', '', d)}"

inherit autotools
inherit systemd update-rc.d

INITSCRIPT_NAME = "hello"
INITSCRIPT_PARAMS = "start 91 5"

S = "${WORKDIR}"

do_compile () {
#        # You will almost certainly need to add additional arguments here
#        cd ${S}
#        oe_runmake
        :
}

do_install () {
        # This is a guess; additional arguments may be required
        cd ${S}
        install -d ${D}${sysconfdir}/init.d
        install -d ${D}${sysconfdir}/rcS.d

        install -m 755 hello.sh ${D}${sysconfdir}/init.d/hello

        ##oe_runmake install DESTDIR=${D}
        update-rc.d -r ${D} hello start 91 S .
}

RPROVIDES_${PN} += "${PN}-systemd"
RREPLACES_${PN} += "${PN}-systemd"
RCONFLICTS_${PN} += "${PN}-systemd"

After booting to kernel, we see that:
- The "hello.sh" is already installed to "/etc/init.d/hello"
- In "/etc/rcS.d/" , we also find out "S91hello" is symlink to  "/etc/init.d/hello"

But at the "hello.sh" is not run.
Could someone give me some suggests or experience the same problem ?

Thank you very much !!

amboar, 2018-03-14:
Hi @ThaiThu - OpenBMC uses [systemd](https://www.freedesktop.org/wiki/Software/systemd/) for its init system, and so you'll need to write a [service file](https://www.freedesktop.org/software/systemd/man/systemd.service.html) to run the applications you need to execute on boot.

ThaiThu, 2018-03-16:
Hi @amboar , Thanks for your support, I've already successfully running script at startup using systemd.
As you said, we need to write a service file and install it into "${D}${systemd_unitdir}/system".

amboar, 2018-03-20:
@ThaiThu great! In that case I'll close the issue.

closed 2018-03-20 by amboar