openbmc/openbmc #2972: p9Coral - LLNL users need to look at firmware levels without having BMC root capability

cdeadmin, 2018-03-06:


rfrandse, 2018-03-06:
@geissonator 
We had a meeting yesterday with Jayesh and the PE team and agreed to enhance the inband IPMI command to retrieve BMC firmware version.

geissonator, 2018-03-07:
Currently if you had a driver of v2.1-223-g2c2a1c5 on your system, you would only get "2 1" in the IPMI data.  This will be enhanced to return "2 1 223" using the auxiliary data in the IPMI command to get version.  Systems can further customize the auxiliary value using the new host ipmid json feature.

rfrandse, 2018-03-13:
<https://gerrit.openbmc-project.xyz/9481> Generate IPMI Aux firmware revision from VERSION_ID field
Resolves  #2972 p9Coral - LLNL users need to look at firmware levels without having BMC root capability

closed 2018-03-27 by geissonator