openbmc/openbmc #2973: p9Field Witherspoon no Processor VPD called out in error logs

cdeadmin, 2018-03-06:


rfrandse, 2018-03-06:
When we have a processor failure the error logs indicated cpu0 or cpu1. They do not give an serial number for the processor in the error logs. This can be figured out via the opbenbmc inventory however as part of RAS I would expect this data to be present in the error log for the mandatory replacement.
how does this help with service?
1) Validating the correct one is removed at time of repair
2) Error logs when they swap cpu's during PDPSI
3) When part comes back for FA

geissonator, 2018-03-07:
Per a discussion today, the thought was to use this issue to start a design discussion on how individual machines or company's can customize the event logs.  We don't want to start a slippery slope of putting whatever a specific company wants in the common event log but we do want a way for them to customize the logs in some way.

One thought was a plugin design, that could monitor for new events getting created and then add to that log within their own specific namespace.

rfrandse, 2018-05-24:
<https://gerrit.openbmc-project.xyz/10795> Restore Callout objects
Resolves  #2973 p9Field Witherspoon no Processor VPD called out in error logs

spinler, 2018-05-24:
The ibm-log-manager is going to create persistent D-Bus objects for callouts like:
```
    "/xyz/openbmc_project/logging/entry/9/callouts/0": {
      "BuildDate": "", 
      "Manufacturer": "", 
      "Model": "2B1D", 
      "PartNumber": "01KL054", 
      "Path": "/xyz/openbmc_project/inventory/system/chassis/motherboard/powersupply0", 
      "SerialNumber": "6AJ108"
    }, 
```

closed 2018-06-04 by bradbishop