openbmc/openbmc #2974: Prevent users from logging in while BMC is not yet ready

AlexanderAmelkin, 2018-03-06:
We would like phosphor-webui to reject user logins while BMC_READY state has not been reached yet.
The questions are:

1. Is this already implemented (can be achieved by means of some configuration) ?
1. Are there any plans to implement such a feature?
1. If not, then what approach would you propose for implementing such a feature?

geissonator, 2018-03-06:
@gtmills - anything like this in the plans?  I could see this making things easier then trying to handle services like BMC/Chassis/Host state stuff being unavailable.

We'd probably have to create a new service, that waits for BMC ready and just polls the dbus interface periodically.  Other services or targets could then just specify to run after this new service.

AlexanderAmelkin, 2018-03-07:
Well, if nothing is implemented in this regard so far, then YADRO team will take this.
Please assign this to me unless there are any other plans on this matter.

Is my understanding correct that BMC can not go from ready to not-ready state without rebooting?

AlexanderAmelkin, 2018-03-07:
Also, not starting the services at all (i.e. making the REST/WebUI not available while not BMC_READY) may be frustrating for the users as it doesn't let to tell a completely broken BMC from a booting one.
Hence, depending on the proposed dbus polling service may not be the best solution.

I would propose altering the REST daemon to return a JSON like this:
```
{
  "data": {
    "description": "BMC is not ready (booting)"
  }, 
  "message": "503 Service Unavailable", 
  "status": "error"
}
```
and handling this object properly in phosphor-webui.

gtmills, 2018-03-07:
Nothing is implemented for this or in the plans.

amboar, 2018-03-14:
@gtmills:

> Please assign this to me unless there are any other plans on this matter.

looks like @AlexanderAmelkin plans to take it on himself

dkodihal, 2018-03-20:
I've posted my comments on https://gerrit.openbmc-project.xyz/#/c/9619/1.

rfrandse, 2018-03-20:
<https://gerrit.openbmc-project.xyz/9619> Prevent users from logging in while BMC is not yet ready.
Step for resolves:  #2974 Prevent users from logging in while BMC is not yet ready

gkeishin, 2018-03-20:
Suppose a system in a bad shape and doesn't reach standby. How are we planning to update or collect data for debugging it if we won't be able to perform REST operation? 

Are we not denying the idea of REST if the services are running but not yet fully loaded to READY?

AlexanderAmelkin, 2018-03-21:
Well, although we here at YADRO think that a BMC not reaching a ready state is something very abnormal and most probably requires interaction via root shell, we nonetheless understand that some other users of openbmc may want to interact with a semi-dead BMC via REST if by some chance it has started successfully while other services haven't.

To allow for that we propose to add a `force: true` parameter to the login request. That way, by default all REST-based services will get the 503 error as described above, but if a force parameter is specified, and the provided login and password are correct, then a normal code 200 will be returned along with a session cookie.

The proposed login request would look like the following:
```
{
    "data": [ "root", "0penBmc" ],
    "force": true
}
```

We prefer this method over what @dkodihal has proposed in Gerrit because this method allows for clear diagnostic of 'why the REST request failed' for all services, even those unaware of Status/BMC or of this 503 code. You will anyway see the failure in the logs. At the same time if you indeed want anything from REST in this state, then you can bypass the check, but do that deliberately and knowingly as opposed to getting a session cookie along with 503, in which case the error code could be just ignored and then some subsequent requests would unpredictably fail without any diagnostics, defeating the very purpose of this change. That is, to minimize the number of 'your bmc is broken' reports for not-yet-booted systems.

What do you think?

bradbishop, 2018-03-26:
Another idea - you could introduce a variable via setup.py and predicate whatever behavior you want to have based on that.  You could enable it with a bbapped and set DISTUTILS_BUILD_ARGS.

Or another idea - simply add a SYSTEMD_OVERRIDE in your layer that adds an ordering dependency to the rest server socket such that it starts whenever you want.

gtmills, 2018-03-27:
Should have this been closed? Looks like the `Partially resolves openbmc/openbmc#2974` in https://gerrit.openbmc-project.xyz/#/c/9618/ closed this.

gtmills, 2018-03-29:
I have reopened the issue. It closed automatically due to the `Partially resolves openbmc/openbmc#2974` in the commit message. The tooling apparently does not recognize the difference between `Partially resolves` and `resolves`. I share George's concern, suppose a system doesn't reach standby, the REST calls could to used to determine why.

AlexanderAmelkin, 2018-03-30:
@gtmills, as I said above, addition of the `force` parameter will let you log in to a non-ready BMC. We're going to update https://gerrit.openbmc-project.xyz/9619 accordingly

bradbishop, 2018-03-30:
@AlexanderAmelkin ok, that makes sense.

how about a command line argument that enables your desired behavior then, to which you can add, again, via bbappend.  I don't think we can simply change the login API behavior on existing systems.

bradbishop, 2018-03-30:
oops.  didn't mean to close.  reopened.

bradbishop, 2018-03-30:
I'd also be fine with enabling the new behavior by default, and expressing the command line option in our layers.  the point is, we need a way to maintain the current behavior on our systems.

if command line option is not palatable perhaps a setup.py --build option could work too.

nest1ing, 2018-04-02:
As @bradbishop proposed, this behavior by default is enabled.
It may be disabled by appending `--without-bmc-check` to command line arguments.

nest1ing, 2018-04-04:
Some notes about changes in API.

It looks clear for me, but maybe this should be clarified.

In authorization request the `force` field is optional. If `force` is not specified server will be interpreting it as `"force": false`. This field is usable at early stages of BMC booting or when something going wrong and BMC unable to reach `Ready` state.

So existing external software does not required any changes, exclude this two cases.

closed 2018-05-30 by geissonator