openbmc/openbmc #2975: Witherspoon: Requested sensor is not present ReasonCode 0x2507 IPMI::RC_SENSOR_NOT_PRESENT

gkeishin, 2018-03-06:
`Driver Info`
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v2.1-219"
VERSION_ID="v2.1-219-g89405da"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v2.1-219"
BUILD_ID="v2.1"
root@witherspoon:~# cat /var/lib/phosphor-software-manager/pnor/ro/VERSION
IBM-witherspoon-ibm-OP9_v1.19_1.142
        op-build-v1.19-583-ge21c3d3-dirty
        buildroot-2017.11-5-g65679be
        skiboot-v5.10-rc4
        hostboot-1e52dd8
        linux-4.14.20-openpower1-p065b180
        petitboot-v1.6.6-pf730dfa
        machine-xml-0fea226
        occ-628b9f6
        hostboot-binaries-3d1af8f
        capp-ucode-p9-dd2-v3
        sbe-706e220
root@witherspoon:~#

```

`SOL log`
```
181.87255öISTEP 20. 1 - host_load_payload
182.33234öISTEP 20. 2 - host_load_hdat
185.77852öISTEP 21. 1 - host_runtime_setup
187.97365ö================================================
187.97829öError reported by ipmi (0x2500) PLID 0x90000A06
187.97680ö  Requested sensor is not present.
187.97680ö  ModuleId   0x03 IPMI::MOD_IPMISENSOR
187.97681ö  ReasonCode 0x2507 IPMI::RC_SENSOR_NOT_PRESENT
187.97682ö  UserData1  BMC IPMI Completion code. : 0x00000000000000cb
187.97684ö  UserData2  bytes Ä0-1Üsensor name bytes Ä2-3Üsensor number bytes Ä4-7ÜHUID of target. : 0xca22000b00010000
187.97684ö------------------------------------------------
187.97685ö  Callout type             : Procedure Callout
187.97686ö  Procedure                : EPUB_PRC_HB_CODE
187.97686ö  Priority                 : SRCI_PRIORITY_HIGH
187.97687ö------------------------------------------------
187.97688ö  Hostboot Build ID: hostboot-1e52dd8/hbicore.bin
187.97688ö================================================
189.66862ö================================================
189.66863öError reported by ipmi (0x2500) PLID 0x90000A07
189.66863ö  Requested sensor is not present.
189.66864ö  ModuleId   0x03 IPMI::MOD_IPMISENSOR
189.66864ö  ReasonCode 0x2507 IPMI::RC_SENSOR_NOT_PRESENT
189.66865ö  UserData1  BMC IPMI Completion code. : 0x00000000000000cb
189.66867ö  UserData2  bytes Ä0-1Üsensor name bytes Ä2-3Üsensor number bytes Ä4-7ÜHUID of target. : 0xca22000b00010000
189.66867ö------------------------------------------------
189.66868ö  Callout type             : Procedure Callout
189.66869ö  Procedure                : EPUB_PRC_HB_CODE
189.66869ö  Priority                 : SRCI_PRIORITY_HIGH
189.66870ö------------------------------------------------
189.66870ö  Hostboot Build ID: hostboot-1e52dd8/hbicore.bin
189.66871ö================================================
191.62738ö================================================
191.62739öError reported by ipmi (0x2500) PLID 0x90000A08
191.62740ö  Requested sensor is not present.
191.62740ö  ModuleId   0x03 IPMI::MOD_IPMISENSOR
191.62741ö  ReasonCode 0x2507 IPMI::RC_SENSOR_NOT_PRESENT
191.62741ö  UserData1  BMC IPMI Completion code. : 0x00000000000000cb
191.62743ö  UserData2  bytes Ä0-1Üsensor name bytes Ä2-3Üsensor number bytes Ä4-7ÜHUID of target. : 0xca22000b00010000
191.62744ö------------------------------------------------
191.62744ö  Callout type             : Procedure Callout
191.62745ö  Procedure                : EPUB_PRC_HB_CODE
191.62745ö  Priority                 : SRCI_PRIORITY_HIGH
191.62746ö------------------------------------------------
191.62746ö  Hostboot Build ID: hostboot-1e52dd8/hbicore.bin
191.62747ö================================================
195.75693öhtmgtöOCCs are now running in ACTIVE state
200.22647öISTEP 21. 2 - host_verify_hdat
200.26076öISTEP 21. 3 - host_start_payload
```

- [x] Logs Journald 
[journald_sensor_log.txt](https://github.com/openbmc/openbmc/files/1786042/journald_sensor_log.txt)

- [x] Logs Errorlog  
[sensor_error_log.txt](https://github.com/openbmc/openbmc/files/1786053/sensor_error_log.txt)

gkeishin, 2018-03-24:
On the latest build more sensors

````
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v2.1-306"
VERSION_ID="v2.1-306-gacd25c4"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v2.1-306"
BUILD_ID="v2.1"
root@witherspoon:~#
```
Journald snippet:
```
Mar 24 17:50:35 witherspoon systemd[1]: Reached target Fan Control Ready.
Mar 24 17:51:18 witherspoon ipmid[1123]: Failed to find Sensor 0x0b
Mar 24 17:51:19 witherspoon ipmid[1123]: Failed to find Sensor 0x0b
Mar 24 17:51:20 witherspoon ipmid[1123]: A host system event with a procedure callout
Mar 24 17:51:21 witherspoon phosphor-fru-fault-monitor[1113]: /xyz/openbmc_project/logging/entry/1
Mar 24 17:51:21 witherspoon ipmid[1123]: Failed to find Sensor 0x0b
Mar 24 17:51:24 witherspoon ipmid[1123]: Failed to find Sensor 0xd8
Mar 24 17:51:24 witherspoon ipmid[1123]: A host system event with a procedure callout
Mar 24 17:51:24 witherspoon phosphor-fru-fault-monitor[1113]: /xyz/openbmc_project/logging/entry/2
```

gkeishin, 2018-03-24:
@dkodihal @tomjoseph83 @geissonator any taker for this one master

geissonator, 2018-03-24:
This should have everything from @devenrao for the turbo sensor now.  Not sure what those one's are.  Same errors in the host traces?

gkeishin, 2018-03-26:
@devenrao ^^^ can u check on this one .

gkeishin, 2018-03-26:
Routing to @tomjoseph83 to have further analysis of the recent changes. Thaks

rfrandse, 2018-03-27:
<https://gerrit.openbmc-project.xyz/9737> Add getObjectFrom* interfaces back in
Partially Resolves  #2975 Witherspoon: Requested sensor is not present ReasonCode 0x2507 IPMI::RC_SENSOR_NOT_PRESENT

closed 2018-03-30 by geissonator