openbmc/openbmc #2977: [kernel-4.13] BMC shutdown or reboot takes longer time than usual

mine260309, 2018-03-07:
With dev-4.13 kernel, it takes longer time in BMC reboot during shutdown.

In 4.10 kernel, it takes about 16 seconds from `reboot` to uboot;
In 4.13 kernel, it takes about 37 seconds.

From the serial console, we can see that systemd takes longer time to stop the services, e.g.
```
...
[  OK  ] Stopped Avahi mDNS/DNS-SD Stack.
[  OK  ] Stopped Phosphor Log Manager.
[  OK  ] Stopped Phosphor Dump core monitor..
[  OK  ] Stopped System Logging Service.
[  OK  ] Stopped Phosphor Settings Daemon.
[  OK  ] Stopped Phosphor User Manager.
[  OK  ] Stopped Phosphor BMC0 Control.
[  OK  ] Stopped Phosphor Inventory Manager.
[  OK  ] Stopped Phosphor IPMI BT DBus Bridge.
[  OK  ] Stopped Kernel Logging Service.
[  OK  ] Stopped Phosphor Download Manager.
[  OK  ] Stopped Phosphor Chassis0 Control.
[  OK  ] Stopped Phosphor Fan Control.
[  OK  ] Stopped Phosphor Download Manager.
[  OK  ] Stopped Phosphor Dump Manager.
[  OK  ] Stopped Phosphor MBOX Daemon.
...
```

mine260309, 2018-03-07:
From testing, the clocksource caused the issue.

* dev-4.10
```
Mar 05 05:57:21 romulus kernel: clocksource: moxart_timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 77222644334 ns
Mar 05 05:57:21 romulus kernel: sched_clock: 32 bits at 100 Hz, resolution 10000000ns, wraps every 21474836475000000ns
Mar 05 05:57:21 romulus kernel: Calibrating delay loop... 789.70 BogoMIPS (lpj=3948544)
```
* dev-4.13
```
Mar 05 05:51:13 romulus kernel: clocksource: FTTMR010-TIMER2: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 77222644334 ns
Mar 05 05:51:13 romulus kernel: sched_clock: 32 bits at 24MHz, resolution 40ns, wraps every 86767015915ns
Mar 05 05:51:13 romulus kernel: Switching to timer-based delay loop, resolution 40ns
Mar 05 05:51:13 romulus kernel: Calibrating delay loop (skipped), value calculated using timer frequency.. 49.50 BogoMIPS (lpj=247500)
```

If we comment below code in `timer-fttmr010.c`
```
diff --git a/drivers/clocksource/timer-fttmr010.c b/drivers/clocksource/timer-fttmr010.c
index 66dd909960c6..fa9e7dbb6456 100644
--- a/drivers/clocksource/timer-fttmr010.c
+++ b/drivers/clocksource/timer-fttmr010.c
@@ -322,8 +322,8 @@ static int __init fttmr010_common_init(struct device_node *np, bool is_aspeed)
                                      "FTTMR010-TIMER2",
                                      fttmr010->tick_rate,
                                      300, 32, clocksource_mmio_readl_down);
-               sched_clock_register(fttmr010_read_sched_clock_down, 32,
-                                    fttmr010->tick_rate);
+//             sched_clock_register(fttmr010_read_sched_clock_down, 32,
+//                                  fttmr010->tick_rate);
        } else {
                writel(0, fttmr010->base + TIMER2_LOAD);
                clocksource_mmio_init(fttmr010->base + TIMER2_COUNT,
```

The issue is gone, and the reboot time becomes back to the same as dev-4.10.

gtmills, 2018-08-20:
@mine260309 We are now on v4.17.14.
 https://gerrit.openbmc-project.xyz/#/c/openbmc/openbmc/+/11953/ 
Still a problem or can this be closed?

mine260309, 2018-08-21:
Just tested dev-4.17 kernel and it takes about 13 seconds on shutdown, which is OK.
Close it.

closed 2018-08-21 by mine260309