openbmc/openbmc #2978: [kernel-4.13] PCIE clock is not enabled

mine260309, 2018-03-07:
In dev-4.13, the PCIE clock is not enabled, which causes PCIE link training errors in skiboot and the VGA is missing in petiboot/OS.

We need to find a way to enable PCIE clock (BCLK, BIT(4) in SCU0c), either in kernel or in user space.

amboar, 2018-03-07:
We probably need to describe it in the device tree and assign the clock there

shenki, 2018-03-08:
Under the current scheme we have two options:

 - Tell the clock driver that this clock is essential, and therefore it should not be disabled
 - Add a driver that enables the clock

mine260309, 2018-03-27:
With the patch http://patchwork.ozlabs.org/patch/890765/, the clock is enabled (by default).
But the pcie/vga device still does not work in Petiboot.
Log is attached: [log_romulus_host_with_bmc_4.13.txt](https://github.com/openbmc/openbmc/files/1850909/log_romulus_host_with_bmc_4.13.txt)

mine260309, 2018-03-28:
Find the root cause, both PCI (bclk) and VGA(dclk) clock are needed.
So enabling bclk/dclk, the issue is gone.

shenki, 2018-03-28:
https://gerrit.openbmc-project.xyz/#/c/9779/

rfrandse, 2018-03-28:
<https://gerrit.openbmc-project.xyz/9779> linux: Host PCIe VGA and random number fixes
Fixes:  #2978 [kernel-4.13] PCIE clock is not enabled

closed 2018-03-28 by geissonator