openbmc/openbmc #2979: IPMI: Implement power-on hours command

dkodihal, 2018-03-07:
IPMI provides an optional POH (Power-on Hours) counter. The function is supposed to increment a counter at a specified rate, as long as the host is powered on. The counter should be in a non-volatile location.

AlexanderAmelkin, 2018-03-26:
I suggest keeping this counter inside BMC VPD EEPROM. Storing it on any flash-type storage will imply erasing/re-writing a whole erase block every hour. Storing in EEPROM will only affect one or several bytes, and each OpenPOWER system, AFAIR, must have a BMC VPD.

I'd suggest introducing a 'UT' (uptime) keyword in VMSC record. The value must be a 32-bit unsigned integer for ease of processing. A 16-bit value would allow for as little as roughly 7.5 years, which is quite restricting. A 24-bit value gives 1915 years but is harder to process. 32-bit value is a total overkill (490 thousand years), but is easy to process.

rfrandse, 2018-04-04:
<https://gerrit.openbmc-project.xyz/9873> Add PowerOnHours interface
Partially Resolves  #2979 IPMI: Implement power-on hours command

rfrandse, 2018-04-09:
<https://gerrit.openbmc-project.xyz/9974> Add Chassis POH Counter
Resolves  #2979 IPMI: Implement power-on hours command

rfrandse, 2018-04-09:
<https://gerrit.openbmc-project.xyz/9960> Implement Get POH Counter command
Partially Resolves  #2979 IPMI: Implement power-on hours command

dkodihal, 2018-04-19:
@AlexanderAmelkin Missed responding to your comment. Agree with what you've stated, although I don't think the BMC has write access to the VPD EEPROM on all OpenPOWER systems. At the same time, I believe that restriction is more of a firmware design rather than a hardware restriction. So that's definitely something that has to be looked at. Writing to the flash can be an interim option or an option where the VPD writes are not possible. @ngorugan we can keep this issue open or write a new one to figure out the VPD writes.

dkodihal, 2018-04-19:
Also does the granularity of the counter update really matter if this is just a long-term host mileage? What if the counter is updated once a day for eg?

AlexanderAmelkin, 2018-04-19:
@dkodihal, it's quite obvious that if you reduce update granularity below once per hour, you can't reliably count hours anymore. Imagine what happens if you set granularity to, say, once per 24 hours and then your chassis gets cut off power after 23 hours since the last update. The IPMI specification allows for starting an hour over in the event of power outage.

If we could make the storage destination configurable, that would be great.

As to writing to an EEPROM... well, I can't say for VPD, as I'm not that much familiar with the spec, but FRUs as per IPMI specification are writable (e.g. for "Asset tag").

AlexanderAmelkin, 2018-04-20:
I don't see how garbage collection in `ubifs` or `jjfs2` is related to device wear. It is a fact though that writing to a flash device (whatever is the upper level filesystem) requires erasing a whole erase block (min 512 bytes) for each write. It is also a fact that EEPROMs usually have higher erase/write cycle count than flash.

Once again, if we had the storage location for POH counter configurable, it would be great. Then each manufacturer could define the location for it as they please.

mdmillerii, 2018-04-20:
My point was the file systems do incremental writes and not all updates require a full block of space so several small writes may be merged into one erase block.  Also the wear will be spread over the size of the file system.

I agree EEPROMs have a higher erase/write count, but was pointing out in a VPD application both the data and the check sum will be repeatedly written to the same locations.   Also what happens if the BMC is reset between the counter and the check sum?

I don't object to having the location configurable, but someone who desires the feature should propose a patch.  Last I remembered we had tooling to parse but not generate VPD (with check sums) in the image.

closed 2018-05-12 by geissonator