openbmc/openbmc #2982: OP920: FVT: IPMID Wrong Turbo Sensor MRW mapping

cdeadmin, 2018-03-07:


rfrandse, 2018-03-07:
```Bus: xyz.openbmc_project.Hwmon-3085209850.Hwmon1, Path: /xyz/openbmc_project/sensors/fan_tach/fan0_0, Interface: xyz.openbmc_project.Control.Host.TurboAllowed```

This is very odd
Why is the TurboAllowed associated with a fan?
Smells like a wrong sensor number or something. Pls open github and/or CQ @gkeishin. This looks concerning.

George Keishing [22:39]
Yeah.. will do ..

Deepak Kodihalli [22:41]
@ojayanth did the turbo allowed sensor get in the MRW?

Jayanth Othayoth [22:42]
yes
Deepak Kodihalli [22:42]
Do you have the commit @ojayanth? (Also pls look at the traces @gkeishin pointed above).
@gkeishin I found the root cause. Pls assign the bug to @devenrao.

rfrandse, 2018-03-13:
<https://gerrit.openbmc-project.xyz/9514> Modify ipmi sensor config yaml to include analog sensors
Resolves  #2982 OP920: FVT: IPMID Wrong Turbo Sensor MRW mapping

closed 2018-03-23 by geissonator