openbmc/openbmc #2985: IPMI: performance improvements

dkodihal, 2018-03-07:
This issue is to track IPMI performance improvements. We are targeting a set of specific ipmi/dcmi commands here, so to start with I'll post the current time measurements on this soon.

dkodihal, 2018-03-14:
Previously, phosphor-hwmon had a polling loop to read the sysfs - it would wakeup every second or when a D-Bus event occurs (in which case it would service D-Bus and then read sysfs). This causes successive D-Bus calls to slow down, because hwmon would be busy in the loop, reading sysfs, after servicing D-Bus.

To solve this, schedule the sysfs read independently of the D-Bus event loop : add a timer handler to an sd event loop, and attach the D-Bus event loop to the sd event loop.

An example of how this helped : the 'dcmi sensors' command (to read 50 temperature sensors) would take more than 60 seconds on Witherspoon (based off of master 00f02f3). With this patch, the time taken is about 15 seconds.

rfrandse, 2018-03-14:
<https://gerrit.openbmc-project.xyz/9549> performance: setup sysfs read independent of D-Bus
Resolves  #2985 IPMI: performance improvements

closed 2018-04-10 by geissonator