openbmc/openbmc #2989: GUI:GA2:OP920: wsbmc014: DD2.2: Warning signs displayed in cores of CPU. No explaination / events for that

cdeadmin, 2018-03-08:


gtmills, 2018-03-08:
The warning icon is displayed when a core is present and not when it is not present. This seems backwards. 

https://github.com/openbmc/phosphor-webui/blob/master/app/server-health/controllers/inventory-overview-controller.html#L50

[image]

bash-4.1$ curl -b cjar -k https://$BMC_IP/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu0/enumerate
{
  "data": {
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu0/core0": {
      "Functional": 1, 
      "Present": 1, 
      "PrettyName": ""
    }, 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu0/core1": {
      "Functional": 1, 
      "Present": 1, 
      "PrettyName": ""
    }, 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu0/core10": {
      "Functional": 1, 
      "Present": 1, 
      "PrettyName": ""
    }, 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu0/core11": {
      "Functional": 1, 
      "Present": 1, 
      "PrettyName": ""
    }, 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu0/core12": {
      "Functional": 0, 
      "Present": 0, 
      "PrettyName": ""
    }, 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu0/core13": {
      "Functional": 0, 
      "Present": 0, 
      "PrettyName": ""
    },

rfrandse, 2018-03-08:
<https://gerrit.openbmc-project.xyz/9473> Display warning icons on non-functional subcomponents
Resolves  #2989 GUI: Warning icons displayed in cores of CPU that are present and functional

closed 2018-03-13 by geissonator