openbmc/openbmc #2992: Witherspoon IPMI enable and disable Power supply redundancy

gkeishin, 2018-03-09:
```
/xyz/openbmc_project/sensors/chassis/enumerate
{
    "/xyz/openbmc_project/sensors/chassis/PowerSupplyRedundancy": {
        "error": 0,
        "units": "",
        "value": "Enabled"
    }
}
```

`To disable`
```
-bash-4.1$ ipmitool -I lanplus -C 3 -P 0penBmc -H xx.xx.xx.xx raw 0x04 0x2d 0x0b
Unable to send RAW command (channel=0x0 netfn=0x4 lun=0x0 cmd=0x2d rsp=0xcb): Requested sensor, data, or record not found
-bash-4.1$
```
and same with enable too.

```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v2.1-240"
VERSION_ID="v2.1-240-gc244f65"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v2.1-240"
BUILD_ID="v2.1"
root@witherspoon:~#
```

`Journald Snippet`
```
Mar 09 13:07:24 witherspoon systemd[1]: Started Network IPMI daemon.
Mar 09 13:07:24 witherspoon kernel: nf_conntrack: default automatic helper assignment has been turned off for security reasons and CT-based  firewall rule not found. Use the iptables CT target to attach helpers instead.
Mar 09 13:07:24 witherspoon kernel[941]: [ 3716.459087] nf_conntrack: default automatic helper assignment has been turned off for security reasons and CT-based  firewall rule not found. Use the iptables CT target to attach helpers instead.
Mar 09 13:07:25 witherspoon netipmid[1372]: Registering handler: /usr/lib/net-ipmid/libapphandler.so.0
Mar 09 13:07:25 witherspoon netipmid[1372]: Registering NetFn:[0x6], Cmd:[0x36]
Mar 09 13:07:25 witherspoon netipmid[1372]: I> Registering Command1836
Mar 09 13:07:25 witherspoon netipmid[1372]: Registering NetFn:[0x6], Cmd:[0xFF]
Mar 09 13:07:25 witherspoon netipmid[1372]: I> Registering Command18ff
Mar 09 13:07:25 witherspoon netipmid[1372]: Registering NetFn:[0x6], Cmd:[0x22]
Mar 09 13:07:25 witherspoon netipmid[1372]: I> Registering Command1822
Mar 09 13:07:25 witherspoon netipmid[1372]: Registering NetFn:[0x6], Cmd:[0x24]
Mar 09 13:07:25 witherspoon netipmid[1372]: I> Registering Command1824
Mar 09 13:07:25 witherspoon netipmid[1372]: I> Registering Command1825
Mar 09 13:07:25 witherspoon netipmid[1372]: Registering NetFn:[0x6], Cmd:[0x1]
Mar 09 13:07:25 witherspoon netipmid[1372]: I> Registering Command1801
Mar 09 13:07:25 witherspoon netipmid[1372]: Registering NetFn:[0x6], Cmd:[0x4]
Mar 09 13:07:25 witherspoon netipmid[1372]: I> Registering Command1804
Mar 09 13:07:25 witherspoon netipmid[1372]: Registering NetFn:[0x6], Cmd:[0x8]
Mar 09 13:07:25 witherspoon netipmid[1372]: I> Registering Command1808
Mar 09 13:07:25 witherspoon netipmid[1372]: Registering NetFn:[0x6], Cmd:[0x6]
Mar 09 13:07:25 witherspoon netipmid[1372]: I> Registering Command1806
Mar 09 13:07:25 witherspoon netipmid[1372]: Registering NetFn:[0x6], Cmd:[0x40]
Mar 09 13:07:25 witherspoon netipmid[1372]: I> Registering Command1840
Mar 09 13:07:25 witherspoon netipmid[1372]: Registering NetFn:[0x6], Cmd:[0x41]
Mar 09 13:07:25 witherspoon netipmid[1372]: I> Registering Command1841
Mar 09 13:07:25 witherspoon netipmid[1372]: Registering NetFn:[0x6], Cmd:[0x42]
Mar 09 13:07:25 witherspoon netipmid[1372]: I> Registering Command1842
Mar 09 13:07:25 witherspoon netipmid[1372]: Registering NetFn:[0x6], Cmd:[0x37]
Mar 09 13:07:25 witherspoon netipmid[1372]: I> Registering Command1837
Mar 09 13:07:25 witherspoon netipmid[1372]: Registering NetFn:[0x4], Cmd:[0xFF]
Mar 09 13:07:25 witherspoon netipmid[1372]: I> Registering Command10ff
Mar 09 13:07:25 witherspoon netipmid[1372]: Registering NetFn:[0x4], Cmd:[0x2F]
Mar 09 13:07:25 witherspoon netipmid[1372]: I> Registering Command102f
Mar 09 13:07:25 witherspoon netipmid[1372]: Registering NetFn:[0x4], Cmd:[0x30]
Mar 09 13:07:25 witherspoon netipmid[1372]: I> Registering Command1030
Mar 09 13:07:25 witherspoon netipmid[1372]: Registering NetFn:[0x4], Cmd:[0x2D]
Mar 09 13:07:25 witherspoon netipmid[1372]: I> Registering Command102d
Mar 09 13:07:25 witherspoon netipmid[1372]: I> Registering Command1022
[... 74 lines cut ...]
Mar 09 13:07:26 witherspoon netipmid[1372]: I> Registering Command140000
Mar 09 13:07:26 witherspoon netipmid[1372]: I> Registering Command1838
Mar 09 13:07:26 witherspoon netipmid[1372]: I> Registering Command183b
Mar 09 13:07:26 witherspoon netipmid[1372]: I> Registering Command183c
Mar 09 13:07:26 witherspoon netipmid[1372]: I> Registering Command10000
Mar 09 13:07:26 witherspoon netipmid[1372]: I> Registering Command1848
Mar 09 13:07:26 witherspoon netipmid[1372]: I> Registering Command1849
Mar 09 13:07:26 witherspoon netipmid[1372]: I> Registering Command184a
Mar 09 13:07:26 witherspoon netipmid[1372]: I> Registering Command3021
Mar 09 13:07:26 witherspoon netipmid[1372]: I> Registering Command3022
Mar 09 13:07:26 witherspoon netipmid[1372]: >> GetChannelCapabilities
Mar 09 13:07:26 witherspoon netipmid[1372]: << GetChannelCapabilities
Mar 09 13:07:26 witherspoon netipmid[1372]: >> GetChannelCapabilities
Mar 09 13:07:26 witherspoon netipmid[1372]: << GetChannelCapabilities
Mar 09 13:07:26 witherspoon netipmid[1372]: >> openSession
Mar 09 13:07:26 witherspoon netipmid[1372]: << openSession
Mar 09 13:07:26 witherspoon netipmid[1372]: >> RAKP12
Mar 09 13:07:26 witherspoon netipmid[1372]: << RAKP12
Mar 09 13:07:26 witherspoon netipmid[1372]: >> RAKP34
Mar 09 13:07:26 witherspoon netipmid[1372]: << RAKP34
Mar 09 13:07:26 witherspoon netipmid[1372]: >> setSessionPrivilegeLevel
Mar 09 13:07:26 witherspoon netipmid[1372]: << setSessionPrivilegeLevel
Mar 09 13:07:26 witherspoon netipmid[1372]: E> Table:: Command Not found: 0xb03e
Mar 09 13:07:26 witherspoon netipmid[1372]: Failed to create a method call: Invalid request descriptorFailed to find Sensor 0x0b
```

gkeishin, 2018-03-09:
````
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v2.1-239"
VERSION_ID="v2.1-239-g98861c5"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v2.1-239"
BUILD_ID="v2.1"
root@witherspoon:~#

```
Yeah seen same on latest master too. @dkodihal

ngorugan, 2018-03-27:
When this issue occurs, busctl monitor giving below given info

```
Type=error  Endian=l  Flags=1  Version=1  Priority=0 Cookie=68  ReplyCookie=29
  Sender=:1.84  Destination=:1.342
  ErrorName=org.freedesktop.DBus.Error.UnknownMethod  ErrorMessage="Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/dbus/service.py", line 654, in _message_cb
    (candidate_method, parent_method) = _method_lookup(self, method_name, interface_name)
  File "/usr/lib/python2.7/site-packages/dbus/service.py", line 246, in _method_lookup
    raise UnknownMethodException('%s is not a valid method of interface %s' % (method_name, dbus_interface))
UnknownMethodException: org.freedesktop.DBus.Error.UnknownMethod: Unknown method: getObjectFromByteId is not a valid method of interface org.openbmc.managers.System
  };
```

It is not able to find the method "getObjectFromByteId" on the interface  "org.openbmc.managers.System"

```
root@witherspoon:/tmp#  busctl introspect org.openbmc.managers.System /org/openbmc/managers/System  |cat  | grep getObjectFromByteId
root@witherspoon:/tmp#
```

But with master build I am not seeing this issue.

```
root@witherspoon:~# busctl introspect org.openbmc.managers.System /org/openbmc/managers/System  |cat  | grep getObjectFromByteId
.getObjectFromByteId                method    sy         (ss) 

root@witherspoon:~# cat  /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v2.1"
VERSION_ID="ibm-v2.1-105-g4397421-r1-0-gbeb0b18"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v2.1"
BUILD_ID="ibm-v2.1-105-g4397421-r1"
```

Still debugging.

gkeishin, 2018-03-27:
Thanks @ngorugan  keep posted.

ngorugan, 2018-03-28:
@gkeishin  It seems this issue is dup of issues #2975.

gkeishin, 2018-03-28:
Yup.. looks like that only.. @geissonator fix should cover this one. I just ran it on his patch and it worked. So we can close this one.

closed 2018-03-28 by gkeishin