openbmc/openbmc #2996: libgpg-error build error on ppc64le

mine260309, 2018-03-14:
Building latest OpenBMC on ppc64le Ubuntu, it gets below error:
```
| DEBUG: Executing python function externalsrc_compile_prefunc
| NOTE: libgpg-error-native: compiling from external source tree /home/leiyu/work/openbmc/build/workspace/sources/libgpg-error-native
| DEBUG: Python function externalsrc_compile_prefunc finished
| DEBUG: Executing shell function do_compile
| cp: cannot stat '/home/leiyu/work/openbmc/build/workspace/sources/libgpg-error-native/src/syscfg/lock-obj-pub.ppc64le-unknown-linux-gnu.h': No such file or directory
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /home/leiyu/work/openbmc/build/tmp/work/ppc64le-linux/libgpg-error-native/1.26-r1/temp/log.do_compile.51500)
ERROR: Task (virtual:native:/home/leiyu/work/openbmc/meta/recipes-support/libgpg-error/libgpg-error_1.26.bb:do_compile) failed with exit code '1'
```

System is Ubuntu 16.04.4 LTS (Xenial Xerus):
```
Linux ubuntu-firestone 4.4.0-112-generic #135-Ubuntu SMP Fri Jan 19 11:48:46 UTC 2018 ppc64le ppc64le ppc64le GNU/Linux
```

geissonator, 2018-03-14:
Yeah, meant to get a bug for this.  We're looking into it.  A quick hack I did was to link one of the other powerpc64le files to the one it's looking for.  It's related to the yocto2.3 update that went in today.

FYI @bradbishop 

```
ln -sf /tmp/andrewg/2018-03-13_openbmc_daily/openbmc/build/tmp/work/ppc64le-linux/libgpg-error-native/1.26-r1/libgpg-error-1.26/src/syscfg/lock-obj-pub.powerpc64le-unknown-linux-gnu.h /tmp/andrewg/2018-03-13_openbmc_daily/openbmc/build/tmp/work/ppc64le-linux/libgpg-error-native/1.26-r1/libgpg-error-1.26/src/syscfg/lock-obj-pub.ppc64le-unknown-linux-gnu.h
```

mine260309, 2018-03-14:
Or use below patch
```
diff --git a/import-layers/yocto-poky/meta/recipes-support/libgpg-error/libgpg-error_1.26.bb b/import-layers/yocto-poky/meta/recipes-support/libgpg-error/libgpg-error_1.26.bb
index 090db1b..ffaf4e0 100644
--- a/import-layers/yocto-poky/meta/recipes-support/libgpg-error/libgpg-error_1.26.bb
+++ b/import-layers/yocto-poky/meta/recipes-support/libgpg-error/libgpg-error_1.26.bb
@@ -42,6 +42,7 @@ do_compile_prepend() {
          mips*el)    TUPLE=mipsel-unknown-linux-gnu ;;
          mips*)      TUPLE=mips-unknown-linux-gnu ;;
          x86_64)     TUPLE=x86_64-pc-linux-gnu ;;
+         ppc64le)    TUPLE=powerpc64le-unknown-linux-gnu ;;
          *)          TUPLE=${TARGET_ARCH}-unknown-linux-gnu ;;
        esac
```

I guess we may need to get such patch upstream for `ppc64le` and `ppc64`?

mine260309, 2018-07-18:
Not reproduced anymore.

closed 2018-07-18 by mine260309