openbmc/openbmc #2997: OCC failed with auto binding during power cycle

ryandcyu, 2018-03-14:
@anoo1 @mine260309 
Observing occ got fail after DC power on. First power on is workable(AC on)
OCC is workable If binding by manually. Is a guess openpower-occ-control binding timing issue? 
Version:
```
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v2.1-24"
VERSION_ID="v2.1-24-g6fb69af"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v2.1-24"
BUILD_ID="v2.1"
```

journal log:
```
Jan 31 01:57:53 zaius kernel: occ-hwmon occ1-dev0: occ bus op failed rc:-5
Jan 31 01:57:53 zaius kernel: occ-hwmon occ1-dev0: failed to get OCC poll response: -5
Jan 31 01:57:53 zaius kernel: occ-hwmon: probe of occ1-dev0 failed with error -5
Jan 31 01:57:53 zaius kernel[931]: [ 2822.150000] occ-hwmon occ1-dev0: occ bus op failed rc:-5
Jan 31 01:57:53 zaius kernel[931]: [ 2822.150000] occ-hwmon occ1-dev0: failed to get OCC poll response: -5
Jan 31 01:57:53 zaius kernel[931]: [ 2822.150000] occ-hwmon: probe of occ1-dev0 failed with error -5
```

eddiejames, 2018-03-14:
-5 indicates a problem at the FSI level. So yea, possibly a timing thing. But we do get random FSI errors on occasion

anoo1, 2018-03-16:
Pinged @cbostic to help us debug this issue at the FSI level.

cbostic, 2018-03-16:
Probably need to dump the low level FSI protocol data to see what type of bus failure we're getting.  To enable trace:

echo 1 > /sys/kernel/debug/tracing/events/fsi/enable;
echo 1 > /sys/kernel/debug/tracing/events/fsi_master_gpio/enable;

# Then cat the trace file after recreate:
cat  /sys/kernel/debug/tracing/trace

amboar, 2018-03-19:
For the record I have a tracing helper script here:

https://github.com/openbmc/openbmc-tools/tree/master/amboar/obmc-scripts/tracing

Run it from your local machine and point it to the BMC that you want to trace. I usually pipe the output through `tee` to also capture a log whilst its running:

```
$ ./trace root@bmc fsi | tee trace-fsi.log
```

ryandcyu, 2018-03-20:
@cbostic 
I have dumped the fsi log follow amboar's comment, please refer the attachment.
[Uploading trace-fsi.log…]()

cbostic, 2018-03-26:
Looking over the trace data I see indications that BREAK commands to the processor off of hub link 1 are occasionally failing.   No real explanation why that may be at this point -will continue to review.   No signs of CRC failures which is what I was expecting to see.

cbostic, 2018-03-29:
I'd recommend trying the latest linux 4.13 kernel if possible to see if this problem still exists.   If it does still exist then would be good to understand the use cases better.  Andrew G. mentioned he wasn't aware of seeing this issue in house.

jk-ozlabs, 2018-03-29:
This looks like the first error from the trace, from the first time we try to access the fsi2 master:

```
 openpower-proc--1570  [000] .n..  6143.020000: fsi_master_break: fsi2:0
 openpower-proc--1570  [000] .n..  6143.020000: fsi_master_write: fsi1:01:00 00080004[4] <= {c0 de 00 00}
 openpower-proc--1570  [000] .n..  6143.020000: fsi_master_write: fsi0:00:00 00180004[4] <= {c0 de 00 00}
 openpower-proc--1570  [000] dn..  6143.020000: fsi_master_gpio_out: fsi-gpio0 <= 230000bc0de0000e[64]
 openpower-proc--1570  [000] dn..  6143.020000: fsi_master_gpio_in: fsi-gpio0 => 0[1]
 openpower-proc--1570  [000] dn..  6143.020000: fsi_master_gpio_in: fsi-gpio0 => 1[1]
 openpower-proc--1570  [000] dn..  6143.020000: fsi_master_gpio_in: fsi-gpio0 => 2[4]
 openpower-proc--1570  [000] dn..  6143.020000: fsi_master_gpio_in: fsi-gpio0 => c[4]
 openpower-proc--1570  [000] .n..  6143.020000: fsi_master_rw_result: fsi0:00:00 00180004[4] <= {c0 de 00 00} ret -5
```

What's happening here:
 - we're issuing a break to the fsi2 master, as part of its initial scan
 - this requires a `c0de0000` write to fsi1, at 0x80004
 - which requires a `c0de0000` write to fsi0, at 0x1800004
 - that write generates a ERR-A response from fsi0

So, some sanity checks:

 - fsi2 is cascaded behind fsi1, which is itself cascaded behind fsi0. Is that correct?
 - do those address space mappings look correct? (ie, that fsi2 is mapped into fsi1 at 0x1000000).
 - could we have the dmesg log from the initial boot?

mine260309, 2018-03-30:
With latest OpenBMC (kernel on dev-4.13), the issue is gone.
The occ/fsi driver are updated on dev-4.13 and it seems to have fixed this issue.

geissonator, 2018-04-05:
Great, closing this one out.

closed 2018-04-05 by geissonator