openbmc/openbmc #3001: How to change kernel configuration

hsuedw, 2018-03-15:
I am trying to use configuration fragment to build **drivers/hwmon/i5k_amb.c** to linux kernel.

```
$ cat temp_sensor.cfg
CONFIG_SENSORS_I5K_AMB=y
CONFIG_SENSORS_I5500=y

$ cat linux-obmc_%.bbappend
FILESEXTRAPATHS_prepend := "${THISDIR}/${PN}:"
SRC_URI_append = " file://device-tree.patch"
SRC_URI_append = " file://temp_sensor.cfg"

```
However, after the code is built, CONFIG_SENSORS_I5K_AMB and CONFIG_SENSORS_I5500 are not included in .config. 

Does anyone know how to change kernel configuration in openbmc??

BRs,
Edward

LindarLiu0724, 2018-03-15:
Below line works in my side:
SRC_URI += "file://xxx.cfg"

hsuedw, 2018-03-16:
Hi LindarLiu0724:

Thanks for your reply. 
I have tried and didn't work. 
I think I am going to have a clean code and try again.

hsuedw, 2018-03-19:
After having a clean code, I can build the driver successfully.

closed 2018-03-19 by hsuedw