openbmc/openbmc #3002: SNMP : Send traps for errors/events occuring on the BMC

dkodihal, 2018-03-15:
At a high-level, this epic will contain stories to deal with the translation of something occurring in the D-Bus model to an SNMP trap.

dkodihal, 2018-03-15:
@ratagupt to send out proposal to the community. Here's a dump of what we've capture so far in discussions :

```
Requirement:- Enable support for SNMP trap notification on openbmc.
Configurable Trap Parameters
============================
1) Manager IP
2) Interested Trap types
    Notice
        Warning
        Critical
How does the SNMP manager provide config information
- Do we need a REST (and hence D-Bus) API?
- Do we need snmp agent code?
How to implement the snmp trap feature
======================================
We have two choices here
1) Either extend the DbusMonitor
   Pros
   ====
      1) Can use the existing code infra
NOTE:-  We have to implement the snmp specific dbus interface
         which would be used to configure the snmp trap specific
         parameters.
  
2) Create seprate process and seprate repo.
   Pros
   ====
     Snmp specific code would be seprate.
   Cons
   ====
     Code would be repititive among DbusMonitor and the new app.
Recommendation:- we should extend the Dbus Monitor app
How we can use existing infra with the use of Phosphor-Dbus-Monitor
===================================================================
- name: create property event
  class: callback
  callback: snmpEvent
  paths: occpath group
  properties: occproctemp property
  eventType: Notice/Warning/Critical
  eventMessage: "User can give the message as per the eventType"
         PropertyChange:- Normal property Change could be eventType as Notice eg:- StateChange/Network Configuration change
         PropertyChange would become Warning:- if we have met a specific condition on a property
                           eg Proc Temprature > some configured tempratute value.
We need to extend the Phosphor-Dbus_monitor Infra to supports
=============================================================
     1) Interface Added signal
           usecases
           ========
           1) error dbus object gets added under /xyz/openbmc_project/logging
                then need to send the Trap with eventType as Critical 
           2) Telemetry aggreagator objects
What attributes would be there in the snmp Event(TRAP) Structure
========================================================
AgentIP   :   BMC IP Address
TimeStamp :   Time when the trap generated
deviceName:-  DbusObjectPath
alertLevel:-  Would be given by the config yaml
Message:-     Would be given by the config yaml and at run time add the 
              actual property name and the value.
We can have also define the category of the Traps
eg:- Device Configuration :-  Network,logging etc
     Software updates 
     
NOTE:- There are certain alerts which would not dependent on watching properties or interface added on dbus objects
       eg:- consecutive three authentication failures  and we need to genarate the trap for that.
         So the application need to provide a way where any application can call a method with 
         the paraemeters to send the trap.
Opensource:- NetSnmp
             Comes up with full suite of tools like snmp agent,snmptrapd, mib2c, bunch of other tools
             Currently as we are supporting the trap so we need minimal of these.
             It is widely used in the linux community for supporting snmp.
             Already available in openbmc meta-openembeded layer.
```

AlexanderAmelkin, 2018-03-19:
We currently have an implementation called phosphor-snmp that has some of the described features implemented. It is not flexible though and can't be used as is for other vendors (has our OUI hard-coded), but can be used as a starter. Please see the attached [README.txt](https://github.com/openbmc/openbmc/files/1824868/README.txt) from the project for more details.

amboar, 2018-03-20:
@AlexanderAmelkin can you publish the code?

ratagupt, 2018-03-20:
Thanks @amboar I wanted to ask the same.

krtaylor, 2018-03-23:
@AlexanderAmelkin your mib extension and sub-agent could be the basis for a generalized OBMC mib. We have an IANA enterprise number - 49871 OpenBMC Project
Can we continue this discussion in the maillist? @ratagupt has started that thread. How about a workgroup to get a basic OBMC mib extension defined?

AlexanderAmelkin, 2018-03-27:
Please take a look at https://github.com/YADRO-KNS/phosphor-snmp

closed 2018-08-13 by rfrandse