openbmc/openbmc #3005: SNMP: Determine the (net-snmp) API to send traps out

dkodihal, 2018-03-15:
- Figure out what pieces of net-snmp is required to send traps out.
- Get those recipes in the Witherspoon layer (if not there already).
- Demo via small test program the ability to send a trap out from the BMC.

ratagupt, 2018-04-19:
To send the trap out we need the libnetsnmp and we need the default mib files which comes via netsnmp.

ratagupt, 2018-05-02:
We have already done it. we can close this.
Few gerrit link for the reference.
https://gerrit.openbmc-project.xyz/#/c/10148/  :- Example code how to send the trap out.
https://gerrit.openbmc-project.xyz/#/c/10284/ :- Recipe changes to bring the netsnmp.
https://gerrit.openbmc-project.xyz/#/c/10285/

closed 2018-05-18 by dkodihal