openbmc/openbmc #3009: Code a sample page for Camvan Nguyen on data refresh on a page

iftekharuli, 2018-03-15:
In the GUI, there is a refresh button.  The is refresh button seems to only refreshes the server status, network info and server health, hostname and mac address..  the task is to code a sample page to refresh the page data so that @camvanng  can follow

rfrandse, 2018-03-21:
<https://gerrit.openbmc-project.xyz/9645> Fix refresh for event log page
Ffixes  #3009 GUI: Code a sample page for data refresh on a page

iftekharuli, 2018-03-23:
This has been coded for event log page and as per discussion, the dev team will code the rest of the pages. This code for log page exists here- https://gerrit.openbmc-project.xyz/#/c/9645/

One point I want to emphasis on coding this to other pages is to remember destroy you listener call at the end of the call otherwise you will encounter chain call issues. All in example code above

closed 2018-03-29 by geissonator