openbmc/openbmc #3011: [LTCTest][OPAL][OP910.20] ipmid[1252]: Failed to get /system/chassis/motherboard/cpu0/core0 busname: (null)

cdeadmin, 2018-03-16:


amboar, 2018-03-20:
Please provide a description of how to reproduce and what versions are affected.

devenrao, 2018-03-21:
Refer to defect SW419195

mdmillerii, 2018-03-21:
@devenrao This is supposed to be a public issue tracker.  The defect database you refer to is not.  Please summarize here.

devenrao, 2018-04-03:
Install a P9 Open Power Hardware with the latest OP910.20 Firmware images.
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v2.0"
VERSION_ID="ibm-v2.0-0-r35-0-g7c7c1af"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v2.0"
BUILD_ID="ibm-v2.0-0-r35"
```
Seeing the following messages in the Open BMC journalctl logs.
```
root@witherspoon:~# journalctl | tail -100
Feb 08 09:12:05 witherspoon ipmid[1252]: Failed to get /system/chassis/motherboard/cpu1/core12 busname: (null)
Feb 08 09:12:05 witherspoon ipmid[1252]: Failed to find Sensor 0x80
Feb 08 09:12:05 witherspoon ipmid[1252]: Failed to get /system/chassis/motherboard/cpu1/core13 busname: (null)
Feb 08 09:12:05 witherspoon ipmid[1252]: Failed to find Sensor 0x83
```

amboar, 2018-04-09:
@devenrao Genuine question, can we publicly access the latest OP910.20 firmware images? If not, this issue should be written in terms of publicly available source. If not, then this issue shouldn't be in the public issue tracker.

devenrao, 2018-05-02:
Rejected
IPMI is not supported for OP910 releases.

closed 2018-05-14 by rfrandse