openbmc/openbmc #3012: Sensor query for user power limit always returns 0 to HTMGT

cdeadmin, 2018-03-16:


amboar, 2018-03-20:
Can you please provide a description of how to reproduce this issue, and what versions are affected?

ngorugan, 2018-03-22:
@amboar Below given info  is provided by @cjcain in CQ.

All queries of the user power limit from HTMGT always return 0.  The user power limit status
does get returned correctly.

When I set a user power limit (via curl commands), the BMC appears to be processing those
command and does return the correct value when queried via curl. OCC is also informed of the
new limit:
```
$ curl -c cjar -b cjar -k -H "Content-Type: application/json" https://$BMCIP/xyz/openbmc_project/control/host0/power_cap
{
 "data": {
    "PowerCap": 0,
    "PowerCapEnable": 0
 },
 "message": "200 OK",
 "status": "ok"
}

$ curl -c cjar -b cjar -k -H "Content-Type: application/json"  -X PUT -d '{"data":1500}' https://$BMCIP/xyz/openbmc_project/control/host0/power_cap/attr/PowerCap
{
 "data": null,
 "message": "200 OK",
 "status": "ok"
}

$ curl -c cjar -b cjar -k -H "Content-Type: application/json"https://$BMCIP/xyz/openbmc_project/control/host0/power_capp
{
 "data": {
    "PowerCap": 1500,
    "PowerCapEnable": 0
 },
 "message": "200 OK",
 "status": "ok"
}

$ curl -c cjar -b cjar -k -H "Content-Type: application/json"  -X PUT -d '{"data":1}' https://$BMCIP/xyz/openbmc_project/control/host0/power_cap/attr/PowerCapEnable
{
 "data": null,
 "message": "200 OK",
 "status": "ok"
}

$ curl -c cjar -b cjar -k -H "Content-Type: application/json"https://$BMCIP/xyz/openbmc_project/control/host0/power_cape
{
 "data": {
    "PowerCap": 1500,
    "PowerCapEnable": 1
 },
 "message": "200 OK",
 "status": "ok"
}
```
```
$ ./occtoolp9 -p
Sending 0x00 command to OCC0 (via opal-prd)...
 Data Length: 0x034D
       Status: 0xC3 ( Master FIRMaster ObsReady ActReady)
    ExtStatus: 0x10 ( QckPwrDrop )
         OCCs: 0x03
      CfgNeed: 0x00
        State: 0x03 (ACTIVE)
...
       Sensor: CAPS - format:3 / 1 sensors (15 bytes/sensor)
         Current Power Cap:   1350 Watts
             Current Power:    328 Watts (output power)
               N Power Cap:   1700 Watts (cap without redundant power)
             Max Power Cap:   3050 Watts
        Hard Min Power Cap:   1700 Watts
        Soft Min Power Cap:    500 Watts
[... 57 lines cut ...]
```

System I was testing on was wsbmxxx.xxx.xxx
```
# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v2.0"
VERSION_ID="ibm-v2.0-0-r41-0-gd0c319e"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v2.0"
BUILD_ID="ibm-v2.0-0-r41"

# cat /var/lib/phosphor-software-manager/pnor/ro/VERSION
open-power-witherspoon-v1.21-55-g309da0f-dirty
        buildroot-2017.11.2-8-g4b6188e
        skiboot-v5.10-45-g84186ef0944c
        hostboot-f5d2c87
        linux-4.15.6-openpower1-pc9a8340
        petitboot-v1.7.1-p78f3192
        machine-xml-c10638f-pc56f45a
        occ-2fe8f2c
        hostboot-binaries-290a4f0
        capp-ucode-p9-dd2-v3
        sbe-75ddac2
```

amboar, 2018-04-09:
@ngorugan thanks for the detail, that makes it much clearer. However, is it possible to hit this issue on the github master builds? The VERSION_ID above shows that this was reported against an internal IBM build.

rfrandse, 2018-05-08:
<https://gerrit.openbmc-project.xyz/10499> Fix Sensor query for derating factor returns 0
Resolves  #3012 Sensor query for user power limit always returns 0 to HTMGT

closed 2018-05-08 by geissonator