openbmc/openbmc #3016: IPL: Common.Error.InternalFailure netipmid errors on wsbmc016 with 910.1804.20180313a

cdeadmin, 2018-03-16:


rfrandse, 2018-03-20:
hit these errors during testing 

eSEL entries - "IPMI Power Off".

"Message": "xyz.openbmc_project.Common.Error.InternalFailure",
"Message": "xyz.openbmc_project.Common.Error.InternalFailure",
/gsa/ausgsa/projects/a/autoipl/FFDC/wsbmc016.180314.090012.BMC_elog.txt

"/xyz/openbmc_project/logging/entry/2": {
            "AdditionalData": [
                "_PID=1791"
            ],
            "Id": 2,
            "Message": "xyz.openbmc_project.Common.Error.InternalFailure",
            "Resolved": 0,
            "Severity": "xyz.openbmc_project.Logging.Entry.Level.Error",
            "Timestamp": 1521038096027,
            "associations": []
        },
        "/xyz/openbmc_project/logging/entry/3": {
            "AdditionalData": [
                "_PID=1791"
            ],
            "Id": 3,
            "Message": "xyz.openbmc_project.Common.Error.InternalFailure",
            "Resolved": 0,
            "Severity": "xyz.openbmc_project.Logging.Entry.Level.Error",
            "Timestamp": 1521038097491,
            "associations": []

rfrandse, 2018-03-29:
<https://gerrit.openbmc-project.xyz/9804> Check if the FRU Id is valid for FRU commands
Resolves  #3016  Common.Error.InternalFailure netipmid errors on wsbmc016 with 910.1804.20180313a

closed 2018-04-28 by geissonator