openbmc/openbmc #3017: obmc-phosphor-fand recipe needs to inherit pkgconfig

MaxPower2005, 2018-03-16:
The build process fails with
`ERROR: obmc-phosphor-fand-1.0-r1 do_compile: oe_runmake failed
ERROR: obmc-phosphor-fand-1.0-r1 do_compile: Function failed: do_compile (log file is located at /home/max/tests/tmp/work/armv6-openbmc-linux-gnueabi/obmc-phosphor-fand/1.0-r1/temp/log.do_compile.4605)                                  
ERROR: Logfile of failure stored in: /home/max/tests/tmp/work/armv6-openbmc-linux-gnueabi/obmc-phosphor-fand/1.0-r1/temp/log.do_compile.4605
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: make -j 8
| make: pkg-config: Command not found
`

Obviously a `inherit pkgconfig` is missing in the recipe.

amboar, 2018-03-21:
@MaxPower2005 any chance you can submit a patch?

amboar, 2018-06-07:
I don't think anyone has looked at fixing it. What system configuration are you building for?

BenjaminFair, 2018-06-12:
Will be fixed by https://gerrit.openbmc-project.xyz/#/c/11085/

rfrandse, 2018-06-13:
<https://gerrit.openbmc-project.xyz/11085> obmc-phosphor-fand.bb: inherit pkgconfig
Resolves  #3017 obmc-phosphor-fand recipe needs to inherit pkgconfig

closed 2018-06-14 by geissonator