openbmc/openbmc #3023: Log event when hardware watchdog causes flash side switch

geissonator, 2018-03-19:
If the hardware watchdog triggers, and an automatic flash side switch occurs, we need to log some sort of indication (and ideally reason) of the event.

Determining if it happened can be pretty specific to the BMC hardware.

Here's how you can determine your side on a AST2500.
```
# BMC side switch  (0=bmc, 1=alt)
a=$(devmem 0x1e785030 b)
echo "$((($a & 0x2) >> 1))
```

mdmillerii, 2018-03-19:
Linux has a defined watchdog API to expose the fact that the watchdog has expired.  This status must be exposed through the watchdog driver and not through `devmem`.  Some API enhancement may be needed to determine which watchdogs cause the side switch when triggered.

amboar, 2018-03-19:
Seconding @mdmillerii's position of not using `devmem`. We want to disable this kernel option, so lets not add more users.

geissonator, 2018-03-20:
@eddiejames is this something you could enhance in the watchdog driver?

geissonator, 2018-03-23:
Our best bet here is probably going to be to generate a new BMC dump when we detect this.  If we're lucky, we were using persistent journal when it happened and the dump will have that info.  Any other cookie crumbles we can gather should be added to this new dump type.  The other tricky part is to only generate the dump once.

geissonator, 2019-03-28:
Still a relevant issue, @eddiejames any chance you've got exposing this data on your list for the watchdog driver?

closed 2019-10-27 by stale[bot]