openbmc/openbmc #3024: GUI: Entering a bad username/password should display relevant error

gtmills, 2018-03-19:
Enter a bad username/password on the GUI login and the error returned is "Server unreachable".
The error should be "Incorrect username or password". 
There is code for this error, but it must not be working. 
https://github.com/openbmc/phosphor-webui/blob/master/app/login/controllers/login-controller.html#L26

[image]

iftekharuli, 2018-03-19:
This has been coded and tested before intel code merge. @gtmills do you want to fix this or what me to handle?

gtmills, 2018-03-20:
@iftekharuli If you don't mind taking this, I have some other issues I am working. Thanks!

gtmills, 2018-03-20:
looks like https://gerrit.openbmc-project.xyz/#/c/9618/  resolves this

rfrandse, 2018-03-21:
<https://gerrit.openbmc-project.xyz/9618> Show error message received from server side
Resolves  #3024 GUI: Entering a bad username/password should display relevant error

closed 2018-03-27 by geissonator