openbmc/openbmc #3027: GUI: Add activate BMC firmware and reboot BMC option

gtmills, 2018-03-20:
When you activate a BMC firmware image, the BMC must be rebooted before it will operate with the new firmware. Add a radio button in the activate confirm modal (pops up when the user activates a BMC image) and the functionality to activate a  BMC image **and** reboot the BMC. 
[image]

gtmills, 2018-03-20:
the last part of the text body should actually be "activation has completed" instead of "when it is done"

rfrandse, 2018-04-02:
<https://gerrit.openbmc-project.xyz/9888> Activate firmware and reboot if selected
Resolves  #3027 GUI: Add activate BMC firmware and reboot BMC option

closed 2018-04-09 by geissonator