openbmc/openbmc #3028: Witherspoon: Dump and logging stale REST object

gkeishin, 2018-03-21:
`Driver Info`
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v2.1-275"
VERSION_ID="v2.1-275-g8280d07"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v2.1-275"
BUILD_ID="v2.1"
root@witherspoon:~#
```
Using Test bucket for dump create/ delete/reboot, we see dump and logging leaving crumb of REST stale object with no properpty.

```
URI:https://xx.xx.xx.xx/xyz/openbmc_project//dump/entry//3, method:Get, args:{}
Response code:200, Content:{
  "data": {
    "Elapsed": 1521638146,
    "Size": 99920
  },
  "message": "200 OK",
  "status": "ok"
}

URI:https://xx.xx.xx.xx/xyz/openbmc_project//dump/action/DeleteAll, method:Post, args:{u'data': {u'data': []}, u'headers': {u'Content-Type': u'application/json'}}
Response code:200, Content:{
  "data": null,
  "message": "200 OK",
  "status": "ok"
}

.URI:https://xx.xx.xx.xx/xyz/openbmc_project//dump/entry//list, method:Get, args:{}
Response code:200, Content:{
  "data": [
    "/xyz/openbmc_project/dump/entry/3"
  ],
  "message": "200 OK",
  "status": "ok"
}

/xyz/openbmc_project/dump/entry/enumerate
{
    "/xyz/openbmc_project/dump/entry/3": {}
}

```

gkeishin, 2018-03-21:
```
Test Suite File     : Test Bmc Dump
Test Case Name      : Delete All BMC Dumps And Verify
Test Source File    : openbmc-test-automation/tests/test_bmc_dump.robot
Failure Time Stamp  : 20180321083058081941
Test Error Message  : 200 != 404
Test Documentation  :
Delete all BMC dumps and verify.
```

FFDC: 
- [x] Journald no pager
[20180321083058081941_BMC_journalctl_nopager.txt](https://github.com/openbmc/openbmc/files/1833672/20180321083058081941_BMC_journalctl_nopager.txt)

-[x] Journald Pretty print JSON 
[20180321083058081941_BMC_journalctl_pretty.txt](https://github.com/openbmc/openbmc/files/1833674/20180321083058081941_BMC_journalctl_pretty.txt)

gkeishin, 2018-03-21:
Recreated manually
```
$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -d "{\"data\": []}" -X POST  https://$BMC_IP/xyz/openbmc_project/dump/action/CreateDump
{
  "data": 2,
  "message": "200 OK",
  "status": "ok"
}
```

```
/xyz/openbmc_project/dump/enumerate
{
    "/xyz/openbmc_project/dump/entry/2": {
        "Elapsed": 1521644391,
        "Size": 99860
    },
    "/xyz/openbmc_project/dump/internal/manager": {}
}
```
Reboot BMC and verify its peristed
```
/xyz/openbmc_project/dump/enumerate
{
    "/xyz/openbmc_project/dump/entry/2": {
        "Elapsed": 1521644391,
        "Size": 99860
    },
    "/xyz/openbmc_project/dump/internal/manager": {}
}
```
Delete dump
```
$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -d "{\"data\": []}" -X POST  https://$BMC_IP/xyz/openbmc_project/dump/entry/2/action/Delete
{
  "data": null,
  "message": "200 OK",
  "status": "ok"
}
```

```
/xyz/openbmc_project/dump/enumerate
{
    "/xyz/openbmc_project/dump/entry/2": {},
    "/xyz/openbmc_project/dump/internal/manager": {}
}
```

gkeishin, 2018-03-21:
@ojayanth and @spinler took a look and saw the generic is broken here for logging and dump.

gkeishin, 2018-03-21:
This is breaking the HW CI and CT runs. So something we need to look into.

geissonator, 2018-03-21:
@ojayanth can you help drive this one? We got a regression somewhere in master it seems.

ojayanth, 2018-03-22:
There is already issue related to the due to mapper issue. https://github.com/openbmc/openbmc/issues/2751.
Looks like the failing system still have https://gerrit.openbmc-project.xyz/#/c/7579/  changes. 
First thing we need to try here is reverting this changes and see still able to recreate this issue.

dkodihal, 2018-03-30:
https://gerrit.openbmc-project.xyz/#/c/7579/  has been reverted already.

ojayanth, 2018-03-30:
In the reboot path there is no `InterfacesRemoved` signal created when issuing the Delete after reboot.  @bradbishop is looking this issue now to see is it is related to the recent commit related to well-known busname .

stewart-ibm, 2018-04-03:
Weirdly, DeleteAll seems to work around this issue.

anoo1, 2018-04-19:
Fixed with https://gerrit.openbmc-project.xyz/#/c/9895/ .

closed 2018-04-19 by anoo1