openbmc/openbmc #3030: Witherspoon: 00-bmc-eth0.network:19: Unknown lvalue 'DHCP' in section 'Route'

gkeishin, 2018-03-26:
Latest master 
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v2.1-306"
VERSION_ID="v2.1-306-gacd25c4"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v2.1-306"
BUILD_ID="v2.1"
root@witherspoon:~#
```
```
root@witherspoon:~# cat /etc/systemd/network/00-bmc-eth0.network
[Match]
Name=eth0
[Network]
LinkLocalAddressing=yes
IPv6AcceptRA=false
NTP=0.asia.pool.ntp.org
DNS=xx.xx.xx.200
Address=xx.xx.xx.42/23
Gateway=xx.xx.xx.1
[Route]
Gateway=xx.xx.xx.1
Destination=xx.xx.xx.0
DHCP=false <-----
[DHCP] 
ClientIdentifier=mac
UseDNS=true
UseNTP=true
UseHostname=true
SendHostname=true
root@witherspoon:~#
```

gkeishin, 2018-03-26:
Thanks @ratagupt for identifying the fix for it

```
Ratan Gupta [14:14]
@ngorugan can you please provide the fix..

https://gerrit.openbmc-project.xyz/#/c/9447/8..9/ethernet_interface.cpp
lineno 627-628 should move to 600
```

rfrandse, 2018-03-26:
<https://gerrit.openbmc-project.xyz/9714> Move DHCP=xxx to Network section in n/w config file.
Resolves  #3030 Witherspoon: 00-bmc-eth0.network:19: Unknown lvalue 'DHCP' in section 'Route'

closed 2018-03-28 by geissonator