openbmc/openbmc #3035: Witherspoon: Secureboot: System boots from different SBE side

cdeadmin, 2018-03-26:
<cde:info> SW421912 </cde:info>

geissonator, 2018-03-26:
The host reboot counter defined in https://github.com/openbmc/phosphor-dbus-interfaces/blob/master/xyz/openbmc_project/Control/Boot/RebootAttempts.interface.yaml is supposed to be reset to it's default on every power on.  It is decremented during reboot attempts on a single power on. If the host fails to boot RebootAttemps times (currently 3) then the bmc goes to the quiesce state.

This issue is to ensure we reset the RebootAttempts on a clean power on.

geissonator, 2018-03-26:
Be nice to just plop this into a service in the power on target but the host reboot target uses the power on target so we're going to need some logic to call systemd dbus api's to determine if the power on is a part of a reboot (in which case don't reset the reboot counter) or if it's a fresh power on (in which case we should reset the reboot counter). That logic already exists in phosphor-state-manager/host_state_manager.cpp (Host::stateActive) so would def be easier to just put into there.

rfrandse, 2018-03-28:
<https://gerrit.openbmc-project.xyz/9778> Reset host reboot attempts on fresh boot
Resolves  #3035 Reset host reboot counter on a power on

closed 2018-04-06 by geissonator