openbmc/openbmc #3040: Network: DHCP not updated after cable plug out and in

mine260309, 2018-03-28:
This is a system vendor reported bug, that the DHCP IP is not updated after the ethernet cable is plugged out, and plugged in with a different network.

Steps to reproduce:
0. Prepare two networks, e.g. Network1 with 192.168.0.xxx, and Network2 with 192.168.1.xxx, setup DHCP server in both networks;
1. BMC network is set to DHCP;
2. Plug Network1's cable, the IP is got automatically in 192.168.0.xxx.
3. Plug the cable out, and plug Network2's cable.
The IP will not be updated within the DHCP lease time.

The expected behavior is that BMC should get Network2's IP after step3.

I have verified that the issue occurs on Romulus as well.

mine260309, 2018-03-28:
According to the vendor, the issue only occurs on the NCSI interface, but not on the dedicate nic.

So it may be related to the NCSI driver not correctly reporting the status, e.g. carrier.
I will verify this tomorrow.

mine260309, 2018-03-29:
Checked the carrier status, it's still 1 when the cable is plugged out.

So the issue should be related to NCSI driver, that it does not reset `carrier` to 0 when the cable is unplugged.

@sammj could you have a look?

sammj, 2018-04-10:
Yep that seems to be the case. We can't just set the carrier to off though since the kernel will then turn off any Rx/Tx on that link - and then we can't talk to the NCSI package to see when the link comes back. Will need to have a deeper look, I remember benh talking about some generic net changes we might need.

mine260309, 2018-07-18:
@sammj Do we have any update on this?
The reason I ask is because it affects some systems with both NCSI and Dedicate ethernet as well, and the system vendor is asking for a solution.

mine260309, 2019-03-28:
Still a problem.

grace414385621, 2019-07-04:
Meet this problem, too. Do we have any update on this?

shannonliang312, 2019-10-11:
Meet this problem, too. +1

mdmillerii, 2019-10-11:
As stated above, significant changes to the kernel need to be made to enable link state monitoring, as the network layer doesn't allow receiving NCSI traffic when the link state is down.

Perhaps it needs a different interface abstraction with the network layer, where each package and port is a separate slave interface and the host NCSI link is "internal".

alvintpwang, 2019-11-14:
Hit the problem, too.

closed 2020-06-14 by stale[bot]

henbin, 2021-08-19:
Hi All,
We also encounter this issue. May I know if any solution to solve it?

Thanks.

mdmillerii, 2021-08-19:
I believe this requires a new dependant device similar to macvlan but with independent device up or like dsa but only fixed mac assignment on, so a new abstraction.    I did some exploration last October but haven't progressed or rfc on netdev.

Alternatively userspace could listen to the NCSI link state messages.   I think some are using a pure userspace NCSI stack.

henbin, 2021-08-19:
> I believe this requires a new dependant device similar to macvlan but with independent device up or like dsa but only fixed mac assignment on, so a new abstraction. I did some exploration last [...]

Hi Miller,
Thanks for your reply.

In the ncsi_aen_handler_lsc() of ncsi-aen.cI, I added the below code to update carrier state.
[image]

But it would cause Kernel to crash after I plugged out the LAN cable.
[image]

It seemed that BMC couldn't communicate with NCSI controller after netif_carrier_off() was called.

mdmillerii, 2021-08-19:
Yes, the kernel blocks transmit and recieve if the carrier is off.  That's why the slave device will be needed.

M-Kamalesh, 2024-10-25:
Hi All,
May I know if there is any solution or patch has done to solve it?

Thanks.

mdmillerii, 2024-10-25:
I'm not aware of any patches.

After studying the Linux network stack and the NC-SI specification I have come to the conclusion that to properly fix this a new ncsi architecture similar to macvlan is required.  I have a mental picture of a proposed organization and several requirements and challenges.    I haven't actually coded any yet nor have I consulted the Linux kernel network naintainers.

If you have someone with significant network stack experience I'll be happy to share my thoughts but it's a scope larger than a simple device driver let alone a patch.

If you are intessted in a tactical hack someone could write a user space monitor application (or add into your chosen dhco client) to observe the network down or network up NC-SI traffic report and via netlink and either kick the DHCP stack or take the whole network down and up hoping to avoid bouncing network and loosing all routes etc.

Alternatively consider tuning your timeouts for DHCP server contact.