openbmc/openbmc #3041: Witherspoon: OCC org::open_power::OCC::Device::Error::OpenFailure

gkeishin, 2018-03-28:
Boot Host to OS, reset BMC

Driver info:
```
root@witherspoon:~# uname -a
Linux witherspoon 4.13.16-462662bc3577d0c73e30cdc96eb28d63c89528df #1 Fri Mar 23 19:17:29 UTC 2018 armv6l GNU/Linux
root@witherspoon:~#

root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v2.1-312"
VERSION_ID="v2.1-312-g37432b4"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v2.1-312"
BUILD_ID="v2.1"
root@witherspoon:~#
```
Journald snippet:

```
Mar 28 10:04:42 witherspoon openpower-occ-control[1209]: Opening OCC device failed.
Mar 28 10:04:42 witherspoon openpower-occ-control[1209]: terminate called after throwing an instance of 'sdbusplus::org::open_power::OCC::Device::Error::OpenFailure'
Mar 28 10:04:42 witherspoon openpower-occ-control[1209]:   what():  org.open_power.OCC.Device.Error.OpenFailure: Opening OCC device failed.
Mar 28 10:04:42 witherspoon systemd[1]: Created slice system-systemd\x2dcoredump.slice.
Mar 28 10:04:43 witherspoon systemd[1]: Started Process Core Dump (PID 1275/UID 0).
```

eddiejames, 2018-03-28:
This is a real OCC error. Not our problem. From the host journal:

Mar 28 10:57:06 ws002p1 opal-prd[4876]: HBRT: HTMGT:>>processOccError(0x0x3fff9d
Mar 28 10:57:06 ws002p1 opal-prd[4876]: FW: firmware signaled OCC error for proc
Mar 28 10:57:06 ws002p1 opal-prd[4876]: HBRT: HTMGT:>>processOccError(0x0x3fff9d
Mar 28 10:57:06 ws002p1 opal-prd[4876]: FW: firmware signaled OCC error for proc
Mar 28 10:57:06 ws002p1 opal-prd[4876]: HBRT: HTMGT:>>processOccError(0x0x3fff9d
Mar 28 10:57:06 ws002p1 opal-prd[4876]: FW: firmware signaled OCC error for proc
Mar 28 10:57:06 ws002p1 opal-prd[4876]: HBRT: HTMGT:>>processOccError(0x0x3fff9d
Mar 28 10:57:06 ws002p1 opal-prd[4876]: FW: firmware signaled OCC error for proc
Mar 28 10:57:06 ws002p1 opal-prd[4876]: HBRT: HTMGT:>>processOccError(0x0x3fff9d
Mar 28 10:57:06 ws002p1 opal-prd[4876]: FW: firmware signaled OCC error for proc
Mar 28 10:57:06 ws002p1 opal-prd[4876]: HBRT: HTMGT:>>processOccError(0x0x3fff9d
Mar 28 10:57:06 ws002p1 opal-prd[4876]: FW: firmware signaled OCC error for proc
Mar 28 10:57:06 ws002p1 opal-prd[4876]: HBRT: HTMGT:<<processOccError()
Mar 28 10:57:06 ws002p1 opal-prd[4876]: HBRT: ERRL:<<sendToHypervisor()
Mar 28 10:57:06 ws002p1 opal-prd[4876]: HBRT: ERRL:<<sendErrLogToBmc
Mar 28 10:57:06 ws002p1 opal-prd[4876]: HBRT: ERRL:I>sendErrLogToBmc callout siz
Mar 28 10:57:06 ws002p1 opal-prd[4876]: HBRT: IPMI:sel: <<sendESEL
Mar 28 10:57:06 ws002p1 opal-prd[4876]: HBRT: IPMI:sel: <<process_esel
Mar 28 10:57:06 ws002p1 opal-prd[4876]: HBRT: IPMI:sel: E>DELETING l_err 0x89003
Mar 28 10:57:06 ws002p1 opal-prd[4876]: HBRT: IPMI:sel: <<send_esel o_err=890031
Mar 28 10:57:06 ws002p1 opal-prd[4876]: HBRT: IPMI:sel: E>error from reserve_sel
Mar 28 10:57:06 ws002p1 opal-prd[4876]: HBRT: ERRL:>>addProcedureCallout(0x55, 0
Mar 28 10:57:06 ws002p1 opal-prd[4876]: HBRT: ERRL:E>Error created : PLID=890031
Mar 28 10:57:06 ws002p1 opal-prd[4876]: HBRT: IPMI:rt: E>Failed sending ipmi msg
Mar 28 10:57:06 ws002p1 opal-prd[4876]: IPMI: response timeout (>5000ms)
Mar 28 10:57:01 ws002p1 opal-prd[4876]: IPMI: sending 0 bytes (netfn 0x0a, cmd 0
Mar 28 10:57:01 ws002p1 opal-prd[4876]: HBRT: IPMI:rt: calling sync a:42  len=0
Mar 28 10:57:01 ws002p1 opal-prd[4876]: HBRT: IPMI:sel: >>send_esel
Mar 28 10:57:01 ws002p1 opal-prd[4876]: HBRT: IPMI:sel: >>sel list size 1
Mar 28 10:57:01 ws002p1 opal-prd[4876]: HBRT: IPMI:sel: >>process_esel
Mar 28 10:57:01 ws002p1 opal-prd[4876]: HBRT: IPMI:sel: >>sendESEL() 1
Mar 28 10:57:01 ws002p1 opal-prd[4876]: HBRT: ERRL:getSensorInfo o_eventOffset 4
Mar 28 10:57:01 ws002p1 opal-prd[4876]: HBRT: ERRL:I>sendErrLogToBmc: sensor 12/
Mar 28 10:57:01 ws002p1 opal-prd[4876]: HBRT: ERRL:getSensorInfo o_eventOffset 4
Mar 28 10:57:01 ws002p1 opal-prd[4876]: HBRT: ERRL:sendErrLogToBmc new priority 
Mar 28 10:57:01 ws002p1 opal-prd[4876]: HBRT: ERRL:>>sendErrLogToBmc errlogId 0x
Mar 28 10:57:01 ws002p1 opal-prd[4876]: HBRT: ERRL:I>Send msg to BMC for errlogI
Mar 28 10:57:01 ws002p1 opal-prd[4876]: HBRT: ERRL:>>saveErrLogToPnor didn't sav
Mar 28 10:57:01 ws002p1 opal-prd[4876]: HBRT: ERRL:<<saveErrLogToPnor returning 
Mar 28 10:57:01 ws002p1 opal-prd[4876]: HBRT: ERRL:>>saveErrLogToPnor eid=890031
Mar 28 10:57:01 ws002p1 opal-prd[4876]: HBRT: ERRL:I>no baseServices, saving to 
Mar 28 10:57:01 ws002p1 opal-prd[4876]: HBRT: ERRL:I>Send an error log to hyperv
Mar 28 10:57:01 ws002p1 opal-prd[4876]: HBRT: ERRL:commitErrLog() called by 2600
Mar 28 10:57:01 ws002p1 opal-prd[4876]: HBRT: HTMGT:E>processOccError(): Error w
Mar 28 10:57:01 ws002p1 opal-prd[4876]: HBRT: HTMGT:E>updateForSafeMode: Safe Mo
Mar 28 10:57:01 ws002p1 opal-prd[4876]: HBRT: ERRL:>>addProcedureCallout(0x55, 0
Mar 28 10:57:01 ws002p1 opal-prd[4876]: HBRT: ERRL:>>addProcedureCallout(0x10, 0
Mar 28 10:57:01 ws002p1 opal-prd[4876]: HBRT: HTMGT:_resetOccs: OCC should have 
Mar 28 10:57:01 ws002p1 opal-prd[4876]: HBRT: HTMGT:E>_resetOccs: Safe Mode (RC:
Mar 28 10:57:01 ws002p1 opal-prd[4876]: HBRT: ERRL:E>Error created : PLID=890031
Mar 28 10:57:01 ws002p1 opal-prd[4876]: HBRT: HTMGT:bldErrLog(mod: 0x07, rc: 0x2
Mar 28 10:57:01 ws002p1 opal-prd[4876]: HBRT: HTMGT:E>_resetOCCs: Retry Threshol
Mar 28 10:57:01 ws002p1 opal-prd[4876]: HBRT: ISTEPS_TRACE:<<resetPMComplex: RC=

geissonator, 2018-04-19:
A BMC reset should not cause OCC errors, if it does we've got some issues.  @eddiejames - will need to work with OCC team to understand why the BMC reset intermittently causes errors for OCC.

closed 2018-09-21 by rfrandse

geissonator, 2018-09-21:
Lots of performance fixes have gone in, assuming these fixed it up.