openbmc/openbmc #3042: GUI: Remove "Activation" from inventory

gtmills, 2018-03-28:
"Activation" is the reverse association of the Software Image to Inventory Item association. 
More information here about the association: 
https://github.com/openbmc/phosphor-dbus-interfaces/tree/master/xyz/openbmc_project/Software#itemupdater
 
[image]

gtmills, 2018-03-28:
We could just ignore this item when we get the inventory in the GUI. But is this reverse association useful? I guess someone could look at `/xyz/openbmc_project/inventory/system/chassis/activation`  for host images or `/xyz/openbmc_project/inventory/system/chassis/motherboar/boxelder/bmc/activation` for the BMC images?

spinler, 2018-04-19:
The callouts in errors logs are also associations that show up in the inventory.  Would be good to remove those too.

[image]

geissonator, 2018-06-19:
This is blocked on #3117 so pushing into backlog until that's resolved.

gtmills, 2019-03-28:
This is done. Closing. https://gerrit.openbmc-project.xyz/c/openbmc/phosphor-webui/+/18722

closed 2019-03-28 by gtmills