openbmc/openbmc #3044: Witherspoon: Secureboot: /usr/local/share/pnor is out of space for BOOTKERNEL flash test

cdeadmin, 2018-03-28:


anoo1, 2018-03-28:
For witherspoon secureboot testing we need to do different flash tests for all OPAL partitions with right and wrong signed keys. So for BOOTKERNEL partition openbmc throwing out of space messages when we keep that image under /usr/local/share/pnor directory.
```
ls bootkernel_prod.bin -lh
-rw-r--r-- 1 root root 15M Mar 24 03:45 bootkernel_prod.bin

mv /tmp/bootkernel_prod.bin /usr/local/share/pnor/BOOTKERNEL
mv /tmp/bootkernel_prod.bin /usr/local/share/pnor/BOOTKERNEL
mv: write error: No space left on device
mv: can't preserve times of '/usr/local/share/pnor/BOOTKERNEL': No space left on device
mv: can't preserve ownership of '/usr/local/share/pnor/BOOTKERNEL': No space left on device
mv: can't preserve permissions of '/usr/local/share/pnor/BOOTKERNEL': No space left on device
```
```
root@bmc:~# ls /usr/local/share/pnor -lh
total 14496
-rw-r--r--    1 root     root       14.2M Mar 24 08:57 BOOTKERNEL
```

anoo1, 2018-03-28:
The /usr/local/share/pnor/ volume is 16MB by default, will increase it to 18MB to accommodate a larger bootkernel patch.
Need to manually remove the patch volume after updating to the fixed driver, then reboot the BMC:
```
ubirmvol /dev/mtd8 -N pnor-patch
```