openbmc/openbmc #3046: OpenBMC CLAs are not hosted anywhere

bradbishop, 2018-03-29:


bradbishop, 2018-03-29:
[OpenBMC CCLA.pdf](https://github.com/openbmc/openbmc/files/1860741/OpenBMC.CCLA.pdf)
[OpenBMC ICLA.pdf](https://github.com/openbmc/openbmc/files/1860742/OpenBMC.ICLA.pdf)

bradbishop, 2018-03-29:
Signed CLAs for maintainers are stored here, until we get a better process:

https://drive.google.com/drive/folders/1Ooi0RdTcaOWF1DWFJUAJDdN7tRKde7Nl

gtmills, 2018-03-30:
Can you give permission to the current maintainers to review?

bradbishop, 2018-04-05:
> Can you give permission

Done I think.  Let me know.

AlexanderAmelkin, 2018-04-20:
We have signed the CLA and have sent it to manager@lfprojects.org yesterday (19/04/2018 14:39 MSK). Can't see it at the google drive link so far.

gtmills, 2018-08-15:
We have a write up here: https://github.com/openbmc/docs/blob/master/maintainer-workflow.md
What is the plan long term for hosting signed CLAs?

krtaylor, 2019-03-27:
The long term plan, a tool for CLA mgmt being developed by the LF, is still pending. Until the LF tool is deployed, we will continue to use the google drive linked above.

closed 2020-11-15 by stale[bot]