openbmc/openbmc #3051: Webui stop at login

ThaiThu, 2018-03-30:
Hi all, the webui is already built to my BMC and when I browse to its address, it stops at "Login required", like this:

[image]

The Readme tell me to navigate to http://localhost:8080, but when I navigate to there, it show: connection to server refused

Thank you.

gtmills, 2018-03-30:
`http://localhost:8080` is for running the GUI locally. 
To run the GUI locally, from your phosphor-webui repo: 
 `npm install`, `npm run-script server`, and then go to http://localhost:8080. You will want to set the `Host` field on the logon page to an OpenBMC machine. The GUI uses this `Host` field to know where to send the REST calls.
 Note: You may need to go to https://<your BMC address> and accept the certificate, otherwise, your traffic to that address may be blocked by your browser.  

[image]

gtmills, 2018-03-30:
What system are you using? The only systems I know that has the WebUI installed in the image are `Witherspoon` and `x86` systems.
https://github.com/openbmc/openbmc/blob/master/meta-openbmc-machines/meta-openpower/meta-ibm/meta-witherspoon/recipes-phosphor/packagegroups/packagegroup-obmc-apps.bbappend#L3

https://github.com/openbmc/openbmc/blob/master/meta-openbmc-machines/meta-x86/meta-intel/meta-common/recipes-intel/packagegroups/packagegroup-intel-apps.bb#L50

ThaiThu, 2018-04-02:
Hi @gtmills ,
Yes, I already added it in my image.

[image]

Thank you.

ThaiThu, 2018-04-02:
Hi, @gtmills ,

When I install it locally, I can browse to the login page, and fill the host ip/user/password, but when I login, the GUI warns that the server unreachable, although I can ping my BMC, and login into it with RestfulApi.

Thank you.

gtmills, 2018-04-02:
@ThaiThu You might need to, in a separate tab on your browser, go to https://<BMC> and accept the certificate.

ThaiThu, 2018-04-03:
Hi @gtmills

Yes I do accept the certificate for the BMC machine.
[image]

Thank you.

gtmills, 2018-04-04:
Is this the latest WebUI code? https://gerrit.openbmc-project.xyz/#/c/9618/ was recently merged to show the actual login error. Before "Server unreachable" was the error displayed when any error on login occurred.  Is there anything on the console? Could you add some console.log() statements in the login path to see where it is failing?

gtmills, 2018-04-08:
@ThaiThu Any luck?

ThaiThu, 2018-04-09:
@gtmills sorry, we recently had a long holiday in our country, I we don't have hardware to check it.
I will check it as soon as possible.

Thanks.

ThaiThu, 2018-04-11:
Hi @gtmills ,

We clone new git branch, and now the webui works

Thank you !!

gtmills, 2018-04-11:
@ThaiThu Glad it worked! Can this issue be closed then?

gtmills, 2018-04-12:
Going to close this issue. Please reopen if there is still a problem.

closed 2018-04-12 by gtmills

EijahDuan, 2018-04-25:
Hi @gtmills,
"What system are you using? It looks to me like the WebUI package is not installed in the image. The only systems I know that has the WebUI installed in the image are Witherspoon and x86 systems."

Do you know how to installed WebUI in aspeed 2500? ths !

ThaiThu, 2018-04-25:
@EijahDuan 
You can refer to this link to install web ui

https://github.com/openbmc/openbmc/blob/master/meta-openbmc-machines/meta-openpower/meta-ibm/meta-witherspoon/recipes-phosphor/packagegroups/packagegroup-obmc-apps.bbappend#L3