openbmc/openbmc #3053: phosphor-ipmi-host package rebuilds every time

anoo1, 2018-03-30:
With commit https://github.com/openbmc/openbmc/commit/f02787a460b9fac4ccdb7974951a3d771da64920 , every time an image is built ( `bitbake obmc-phosphor-image` ) the phosphor-ipmi-host package gets rebuilt, including the packages depending on it, causing the build to now take much longer to rebuild simple changes unrelated to ipmi.

anoo1, 2018-03-30:
@tomjoseph83 , Brad was saying that it can probably help to have the config file go into its own package.

amboar, 2018-04-23:
I've pushed a patch which moves the configuration to a virtual that phosphor-ipmi-host RDEPENDs on:

https://gerrit.openbmc-project.xyz/#/c/9957/

geissonator, 2018-05-17:
This has been merged

anoo1, 2018-05-23:
Closing since it's being merged.

closed 2018-05-23 by anoo1