openbmc/openbmc #3054: GUI: After DownloadViaTFTP an image the firmware page should display the image

gtmills, 2018-03-30:
DownloadViaTFTP was successful but had to refresh to get the image to show up on the GUI. The GUI should refresh automatically.

gtmills, 2018-03-30:
This is similar to https://github.com/openbmc/openbmc/issues/2960 but for the DownloadViaTFTP

gtmills, 2018-04-17:
https://github.com/openbmc/openbmc/issues/1691 is the story to return the id of the newly created image. In the short time, could look at `/xyz/openbmc_project/software` objects, if a new object doesn't show up in ~1 min, we know it failed and should show the user an error. 
If a new object does show, we should update the GUI.

rfrandse, 2018-04-23:
<https://gerrit.openbmc-project.xyz/10263> Refresh the firmware page after downloading image
Resolves  #3054 GUI: After DownloadViaTFTP an image the firmware page should display the image