openbmc/openbmc #3062: romulus: phosphor-host-ipmi crashes on host boot

mine260309, 2018-04-03:
On latest OpenBMC build (cfe0fd0), `ipmid` crashes for several times during build with logs like below.
```
Apr 02 08:37:09 rom285 systemd[1]: Started Process Core Dump (PID 1584/UID 0).
Apr 02 08:37:10 rom285 ipmid[1209]: IPMI SET_SENSO
Apr 02 08:37:10 rom285 systemd[1]: phosphor-ipmi-host.service: Main process exited, code=dumped, status=11/SEGV
Apr 02 08:37:10 rom285 systemd[1]: phosphor-ipmi-host.service: Unit entered failed state.
Apr 02 08:37:10 rom285 systemd[1]: phosphor-ipmi-host.service: Failed with result 'core-dump'.
Apr 02 08:37:11 rom285 systemd[1]: phosphor-ipmi-host.service: Service hold-off time over, scheduling restart.
Apr 02 08:37:11 rom285 systemd[1]: Stopped Phosphor Inband IPMI.
Apr 02 08:37:11 rom285 systemd[1]: Started Phosphor Inband IPMI.
```

The coredump back trace:
```
#0  __GI_strcmp (p1=0x76f72c42 <g_SensorTypeMap+2> "Temp", p2=0x1 <error: Cannot access memory at address 0x1>) at /usr/src/debug/glibc/2.26-r0/git/string/strcmp.c:39
#1  0x76f0acdc in dbus_to_sensor_type (p=0x1 <error: Cannot access memory at address 0x1>) at /usr/src/debug/phosphor-ipmi-host/1.0-r1/git/sensorhandler.cpp:270
#2  0x76f0ad5c in get_type_from_interface (dbus_if=...) at /usr/src/debug/phosphor-ipmi-host/1.0-r1/git/sensorhandler.cpp:301
#3  0x76f0ada8 in find_type_for_sensor_number (num=<optimized out>) at /usr/src/debug/phosphor-ipmi-host/1.0-r1/git/sensorhandler.cpp:316
#4  0x76f23694 in updateSensorRecordFromSSRAESC (record=record@entry=0x6b1690) at /usr/src/debug/phosphor-ipmi-host/1.0-r1/git/ipmisensor.cpp:265
#5  0x76f0b828 in ipmi_sen_set_sensor (netfn=<optimized out>, cmd=cmd@entry=48 '0', request=request@entry=0x6b1690, response=response@entry=0x7ede9749,
    data_len=data_len@entry=0x7ede972c, context=0x0) at /usr/src/debug/phosphor-ipmi-host/1.0-r1/git/sensorhandler.cpp:408
#6  0x00016a28 in ipmi_netfn_router (netfn=<optimized out>, cmd=<optimized out>, request=0x6b1690, response=0x7ede9748, response@entry=0x7ede9740, data_len=0x7ede972c,
    data_len@entry=0x7ede9724) at /usr/src/debug/phosphor-ipmi-host/1.0-r1/git/ipmid.cpp:231
#7  0x00016b70 in handle_ipmi_command (m=0x6a8160, user_data=<optimized out>, ret_error=<optimized out>) at /usr/src/debug/phosphor-ipmi-host/1.0-r1/git/ipmid.cpp:379
#8  0x4de184d4 in bus_match_run (bus=bus@entry=0x6a3218, node=0x6ece00, m=m@entry=0x6a8160) at ../git/src/libsystemd/sd-bus/bus-match.c:327
#9  0x4de18708 in bus_match_run (bus=bus@entry=0x6a3218, node=0x6edc68, m=m@entry=0x6a8160) at ../git/src/libsystemd/sd-bus/bus-match.c:301
#10 0x4de18a80 in bus_match_run (bus=bus@entry=0x6a3218, node=0x6eead0, m=m@entry=0x6a8160) at ../git/src/libsystemd/sd-bus/bus-match.c:416
#11 0x4de18708 in bus_match_run (bus=bus@entry=0x6a3218, node=0x6ef938, m=m@entry=0x6a8160) at ../git/src/libsystemd/sd-bus/bus-match.c:301
#12 0x4de18a80 in bus_match_run (bus=bus@entry=0x6a3218, node=0x6adb90, m=m@entry=0x6a8160) at ../git/src/libsystemd/sd-bus/bus-match.c:416
#13 0x4de18708 in bus_match_run (bus=bus@entry=0x6a3218, node=0x6adb70, m=m@entry=0x6a8160) at ../git/src/libsystemd/sd-bus/bus-match.c:301
#14 0x4de18a80 in bus_match_run (bus=bus@entry=0x6a3218, node=0x6b0098, m=m@entry=0x6a8160) at ../git/src/libsystemd/sd-bus/bus-match.c:416
#15 0x4de18a68 in bus_match_run (bus=0x6a3218, node=0x6a3278, m=0x6a8160) at ../git/src/libsystemd/sd-bus/bus-match.c:291
#16 0x4ddc9cdc in process_match (bus=<optimized out>, m=<optimized out>) at ../git/src/libsystemd/sd-bus/sd-bus.c:2456
#17 0x4dddadd4 in process_message (m=0x6a8160, bus=0x6a3218) at ../git/src/libsystemd/sd-bus/sd-bus.c:2572
#18 process_running (ret=0x0, priority=<optimized out>, hint_priority=<optimized out>, bus=0x6a3218) at ../git/src/libsystemd/sd-bus/sd-bus.c:2622
#19 bus_process_internal (bus=0x6a3218, hint_priority=hint_priority@entry=false, priority=<optimized out>, ret=ret@entry=0x0) at ../git/src/libsystemd/sd-bus/sd-bus.c:2854
#20 0x4dddb3a8 in sd_bus_process (bus=<optimized out>, ret=ret@entry=0x0) at ../git/src/libsystemd/sd-bus/sd-bus.c:2873
#21 0x4dddb420 in io_callback (s=<optimized out>, fd=<optimized out>, revents=<optimized out>, userdata=<optimized out>) at ../git/src/libsystemd/sd-bus/sd-bus.c:3143
#22 0x4ddf541c in source_dispatch (s=s@entry=0x6ad328) at ../git/src/libsystemd/sd-event/sd-event.c:2272
#23 0x4ddf5880 in sd_event_dispatch (e=e@entry=0x6a4b70) at ../git/src/libsystemd/sd-event/sd-event.c:2631
#24 0x4ddf5a5c in sd_event_run (e=0x6a4b70, timeout=<optimized out>) at ../git/src/libsystemd/sd-event/sd-event.c:2690
#25 0x00015e74 in main (argc=<optimized out>, argv=<optimized out>) at /usr/src/debug/phosphor-ipmi-host/1.0-r1/git/ipmid.cpp:609
```

mine260309, 2018-04-03:
The problem is that BMC gets a sensor with id 0, then it tries to call below methods:
`find_openbmc_path()` -> `legacy_dbus_openbmc_path()`

In `legacy_dbus_openbmc_path()`, it calls `getObjectFromByteId` to `org.openbmc.managers.System`, and get empty path without error.
```
# busctl call org.openbmc.managers.System /org/openbmc/managers/System org.openbmc.managers.System getObjectFromByteId  sy SENSOR 0
(ss) "" "org.openbmc.SensorValue"
```
After that, the code tries to use the empty path and gets the crash.

rfrandse, 2018-04-03:
<https://gerrit.openbmc-project.xyz/9902> Handle sensors that can not be found in SystemManager
Resolves  #3062 romulus: phosphor-host-ipmi crashes on host boot

closed 2018-04-10 by geissonator