openbmc/openbmc #3063: How can I add a sensor to OpenBMC?

momozizi1, 2018-04-04:
I'm beginner engineer of OpenBMC and embedded Linux.

I added the temp sensor lm75 on I2C bus our server board.
I did the follows step:
   1) Add I2C sensor config to devicetree
`temp1: temp1@48{
                        compatible = "national,lm75";
                        reg = <0x48>;
                        status = "okay";
                };
`
   2) Add sensor recipe(?) to recipes-phosphor/sensors/phosphor-hwmon%/obmc/hwmon/ahb/apb/i2c@1e78a000/i2c-bus@300/temp1@48.conf and bbappend file.
`MODE_temp1 = "label"
LABEL_temp1=cpu1
WARNLO_temp1=10000
WARNHI_temp1=40000`

   3) Build image.

In BMC, It seems to works good when I read use 'cat' cmd.

> ./sys/devices/platform/ahb/ahb:apb/ahb:apb:i2c@1e78a000/1e78a300.i2c-bus/i2c-7/7-0048/hwmon/hwmon1/temp1_input
26500

Then, I want to mapping sensor to d-bus xyz/openbmc_project/sensor. (I did not comprehend structure of Linux.. its right?) 
I searched xyz/openbmc_project/sensor structure with curl cmd, It is not exist.

> curl -b cjar -k https://192.168.10.221/xyz/openbmc_project/
"data": [
    "/xyz/openbmc_project/dump", 
    "/xyz/openbmc_project/software", 
    "/xyz/openbmc_project/control", 
    "/xyz/openbmc_project/network", 
    "/xyz/openbmc_project/logging", 
    "/xyz/openbmc_project/inventory", 
    "/xyz/openbmc_project/user", 
    "/xyz/openbmc_project/time", 
    "/xyz/openbmc_project/led", 
    "/xyz/openbmc_project/state"
  ], 
  "message": "200 OK", 
  "status": "ok"

Is there something as manual to add sensor or understand openbmc_project structure?
I need help.
Thank you.

mine260309, 2018-04-04:
>
> I'm beginner engineer of OpenBMC and embedded Linux.
>
> I added the temp sensor lm75 on I2C bus our server board.
> I did the follows step:
>
> Add I2C sensor config to devicetree
> temp1: temp1@48{ compatible = "national,lm75"; reg = <0x48>; status = "okay"; };
>
> Add sensor recipe(?) to recipes-phosphor/sensors/phosphor-hwmon%/obmc/hwmon/ahb/apb/i2c@1e78a000/i2c-bus@300/temp1@48.conf and bbappend file.
> MODE_temp1 = "label" LABEL_temp1=cpu1 WARNLO_temp1=10000 WARNHI_temp1=40000

This is OK, but there is actually no need to use `MODE_temp1 = "label"` here,
because you know temp1 is for cpu1.

> Build image.
>
> In BMC, It seems to works good when I read use 'cat' cmd.
>
> ./sys/devices/platform/ahb/ahb:apb/ahb:apb:i2c@1e78a000/1e78a300.i2c-bus/i2c-7/7-0048/hwmon/hwmon1/temp1_input
> 26500
>
> Then, I [...]

You are doing right. But since you did not see the sensor from DBUS, you may
want to check:
1. Where does /sys/class/hwmon/hwmon1/of-node point to?
   It should be the file in device tree, make sure it matches your .conf file
2. Make sure the conf file is correctly install into
   "/etc/default/obmc/hwmon/ahb/apb/..."
3. Check if phosphor-hwmon-readd is working on your device.
   e.g. by ps | grep hwmon

mine260309, 2018-04-04:
>
> Thanks for reply.
>
> Where does /sys/class/hwmon/hwmon1/of-node point to?
> It should be the file in device tree, make sure it matches your .conf file
>
> Make sure the conf file is correctly install into
> "/etc/default/obmc/hwmon/ahb/apb/..."
>
> I did not make conf file. I don't know where I should do that. Can you provide to me some example?

The conf file is the one in your original mail
(recipes-phosphor/sensors/phosphor-hwmon%/obmc/hwmon/ahb/apb/i2c@1e78a000/i2c-bus@300/temp1@48.conf)

>
> Check if phosphor-hwmon-readd is working on your device.
> phosphor-hwmon-readd is exist /usr/sbin folder.
> But it is not working when I check 'ps | grep hwmon'.
>
> Do I have to study the source code of 'phosphor-hwmon' for understand to adding sensor? or more things?

No, if you have correctly put the conf file, the phosphor-hwmon will be
started by udev rule automatically and it will work just fine.

momozizi1, 2018-04-04:
You gave me the right direction.
Thank you. I will try more.

closed 2018-04-04 by momozizi1

aaronw2, 2019-01-03:
I'm basically having a similar problem to the original author here. In my case, the .conf files are not being installed. As far as I can tell they are in the correct subdirectories which match what is shown from the of_node entries. Do I need to do anything special for the conf files to be installed?

Student2C0, 2021-09-16:
Hi. I am trying to add a sensor to my OpenBMC environment on RaspberryPi 4-b platform but struggling this very much.
So,could someone please give me advice about my case ?

I am writing my steps below.

**Step#1: I physically connected an ADS1115 sensor to my Rpi4 board and saw where it is connected on my Raspberry Pi OS .**
  I made sure the sensor has 0x48 address and is connected to 12C Bus #1 which is shared with i2C@7e804000 on BCM2835 before I actually started to build openbmc image for this platform.

    pi@raspberrypi:/boot/overlays $ sudo i2cdetect -l
    i2c-1	i2c       	bcm2835 (i2c@7e804000)          	I2C adapter

    pi@raspberrypi:/boot/overlays $ 
    pi@raspberrypi:/boot/overlays $ sudo i2cdetect -y 1
         0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
    00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
    10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
    20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
    30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
    40: -- -- -- -- -- -- -- -- 48 -- -- -- -- -- -- -- 
    50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
    60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
    70: -- -- -- -- -- -- -- --                         
    pi@raspberrypi:/boot/overlays $

**Step#2: I set up a new OpenBMC environment for Rpi 4 board.**

**Step#2-1: git clone**

     yocto@yocto-VM:~/Rpi4-2.8.0$ git clone https://github.com/openbmc/openbmc.git -b 2.8.0

I chose 2.8.0 due to #3812.

**Step#2-2: Set TEMPLATECONT**

    yocto@yocto-VM:~/Rpi4-2.8.0/openbmc$ export TEMPLATECONF=meta-evb/meta-evb-raspberrypi/conf
    yocto@yocto-VM:~/Rpi4-2.8.0/openbmc$ echo $TEMPLATECONF
        meta-evb/meta-evb-raspberrypi/conf

**Step#2-3: Set open-env**

    yocto@yocto-VM:~/Rpi4-2.8.0/openbmc/. openbmc-env
    yocto@yocto-VM:~/Rpi4-2.8.0/openbmc/build

**Step#2-4: Modify local.conf**

    yocto@yocto-VM:~/Rpi4-2.8.0/openbmc/build/conf$ diff local.conf local.conf.bk 
    1c1
    < MACHINE ??= "raspberrypi4"
    ---
    > MACHINE ??= "raspberrypi"
    18,22d17
    < 
    < DL_DIR ?= "/home/yocto/Yocto/downloads-4-2.8.0"
    < SSTATE_DIR ?= "/home/yocto/Yocto/sstate-cache-4-2.8.0"
    < IMAGE_FSTYPES += " rpi-sdimg"
    < 

**Step#2-5: Modify image_types_phosphor.bbclass file.**

    yocto@yocto-VM:~/Rpi4-2.8.0/openbmc/meta-phosphor/classes$ diff image_types_phosphor.bbclass image_types_phosphor.bbclass.bk 
    38c38
    < FLASH_SIZE ?= "131072"
    ---
    > FLASH_SIZE ?= "32768"

This is due to #3590

**Step#3: Set up a config file for ads1115 sensor**

    yocto@yocto-VM:~/Rpi4-2.8.0/openbmc/meta-evb/meta-evb-raspberrypi/recipes-phosphor/sensors/phosphor-hwmon/obmc/hwmon/soc/i2c@7e804000$ ls -la
[... 189 lines cut ...]
    Reading rules file: /lib/udev/rules.d/70-mouse.rules
    Reading rules file: /lib/udev/rules.d/70-touchpad.rules
    Reading rules file: /lib/udev/rules.d/75-net-description.rules
    Reading rules file: /lib/udev/rules.d/75-probe_mtd.rules
    Reading rules file: /lib/udev/rules.d/78-sound-card.rules
    Reading rules file: /lib/udev/rules.d/80-drivers.rules
    Reading rules file: /lib/udev/rules.d/80-net-setup-link.rules
    Reading rules file: /lib/udev/rules.d/80-obmc-console-uart.rules
    Reading rules file: /etc/udev/rules.d/99-com.rules
    /etc/udev/rules.d/99-com.rules:10 Invalid value "/bin/sh -c 'ALIASES=/proc/device-tree/aliases; if cmp -s $ALIASES/uart0 $ALIASES/serial0; then echo 0;elif cmp -s $ALIASES/uart0 $ALIASES/serial1; then echo 1; else exit 1; fi'" for PROGRAM (char 58: invalid substitution type), ignoring, but please fix it.
    /etc/udev/rules.d/99-com.rules:21 Invalid value "/bin/sh -c 'ALIASES=/proc/device-tree/aliases; if cmp -s $ALIASES/uart1 $ALIASES/serial0; then echo 0; elif cmp -s $ALIASES/uart1 $ALIASES/serial1; then echo 1; else exit 1; fi '" for PROGRAM (char 58: invalid substitution type), ignoring, but please fix it.
    Reading rules file: /lib/udev/rules.d/99-systemd.rules
    Reading rules file: /etc/udev/rules.d/can.rules
    Reading rules file: /etc/udev/rules.d/touchscreen.rules
    >>Failed to open device '/sys/class/hwmon/hwmon': No such device
    Unload module index
    Unloaded link configuration context.
    root@raspberrypi4:~# 

As you can see, there is no sensor found on my Rpi platform and I am wondering if I overlooked something or took wrong steps.
There is one thing I suspect is this is a Raspberry Pi board and might need to change i2c interface to enable by rspi-config before I use i2c but I didn't do this.
Now, I will try to do this but I am not sure if this works.
So, I would really appreciate someone give me advice about this.

[bcm2711-rpi-4-b.dts.txt](https://github.com/openbmc/openbmc/files/7179096/bcm2711-rpi-4-b.dts.txt)

AloneWarrior11011, 2025-04-07:
> On Wed, Apr 4, 2018 at 12:20 PM, momozizi1 ***@***.***> wrote:
> 
>  Thanks for reply.
> 
>  Where does /sys/class/hwmon/hwmon1/of-node point to?
>  It should be the file in device tree, make sure it matches your .conf file
> 
>  Make sure the conf file is correctly install into
>  "/etc/default/obmc/hwmon/ahb/apb/..."
> 
>  I did not make conf file. I don't know where I should do that. Can you provide to me some example?
> The conf file is the one in your original mail
> (recipes-phosphor/sensors/phosphor-hwmon%/obmc/hwmon/ahb/apb/i2c@1e78a000/i2c-bus@300/temp1@48.conf)
> 
>  Check if phosphor-hwmon-readd is working on your device.
>  phosphor-hwmon-readd is exist /usr/sbin folder.
>  But it is not working when I check 'ps | grep hwmon'.
> 
>  Do I have to study the source code of 'phosphor-hwmon' for understand to adding sensor? or more things?
> No, if you have correctly put the conf file, the phosphor-hwmon will be
> started by udev rule automatically and it will work just fine.
> […](#)

In my case the configuration file of tmp75@4d perfectly installed into /etc/default/obmc/hwmon/i2c-1/[tmp75@4d.conf](mailto:tmp75@4d.conf), but devide node not created, tmp75.dts file content looks
fine, but I think there might be some issue in linux-raspberrypi_%.bbappend recipe that don't place dts file in the right location while building openbmc image for raspberrypi.

If I get your complete recipe structure, it would really helpful for me to figure out the issue.

if you need complete recipe structure to get the actual issue I will post here as well.

```
**Configuration file been installed :** 

root@raspberrypi3:~# ls -la /etc/default/obmc/hwmon/i2c-1
drwxr-xr-x    2 root     root          1024 Apr  5  2011 .
drwxr-xr-x    3 root     root          1024 Apr  5  2011 ..
-rw-r--r--    1 root     root           211 Apr  5  2011 tmp75@4d.conf
```

```
**journalctl shows skipped logs :** 

root@raspberrypi3:~# journalctl | grep Hwmon
Dec 19 21:25:51 raspberrypi3 systemd[1]: Created slice Slice /system/xyz.openbmc_project.Hwmon.
Dec 19 21:25:54 raspberrypi3 systemd[1]: Phosphor Hwmon Poller was skipped because of an unmet condition check (ConditionFileNotEmpty=/etc/default/obmc/hwmon/devices/virtual/thermal/thermal_zone0/hwmon0.conf).

```

```
**Conf file been installed to the right directory :** 

root@raspberrypi3:~# ls -l /etc/default/obmc
drwxr-xr-x    3 root     root          1024 Apr  5  2011 hwmon
drwxr-xr-x    2 root     root          1024 Apr  5  2011 watchdog
```

```
**No entry here for tmp75 sensor (device tree entries) :** 

root@raspberrypi3:~# ls -la /sys/firmware/devicetree/base
-r--r--r--    1 root     root             4 Apr  4 10:53 #address-cells
-r--r--r--    1 root     root             4 Apr  4 10:53 #size-cells
drwxr-xr-x   25 root     root             0 Dec 19 21:25 .
drwxr-xr-x    3 root     root             0 Dec 19 21:25 ..
drwxr-xr-x    2 root     root             0 Apr  4 10:53 __overrides__
drwxr-xr-x    2 root     root             0 Apr  4 10:53 __symbols__
drwxr-xr-x    2 root     root             0 Apr  4 10:53 aliases
drwxr-xr-x    2 root     root             0 Apr  4 10:53 arm-pmu
drwxr-xr-x    3 root     root             0 Apr  4 10:53 axi
drwxr-xr-x    2 root     root             0 Apr  4 10:53 cam0_clk
[... 82 lines cut ...]
Reading rules file: /usr/lib/udev/rules.d/80-obmc-console-uart.rules
Reading rules file: /usr/lib/udev/rules.d/81-net-dhcp.rules
Reading rules file: /etc/udev/rules.d/99-com.rules
/etc/udev/rules.d/99-com.rules:2 Unknown group 'i2c', ignoring.
/etc/udev/rules.d/99-com.rules:3 Unknown group 'spi', ignoring.
/etc/udev/rules.d/99-com.rules:4 Unknown group 'gpio', ignoring.
/etc/udev/rules.d/99-com.rules:10 Unknown group 'gpio', ignoring.
Reading rules file: /usr/lib/udev/rules.d/99-systemd.rules
Reading rules file: /etc/udev/rules.d/can.rules
Reading rules file: /etc/udev/rules.d/fb.rules
Reading rules file: /etc/udev/rules.d/touchscreen.rules
hwmon0: /usr/lib/udev/rules.d/70-hwmon.rules:1 RUN '/usr/bin/start_hwmon.sh start %p %E{OF_FULLNAME}'
Properties:
  DEVPATH=/devices/virtual/thermal/thermal_zone0/hwmon0
  ACTION=add
  SUBSYSTEM=hwmon
  USEC_INITIALIZED=307964567
Queued commands:
  RUN{program} : /usr/bin/start_hwmon.sh start /devices/virtual/thermal/thermal_zone0/hwmon0 
Unloaded link configuration context.

```
device not created in sysfs despite being tmp75@4d.conf installed into /etc/default/obmc/hwmon/i2c-1/tmp75@4d.conf. 

do resolve my problem, I would be thankful to you always...!