openbmc/openbmc #3064: Legacy code update is broken.

mine260309, 2018-04-04:
As https://github.com/openbmc/openbmc/commit/c3f74dc48e971a9a5421fa8b41f83e8a411901f3 removes the bmc control package, the legacy code update is broken:
1. It needs to call `PrepareForupdate()` in `bmc_update.py`
2. `bmc_update.py` replies on bmc control to do `warmReset()`.

As bmc control package is removed the above is broken.
We need to fix `bmc_update.py` for now.

mine260309, 2018-04-04:
After `bmc_update.py` is fixed, the next reboot of BMC result in maintainance mode:
```
rofs = mtd4 squashfs rwfs = mtd5 jffs2
copy-files-to-ram copy-base-filesystem-to-ram
update: --copy-files --clean-saved-files --no-restore-files
[    6.648236] jffs2: notice: (525) jffs2_build_xattr_subsystem: complete building xattr subsystem, 28 of xdatum (27 unchecked, 1 orphan) and 95 of xref (1 dead, 0 orphan) found.
Skipping empty update of rwfs.
copy-files-to-ram copy-base-filesystem-to-ram
mount: can't setup loop device: No such device or address
Running with read-write overlay in RAM for this boot.
No state will be preserved unless flash update performed.
[   11.770647] overlayfs: upper fs does not support xattr.
chroot: can't execute '/bin/sh': No such file or directory

Unable to confirm /sbin/init is an executable non-empty file
in merged file system mounted at /root.

Change Root test failed!  Invoking emergency shell.
Enter password to try to manually fix.
After fixing run exit to continue this script, or reboot -f to retry, or
touch /takeover and exit to become PID 1 allowing editing of this script.
Give root password for maintenance
```

amboar, 2018-04-04:
Wow, I  remember discussing that error message with @mdmillerii when I first started on the project. Somewhat surprised to hit it.

Do we have any logs from the update of the rootfs prior to the reboot?

mine260309, 2018-04-04:
@amboar Yup, here's the log before reboot:
```
...
[  OK  ] Stopped Apply Kernel Variables.
[  OK  ] Stopped Network Time Synchronization.
[  OK  ] Stopped Load/Save Random Seed.
[  OK  ] Stopped Create Volatile Files and Directories.
[  OK  ] Stopped target Local File Systems.
         Unmounting /var/volatile...
         Unmounting Temporary Directory (/tmp)...
[FAILED] Failed unmounting /var/volatile.
[FAILED] Failed unmounting Temporary Directory (/tmp).
[  OK  ] Stopped target Swap.
[  OK  ] Reached target Unmount All Filesystems.
[  OK  ] Stopped target Local File Systems (Pre).
[  OK  ] Stopped Remount Root and Kernel File Systems.
[  OK  ] Stopped Create Static Device Nodes in /dev.
[  OK  ] Reached target Shutdown.
[11486.562102] watchdog: watchdog0: watchdog did not stop!
[11486.599799] systemd-shutdow: 38 output lines suppressed due to ratelimiting
[11486.643729] systemd-shutdown[1]: Sending SIGTERM to remaining processes...
[11486.671196] systemd-journald[538]: Received SIGTERM from PID 1 (systemd-shutdow).
[11486.692236] systemd-shutdown[1]: Sending SIGKILL to remaining processes...
[11486.714387] systemd-shutdown[1]: Hardware watchdog 'aspeed_wdt', version 0
[11486.727670] systemd-shutdown[1]: Unmounting file systems.
[11486.734802] systemd-shutdown[1]: Remounting '/var/volatile' read-only with options ''.
[11486.743436] systemd-shutdown[1]: Unmounting /var/volatile.
[11486.749486] systemd-shutdown[1]: Remounting '/tmp' read-only with options ''.
[11486.757334] systemd-shutdown[1]: Unmounting /tmp.
[11486.762516] systemd-shutdown[1]: Remounting '/' read-only with options 'lowerdir=run/initramfs/ro,upperdir=run/initramfs/rw/cow,workdir=run/initramfs/rw/work'.
[11486.791930] systemd-shutdown[1]: Remounting '/run/initramfs/rw' read-only with options ''.
[11486.844950] watchdog: watchdog0: watchdog did not stop!
[11486.850231] shutdown: 16 output lines suppressed due to ratelimiting
shutdown: reboot --log-level 6 --log-target kmsg --log-color
+ sort -r
+ awk /oldroot|mnt/ { print $2 }
+ umount /oldroot/sys/kernel/debug
+ umount /oldroot/sys/fs/cgroup/unified
+ umount /oldroot/sys/fs/cgroup/systemd
+ umount /oldroot/sys/fs/cgroup
+ umount /oldroot/sys
+ umount /oldroot/proc
+ umount /oldroot/dev/shm
+ umount /oldroot/dev/pts
+ umount /oldroot/dev
+ umount /oldroot
+ umount /mnt/initramfs/rw
+ umount /mnt/initramfs/ro
+ umount /mnt
+ set +x
Remaining mounts:
tmpfs / tmpfs rw,nosuid,nodev,mode=755 0 0
sys /sys sysfs rw,relatime 0 0
dev /dev devtmpfs rw,relatime,size=213908k,nr_inodes=53477,mode=755 0 0
tmpfs /run tmpfs rw,nosuid,nodev,mode=755 0 0
proc /proc proc rw,relatime 0 0
[11487.352607] reboot: Restarting system
```

mine260309, 2018-04-04:
The issue is that mounting rofs as loop device failed:
```
/ # mount /run/image-rofs /run/initramfs/ro -t squashfs -o ro,loop
mount: can't setup loop device: No such device or address
```
And the kernel config seems to be lacking of `CONFIG_BLK_DEV_LOOP`, although it's defined in defconfig.

rfrandse, 2018-04-04:
<https://gerrit.openbmc-project.xyz/9920> AST2500: Add loop device support in defconfig
Resolves  #3064 Legacy code update is broken.

closed 2018-04-05 by geissonator

rfrandse, 2018-04-05:
<https://gerrit.openbmc-project.xyz/9919> bmc_update.py: Use BMC state manager to reboot BMC
Partially resolves  #3064 Legacy code update is broken.