openbmc/openbmc #3065: Building errors for meta-q71l

Ananth-CG, 2018-04-05:
Hi,
I'm trying to build the OpenBMC image for **meta-q71l**. The procedure that I am following is:

1. Downloaded the source as mentioned in the Section-2 of Setting up your OpenBMC project in _https://github.com/openbmc/openbmc_
2. Then build the source:
          (i) export TEMPLATECONF=meta-openbmc-machines/meta-x86/meta-quanta/meta-q71l/conf
          (ii) . openbmc-env
          (iii) bitbake obmc-phosphor-image
The first question is - Is this the right way to build for q71l ?

The first error that I encountered was that extra-properties.yaml is not found. I resolved this error by copying this file in the required directory from another build.

Now, the error is - 
_.so -lsdbusplus -lsystemd -lmapper
| **./.libs/libwritefrudata.so: undefined reference to `frus'**
| collect2: error: ld returned 1 exit status
| Makefile:595: recipe for target 'phosphor-read-eeprom' failed
| make[1]: *** [phosphor-read-eeprom] Error 1
| make[1]: Leaving directory '/home/ananth/IBM_OpenBMC/openbmc/build/tmp/work/armv5e-openbmc-linux-gnueabi/phosphor-ipmi-fru/1.0-r1/build'
| Makefile:416: recipe for target 'all' failed
| make: *** [all] Error 2
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /home/ananth/IBM_OpenBMC/openbmc/build/tmp/work/armv5e-openbmc-linux-gnueabi/phosphor-ipmi-fru/1.0-r1/temp/log.do_compile.21348)
ERROR: Task (/home/ananth/IBM_OpenBMC/openbmc/meta-phosphor/common/recipes-phosphor/ipmi/phosphor-ipmi-fru.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 2863 tasks of which 2856 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/ananth/IBM_OpenBMC/openbmc/meta-phosphor/common/recipes-phosphor/ipmi/phosphor-ipmi-fru.bb:do_compile
Summary: There were 2 ERROR messages shown, returning a non-zero exit code._

[image]

Any idea why these errors related to ipmi-fru occur ? Is there a dependency from openpower to x86 machine that needs to be taken care of ? Kindly suggest.

Thanks in advance.

geissonator, 2018-04-05:
Any thoughts @dkodihal or @tomjoseph83 ?

dkodihal, 2018-04-06:
I'm not sure why you hit this because the relevant recipes seem to be in place (the one for installing extra-properties.yaml is there as well) : https://github.com/openbmc/openbmc/tree/master/meta-openbmc-machines/meta-x86/meta-quanta/meta-q71l/recipes-phosphor/ipmi

I tried building myself and hit a different error :
```
ERROR: ExpansionError during parsing /esw/san5/dkodihal/obmc/openbmc/meta-phosphor/common/recipes-phosphor/sensors/phosphor-hwmon-config-mrw.bb                     | ETA:  0:00:38
Traceback (most recent call last):
bb.data_smart.ExpansionError: Failure expanding variable SYSTEMD_ENVIRONMENT_FILE_${PN}_append_quanta-q71l[:=], expression was  ${@compose_list(d, 'ENVS', 'ITEMS')} which triggered exception NameError: name 'compose_list' is not defined
```

I wonder if @pstrinkle has thoughts on this.

bradbishop, 2018-04-12:
@pstrinkle Can you help out here?  @yuennancy fyi...

pstrinkle, 2018-04-12:
So we hit this downstream as well when we picked up the ipmi-fru stuff in phosphor-host-ipmid because the defaults don't work or weren't applied to other platforms that don't use mrw

pstrinkle, 2018-04-12:
@bradbishop @geissonator Is the quanta-q71l machine part of the CI for openbmc builds?  As this should have been caught when the phosphor-host-ipmid recipe stuff was updated.

geissonator, 2018-04-12:
Nope, if you want to add it into https://github.com/openbmc/openbmc-build-scripts/blob/master/build-setup.sh#L114 and get this regression fixed then I can add it into our CI job.

pstrinkle, 2018-04-12:
Ok, I'll try to figure out what CL broke the recipes for non-mrw platforms.  I can just submit my own preferred_version for the thing, but any future platforms that are non-mrw will also fail in a weird way.

bradbishop, 2018-04-12:
Great!  But I'd guess you'd both agree..this is a short term fix.  The 'public' resources we have aren't going to scale to the point of being able to test everyones platforms for them.  I'd guess the other folks that have machine trees upstream will follow suit and ask for their machines to be tested for them as well (I know I would).

We need a way for system maintainers like @pstrinkle to test their own platforms against pull requests, and then push those results back up to the pull request.

What do you guys think we need to change, to send the clear message that the "community" does _not_ test peoples platforms for them?  You have to do that yourself.

With Poky, this is easy.  Poky itself does all its testing on its reference platforms (the qemu models).  Projects based on poky (like OpenBMC for example) are responsible for testing Poky changes themselves - the Poky project does not do this for them).  Poky has an additional constraint in that it doesn't even allow support for other platforms in its tree.  This is the direction I'd like to go as well.  This is precisely why I've been working towards moving us to tracking Poky HEAD - so we can test Poky changes and discover breakage as it happens. 

But OpenBMC doesn't really have reference platforms.  At least not to any agreed-to metric.  So I can understand the confusion where the responsibility lies when it comes to doing this kind of regression testing.  Initially I thought the concept of reference platforms would be almost impossible, considering the companies involved and and the scope of the project (servers, switches, raid appliances, you name it...), but I wonder if we could come up with some kind of metric for establishing reference platforms?  Having that makes it really obvious where the testing responsibility lies.  Thoughts?  Possible metrics for reference platform classification?

pstrinkle, 2018-04-12:
I'm working on parsing this, but my first thought is -- if we have machines registered, how many do we anticipate over the next few years?  If it's dozens, at what point is it too much for the CI?  I honestly don't have a sense for this.  When I do builds, like I'm doing now to track down the tuple mock issue, it takes ~1 hour to build things, but that's without a shared cache.  With OpenBMC it's somewhat easy to update a recipe and test your own platform, but even downstream we've had failures along these lines.  Specifically with the ipmi-fru change, we fixed it for the quanta-q71l downstream and then were like, "ok it's done," and broke two other platforms we maintain.  Only when I do a merge downstream, say given a tag like when I merge down from 2.1 do I actually test all of our platforms, so it's certainly not a problem we've addressed ourselves.  That said, with the CI builds -- at some point it won't really scale, and I wonder where we'll hit that -- and per your comment, who's responsible for testing that something doesn't break other platforms.

> What do you guys think we need to change, to send the clear message that the "community" does not test peoples platforms for them? You have to do that yourself.

With groups that don't follow HEAD closely, I think it's fine because one can certainly manage their acceptance and pace of pulling in changes.  However, this'll happen where, I don't test the quanta-q71l regularly.  I left it in a working state and haven't been poking it much as the project is slowing down in terms of updates.  Once a platform from any company has reached a certain level fo maturity, it's reasonable that the maintainer won't check in with every patch, but just with some regularity.

pstrinkle, 2018-04-12:
> IMHO this is not how its supposed to work. The way I look at it, having your platform upstream is a privilege. By your I don't mean you @pstrinkle but in the general sense. If its in the upstream tree, it should work. All the time. And someone has to do that maintenance. Thats what it means to be a maintainer.

Fair enough.  With many people changing projects regularly from the many companies now involved, it may be worthwhile bringing this point up with the TSC.  Make sure all contributing companies have this in mind.  I know for a fact that although I have been allocating time (maybe not enough recently) with phosphor-hwmon maintenance and review, I haven't been allocating time for quanta-q71l upstream maintenance.  Should the quanta-q71l be deprecated?  I wouldn't object, but I imagine others might -- so getting all companies on the same page w.r.t the guidance for their platforms is necessary.  I understand completely that if someone isn't actively maintaining a platform that it will go into disrepair and that it is basically just dragging.  I know personally as a member of the NPI team and not core BMC, I will be moving onto another platform in a couple of months and passing the mantle of maintainership of quanta-q71l to the core team.  However, at that point, it might make sense to drop the quanta-q71l -- I don't know.

bradbishop, 2018-04-12:
Indeed.  I'm not trying to single you or quanta-q71l out...it just happened that way because the issue happened to be opened against that platform.  As you say I just think its a conversation the project needs to have.

pstrinkle, 2018-04-12:
@bradbishop I'm actually glad this issue has been pushed to the top. :) as things start to move forward a lot of things that were more or less "ok" before will shake out.  And I don't mind quanta-q71l being on the chopping block because it was an early out-side platform and we need to evaluate the long-term maintenance -- within Google and for OpenBMC.

amboar, 2018-04-13:
#3088 is roughly related by @dkodihal 's comment [above](#issuecomment-379180768)

pstrinkle, 2018-04-13:
I should have a patch for this later today.

pstrinkle, 2018-04-13:
I got it past the one issue, but now uboot won't compile.

ERROR: u-boot-1_v2016.07+gitAUTOINC+e4c85cd4de-r0 do_patch: Command Error: 'quilt --quiltrc /usr/local/google/git/workspaces/openbmc/build/tmp/work/quanta_q71l-openbmc-linux-gnueabi/u-boot/1_v2016.07+gitAUTOINC+e4c85cd4de-r0/recipe-sysroot-native/etc/quiltrc push' exited with 0  Output:
/bin/sh: quilt: command not found
ERROR: u-boot-1_v2016.07+gitAUTOINC+e4c85cd4de-r0 do_patch: Function failed: patch_do_patch
ERROR: Logfile of failure stored in: /usr/local/google/git/workspaces/openbmc/build/tmp/work/quanta_q71l-openbmc-linux-gnueabi/u-boot/1_v2016.07+gitAUTOINC+e4c85cd4de-r0/temp/log.do_patch.202336
ERROR: Task (/usr/local/google/git/workspaces/openbmc/meta-openbmc-bsp/meta-aspeed/common/recipes-bsp/u-boot/u-boot_2016.07.bb:do_patch) failed with exit code '1'

pstrinkle, 2018-04-13:
Just for the bug itself, the problem is compose_list comes from a bbclass that was no longer being inherited from the parent recipe.

mdmillerii, 2018-04-13:
@pstrinkle  It appears you are missing the `quilt` tool.  
Either that or the recipe managed to hide it from the effective path. I would have expected it to be a sanity check requirement or downloaded by yocto.  Maybe it downloads but misses a dependency ...

amboar, 2018-04-13:
Looks like i broke u-boot for everyone but myself. There's a patch here:

https://gerrit.openbmc-project.xyz/#/c/10117/

pstrinkle, 2018-04-16:
I have two patches queued that should address this: 
https://gerrit.openbmc-project.xyz/#/c/10126/
https://gerrit.openbmc-project.xyz/#/c/10127/

closed 2018-06-04 by Ananth-CG