openbmc/openbmc #3069: Code update does not got to Active, stuck in Activating

geissonator, 2018-04-05:
This popped up with a recent master build, the suspicion is https://github.com/openbmc/phosphor-bmc-code-mgmt/commit/2225fda1dbe6e5fe87c009174d941369ab7e7b3a since that was what was on the system when this fail was hit but no hard proof.

```
v2.1-377-g4d02df1

/xyz/openbmc_project/software/69d56586
{
    "associations": [
        [
            "inventory", 
            "activation", 
            "/xyz/openbmc_project/inventory/system/chassis/motherboard/boxelder/bmc"
        ]
    ], 
    "Activation": "xyz.openbmc_project.Software.Activation.Activations.Activating", 
    "Version": "ibm-v2.0-0-r45-0-g4fc013e", 
    "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.BMC", 
    "Progress": 80, 
    "Path": "/tmp/images/69d56586", 
    "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.Active"
}
```

```
Apr 05 19:26:02 w54 dropbear[1631]: PAM password auth succeeded for 'root' from ::ffff:9.41.165.103:55206
Apr 05 19:26:04 w54 systemd[1]: Created slice system-obmc\x2dflash\x2dbmc\x2dubiro\x2dremove.slice.
Apr 05 19:26:04 w54 systemd[1]: Starting Deletes read-only and kernel ubi volume c6c2106d...
Apr 05 19:26:05 w54 systemd[1]: Starting Set U-Boot environment variable...
Apr 05 19:26:05 w54 systemd[1]: Created slice system-obmc\x2dflash\x2dbmc\x2dupdateubootvars.slice.
Apr 05 19:26:05 w54 systemd[1]: Starting Updates the u-boot variable to point BMC version to 2da457c6...
Apr 05 19:26:07 w54 kernel: block ubiblock0_0: released
Apr 05 19:26:07 w54 kernel[974]: [ 7363.728738] block ubiblock0_0: released
Apr 05 19:26:07 w54 systemd[1]: Started Set U-Boot environment variable.
Apr 05 19:26:15 w54 systemd[1]: Started Updates the u-boot variable to point BMC version to 2da457c6.
Apr 05 19:26:15 w54 phosphor-image-updater[1216]: BMC image activating - BMC reboots are disabled.
Apr 05 19:26:15 w54 systemd[1]: Starting Enable a guard that blocks BMC reboot...
Apr 05 19:26:15 w54 systemd[1]: Starting Create BMC read-write ubi volume...
Apr 05 19:26:16 w54 systemd[1]: Created slice system-obmc\x2dflash\x2dbmc\x2dubiro.slice.
Apr 05 19:26:16 w54 systemd[1]: Starting Store read-only images 18dce02b to BMC storage...
Apr 05 19:26:16 w54 systemd[1]: Started Enable a guard that blocks BMC reboot.
Apr 05 19:26:17 w54 systemd[1]: Started Create BMC read-write ubi volume.
Apr 05 19:26:18 w54 systemd[1]: Started SSH Per-Connection Server (9.41.165.103:55226).
Apr 05 19:26:18 w54 dropbear[1699]: Child connection from ::ffff:9.41.165.103:55226
Apr 05 19:26:20 w54 dropbear[1699]: PAM password auth succeeded for 'root' from ::ffff:9.41.165.103:55226
Apr 05 19:26:20 w54 systemd[1]: Started Deletes read-only and kernel ubi volume c6c2106d.
Apr 05 19:27:37 w54 obmc-flash-bmc[1725]: Volume ID 0, size 266 LEBs (17398528 bytes, 16.6 MiB), LEB size 65408 bytes (63.9 KiB), static, name "rofs-18dce02b", alignment 1
Apr 05 19:28:18 w54 kernel: block ubiblock0_0: created from ubi0:0(rofs-18dce02b)
Apr 05 19:28:18 w54 kernel[974]: [ 7494.623719] block ubiblock0_0: created from ubi0:0(rofs-18dce02b)
Apr 05 19:28:30 w54 obmc-flash-bmc[1775]: Volume ID 1, size 37 LEBs (2420096 bytes, 2.3 MiB), LEB size 65408 bytes (63.9 KiB), static, name "kernel-18dce02b", alignment 1
Apr 05 19:28:38 w54 obmc-flash-bmc[1775]: Volume ID 3, size 37 LEBs (2420096 bytes, 2.3 MiB), LEB size 65408 bytes (63.9 KiB), static, name "kernel-18dce02b", alignment 1
Apr 05 19:28:48 w54 obmc-flash-bmc[1862]: [130B blob data]
Apr 05 19:28:49 w54 obmc-flash-bmc[1862]: [1.9K blob data]
Apr 05 19:28:49 w54 obmc-flash-bmc[1862]: [2.0K blob data]
Apr 05 19:28:49 w54 obmc-flash-bmc[1862]: [149B blob data]
Apr 05 19:28:49 w54 systemd[1]: Started Store read-only images 18dce02b to BMC storage.
Apr 05 19:31:44 w54 dbus[988]: [system] Activating via systemd: service name='org.freedesktop.timedate1' unit='dbus-org.freedesktop.timedate1.service'
Apr 05 19:31:44 w54 systemd[1]: Starting Time & Date Service...
Apr 05 19:31:44 w54 dbus[988]: [system] Successfully activated service 'org.freedesktop.timedate1'
Apr 05 19:31:44 w54 systemd-timedated[1883]: /etc/localtime should be a symbolic link to a time zone data file in /usr/share/zoneinfo/.
Apr 05 19:31:44 w54 systemd[1]: Started Time & Date Service.
Apr 05 19:32:26 w54 systemd[1]: Started SSH Per-Connection Server (9.41.165.103:55352).
Apr 05 19:32:26 w54 dropbear[1899]: Child connection from ::ffff:9.41.165.103:55352
Apr 05 19:32:26 w54 dropbear[1899]: PAM password auth succeeded for 'root' from ::ffff:9.41.165.103:55352
Apr 05 19:32:26 w54 dropbear[1620]: Exit (root): Error reading: Connection reset by peer
Apr 05 19:32:26 w54 dropbear[1615]: Exit (root): Error reading: Connection reset by peer
```

geissonator, 2018-04-10:
For now we've reverted the change.  We'll keep this open to get to root cause.

gtmills, 2018-05-30:
@geissonator This got fixed correct? Can we close?

gtmills, 2018-08-20:
@anoo1 @geissonator Can we close?

anoo1, 2018-08-27:
I guess we can, we reverted the change to try to wait for systemd services, so if that's an issue like when a bmc is rebooted before the priorities are updated, we can revisit a solution for it and open a new issue.

closed 2018-08-27 by anoo1