openbmc/openbmc #3070: [witherspoon]: kernel panic and UBI error: cannot open mtd obmc-ubi, error -2

ouyangxx10, 2018-04-08:
[    1.346599] ip6_tables: (C) 2000-2006 Netfilter Core Team
[    1.352540] sit: IPv6, IPv4 and MPLS over IPv4 tunneling driver
[    1.359974] NET: Registered protocol family 17
[    1.364482] 8021q: 802.1Q VLAN Support v1.8
[    1.378858] UBI error: cannot open mtd obmc-ubi, error -2
[    1.378916] UBI error: cannot open mtd alt-obmc-ubi, error -2
[    1.384339] UBI: block: can't open volume on ubi0_1, err=-19
[    1.390158] console [netcon0] enabled
[    1.399474] netconsole: network logging started
[    1.405573] input: gpio-keys as /devices/platform/gpio-keys/input/input2
[    1.413978] hctosys: unable to open rtc device (rtc0)
[    1.422820] VFS: Cannot open root device "ubiblock0_1" or unknown-block(0,0): error -6
[    1.430750] Please append a correct "root=" boot option; here are the available partitions:
[    1.439107] 0100            4096 ram0
[    1.439115]  (driver?)
[    1.445381] 0101            4096 ram1
[    1.445392]  (driver?)
[    1.451502] 0102            4096 ram2
[    1.451507]  (driver?)
[    1.457609] 0103            4096 ram3
[    1.457614]  (driver?)
[    1.463712] 0104            4096 ram4
[    1.463715]  (driver?)
[    1.469821] 0105            4096 ram5
[    1.469826]  (driver?)
[    1.475931] 0106            4096 ram6
[    1.475935]  (driver?)
[    1.482030] 0107            4096 ram7
[    1.482034]  (driver?)
[    1.488133] 0108            4096 ram8
[    1.488136]  (driver?)
[    1.494234] 0109            4096 ram9
[    1.494237]  (driver?)
[    1.500336] 010a            4096 ram10
[    1.500340]  (driver?)
[    1.506524] 010b            4096 ram11
[    1.506527]  (driver?)
[    1.512711] 010c            4096 ram12
[    1.512716]  (driver?)
[    1.519018] 010d            4096 ram13
[    1.519028]  (driver?)
[    1.525221] 010e            4096 ram14
[    1.525226]  (driver?)
[    1.531414] 010f            4096 ram15
[    1.531418]  (driver?)
[    1.537608] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
[    1.545873] CPU: 0 PID: 1 Comm: swapper Not tainted 4.13.16-158c8e15d577cebdb047f5d553e12054b15b1b84 #1
[    1.555259] Hardware name: Generic DT based system
[    1.560101] [<80109930>] (unwind_backtrace) from [<80106bc4>] (show_stack+0x20/0x24)
[    1.567878] [<80106bc4>] (show_stack) from [<80581c04>] (dump_stack+0x20/0x28)
[    1.575123] [<80581c04>] (dump_stack) from [<80116120>] (panic+0xe0/0x284)
[    1.582023] [<80116120>] (panic) from [<808016e4>] (mount_block_root+0x234/0x338)
[    1.589522] [<808016e4>] (mount_block_root) from [<80801978>] (mount_root+0x68/0x84)
[    1.597274] [<80801978>] (mount_root) from [<80801b04>] (prepare_namespace+0x170/0x1d4)
[    1.605280] [<80801b04>] (prepare_namespace) from [<80801234>] (kernel_init_freeable+0x188/0x1d0)
[    1.614168] [<80801234>] (kernel_init_freeable) from [<80596fc8>] (kernel_init+0x18/0x120)
[    1.622449] [<80596fc8>] (kernel_init) from [<801026f8>] (ret_from_fork+0x14/0x3c)

ast# env print
baudrate=115200
bootargs=console=ttyS4,115200n8 ubi.mtd=obmc-ubi,0,0,0 ubi.mtd=alt-obmc-ubi,0,0,4 ro rootfstype=squashfs
bootcmd=run set_bootargs; run obmc_bootcmd
bootdelay=2
do_rwreset=if test "${rwreset}" = "true"; then ubi remove rwfs; ubi create rwfs ${rwfs_size}; fi
ethact=FTGMAC100#0
kernelname=kernel-77a2fa21
loadaddr=80800000
mtdids=nor0=bmc
mtdparts=mtdparts=bmc:384k(u-boot),128k(u-boot-env),-(obmc-ubi)
obmc_bootcmd=ubi part obmc-ubi; run do_rwreset; ubi read ${loadaddr} ${kernelname}; bootm ${loadaddr}
root=/dev/ubiblock0_1
rwfs_size=0x600000
set_bootargs=setenv bootargs console=ttyS4,115200n8 ubi.mtd=obmc-ubi,0,0,0 ubi.mtd=alt-obmc-ubi,0,0,4 ro rootfstype=squashfs ubi.block=${ubiblock} root=${root}
spi_dma=yes
stderr=serial
stdin=serial
stdout=serial
ubiblock=0,1
verify=yes

Environment size: 844/65531 bytes
ast#

amboar, 2018-04-09:
Hello @ouyangxx10, thanks for filing the bug.

However, there's not enough context here to understand the source of the problem. It looks like you are missing mtd device support in the kernel, but I have no idea of the state of your source tree or how you've chosen to configure the build for Witherspoon. Can you please provide a link to the exact source tree you're trying to build and any commands you used to build the image and write it to flash on your Witherspoon?

ouyangxx10, 2018-04-09:
the newest commit id of my openbmc folder is : 
commit 65b556c79c4af44c69f3fb278e63515a5346a3bd
Author: Andrew Geissler <geissonator@yahoo.com>
Date:   Fri Apr 6 18:30:09 2018 +0000

    phosphor-software-manager.bb: bump version

    Change-Id: Ia3ee6911e600c60fba86c49a40e17f005dee623d
    Signed-off-by: Andrew Geissler <geissonator@yahoo.com>

my build env is: 
Build Configuration:
BB_VERSION           = "1.36.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-16.04"
TARGET_SYS           = "arm-openbmc-linux-gnueabi"
MACHINE              = "witherspoon"
DISTRO               = "openbmc-phosphor"
DISTRO_VERSION       = "0.1.0"
TUNE_FEATURES        = "arm armv6 thumb arm1176jzs"
TARGET_FPU           = "soft"
meta
meta-poky
meta-oe
meta-networking
meta-perl
meta-python
meta-virtualization
meta-phosphor
meta-aspeed
meta-ast2500
meta-openpower
meta-ibm
meta-witherspoon
workspace            = "master:65b556c79c4af44c69f3fb278e63515a5346a3bd"

The compile step is performed according to the readme:
git clone git@github.com:openbmc/openbmc.git
cd openbmc
export TEMPLATECONF=meta-openbmc-machines/meta-openpower/meta-ibm/meta-witherspoon/conf
. openbmc-env
bitbake obmc-phosphor-image

When compiled successfully, use DediProg SF100 flash the image to 32MB BMC flash chip.(the image name is : obmc-phosphor-image-witherspoon.ubi.mtd)

I haven't made any changes to source code.

geissonator, 2018-04-09:
That ubiblock env variable looks off.  Try running this from the uboot prompt:

```
setenv ubiblock 0,0
setenv root /dev/ubiblock0_0
saveenv
boot
```

ouyangxx10, 2018-04-09:
No effect.

ast# setenv ubiblock 0,0
ast# setenv root /dev/ubiblock0_0
ast# saveenv
Saving Environment to Flash...
Un-Protected 1 sectors
Un-Protected 1 sectors
Erasing Flash...
. done
Erased 1 sectors
Writing to Flash... done
Protected 1 sectors
Protected 1 sectors
ast# boot
ubi0: attaching mtd1
ubi0: scanning is finished
ubi0: attached mtd1 (name "mtd=2", size 31 MiB)
ubi0: PEB size: 65536 bytes (64 KiB), LEB size: 65408 bytes
ubi0: min./max. I/O unit sizes: 1/1, sub-page size 1
ubi0: VID header offset: 64 (aligned 64), data offset: 128
ubi0: good PEBs: 504, bad PEBs: 0, corrupted PEBs: 0
ubi0: user volume: 3, internal volumes: 1, max. volumes count: 128
ubi0: max/mean erase counter: 1/0, WL threshold: 4096, image sequence number: 1434730415
ubi0: available PEBs: 100, total reserved PEBs: 404, PEBs reserved for bad PEB handling: 0
Read 0 bytes from volume kernel-77a2fa21 to 80800000
No size specified -> Using max size (2388192)
## Loading kernel from FIT Image at 80800000 ...
   Using 'conf@aspeed-bmc-opp-witherspoon.dtb' configuration
   Trying 'kernel@1' kernel subimage
     Description:  Linux kernel
     Type:         Kernel Image
     Compression:  uncompressed
     Data Start:   0x8080011c
     Data Size:    2352096 Bytes = 2.2 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: 0x80001000
     Entry Point:  0x80001000
     Hash algo:    sha1
     Hash value:   1fbe55737662c45f363a0f8216a87f3df39caff4
   Verifying Hash Integrity ... sha1+ OK
## Loading fdt from FIT Image at 80800000 ...
   Using 'conf@aspeed-bmc-opp-witherspoon.dtb' configuration
   Trying 'fdt@aspeed-bmc-opp-witherspoon.dtb' fdt subimage
     Description:  Flattened Device Tree blob
     Type:         Flat Device Tree
     Compression:  uncompressed
     Data Start:   0x80a3e60c
     Data Size:    34099 Bytes = 33.3 KiB
     Architecture: ARM
     Hash algo:    sha1
     Hash value:   6f122afe5c31655bf72aec062a8ecf3ceebcbc89
   Verifying Hash Integrity ... sha1+ OK
   Booting using the fdt blob at 0x80a3e60c
   Loading Kernel Image ... OK
   Loading Device Tree to 9eb7d000, end 9eb88532 ... OK

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] random: get_random_bytes called from start_kernel+0x3c/0x3e4 with crng_init=0
[    0.000000] Linux version 4.13.16-158c8e15d577cebdb047f5d553e12054b15b1b84 (oe-user@oe-host)                                                                         (gcc version 7.3.0 (GCC)) #1 Fri Apr 6 23:14:14 EDT 2018
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt: Machine model: Witherspoon BMC
[    0.000000] Memory policy: Data cache writeback
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 109600
[    0.000000] Kernel command line: console=ttyS4,115200n8 ubi.mtd=obmc-ubi,0,0,0 ubi.mtd=alt-ob                                                                        mc-ubi,0,0,4 ro rootfstype=squashfs ubi.block=0,0 root=/dev/ubiblock0_0
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[... 147 lines cut ...]
[    1.496557]  (driver?)
[    1.502654] 010a            4096 ram10
[    1.502658]  (driver?)
[    1.508843] 010b            4096 ram11
[    1.508846]  (driver?)
[    1.515031] 010c            4096 ram12
[    1.515034]  (driver?)
[    1.521218] 010d            4096 ram13
[    1.521221]  (driver?)
[    1.527406] 010e            4096 ram14
[    1.527409]  (driver?)
[    1.533713] 010f            4096 ram15
[    1.533723]  (driver?)
[    1.539921] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
[    1.548186] CPU: 0 PID: 1 Comm: swapper Not tainted 4.13.16-158c8e15d577cebdb047f5d553e12054b                                                                        15b1b84 #1
[    1.557568] Hardware name: Generic DT based system
[    1.562411] [<80109930>] (unwind_backtrace) from [<80106bc4>] (show_stack+0x20/0x24)
[    1.570178] [<80106bc4>] (show_stack) from [<80581c04>] (dump_stack+0x20/0x28)
[    1.577416] [<80581c04>] (dump_stack) from [<80116120>] (panic+0xe0/0x284)
[    1.584316] [<80116120>] (panic) from [<808016e4>] (mount_block_root+0x234/0x338)
[    1.591814] [<808016e4>] (mount_block_root) from [<80801978>] (mount_root+0x68/0x84)
[    1.599568] [<80801978>] (mount_root) from [<80801b04>] (prepare_namespace+0x170/0x1d4)
[    1.607577] [<80801b04>] (prepare_namespace) from [<80801234>] (kernel_init_freeable+0x188/0x                                                                        1d0)
[    1.616463] [<80801234>] (kernel_init_freeable) from [<80596fc8>] (kernel_init+0x18/0x120)
[    1.624743] [<80596fc8>] (kernel_init) from [<801026f8>] (ret_from_fork+0x14/0x3c)

geissonator, 2018-04-09:
Maybe the other bank?
```
setenv ubiblock 4,0
setenv root /dev/ubiblock4_0
saveenv
boot
```

amboar, 2018-04-09:
Looks like the flash chip is toast? Unrecognised JEDEC ID:

```
[ 0.731689] aspeed-lpc-ctrl 1e789080.lpc-ctrl: Loaded at [mem 0x98000000-0x9bffffff flags 0x2 00]
[ 0.745966] aspeed-smc 1e620000.flash-controller: w25q256 (32768 Kbytes)
[ 0.752704] aspeed-smc 1e620000.flash-controller: CE0 window [ 0x20000000 - 0x22000000 ] 32MB
[ 0.761236] aspeed-smc 1e620000.flash-controller: CE1 window [ 0x22000000 - 0x2a000000 ] 128M B
[ 0.769846] aspeed-smc 1e620000.flash-controller: read control register: 203b0641
[ 0.777354] 3 ofpart partitions found on MTD device bmc
[ 0.786757] Creating 3 MTD partitions on "bmc":
[ 0.791326] 0x000000000000-0x000000060000 : "u-boot"
[ 0.800271] 0x000000060000-0x000000080000 : "u-boot-env"
[ 0.809621] 0x000000080000-0x000002000000 : "obmc-ubi"
[ 0.818921] aspeed-smc 1e620000.flash-controller: unrecognized JEDEC id bytes: 00, 00, 00
[ 0.843380] aspeed-smc 1e620000.flash-controller: Aspeed SMC probe failed -2
[ 0.855994] aspeed-smc: probe of 1e620000.flash-controller failed with error -2
[ 0.864955] aspeed-smc 1e630000.flash-controller: unrecognized JEDEC id bytes: 00, 00, 00
[ 0.873161] aspeed-smc 1e630000.flash-controller: Aspeed SMC probe failed -2
[ 0.882525] aspeed-smc: probe of 1e630000.flash-controller failed with error -2
```

amboar, 2018-04-09:
@ouyangxx10 by the way, you can format your Github comments using Markdown for readability. Specifically, you can use the triple back-tick markdown when pasting code or console output into github issues. It prevents github's rendering engine interpreting your log output as markdown. What you write looks like:

    ```
    My code, console or log output here
    ```

A guide to using github-flavoured markdown is here:

https://guides.github.com/features/mastering-markdown/

ouyangxx10, 2018-04-09:
Thanks to amboar for his help and reminder. 
I think the "Unrecognised JEDEC ID" problem maybe not the root cause of "UBI error: cannot open mtd obmc-ubi, error -2", I used to use the old version of openbmc, and it also report "Unrecognised JEDEC ID" problem too, but it can be started normally.(I have forget the version of the openbmc, I only remember the approximate date of April 2017)

```
loop: module loaded
aspeed-smc 1e620000.flash-controller: Using IRQ 17
aspeed-smc 1e620000.flash-controller: DMA support deactivated.
aspeed-smc 1e620000.flash-controller: control register changed to: 00000640
platform 1e620000.flash-controller:flash@0: mx25l25635e (32768 Kbytes)
5 ofpart partitions found on MTD device bmc
Creating 5 MTD partitions on "bmc":
0x000000000000-0x000000060000 : "u-boot"
0x000000060000-0x000000080000 : "u-boot-env"
0x000000080000-0x0000004c0000 : "kernel"
0x0000004c0000-0x000001c00000 : "rofs"
0x000001c00000-0x000002000000 : "rwfs"
aspeed-smc 1e630000.flash-controller: no DMA support.
platform 1e630000.flash-controller:flash@0: unrecognized JEDEC id bytes: 00, 00, 00
aspeed-smc 1e631000.flash-controller: no DMA support.
ftgmac100 1e660000.ethernet: Generated random MAC address ca:94:a8:1a:2f:79
libphy: ftgmac100_mdio: probed
ftgmac100 1e660000.ethernet eth0: irq 18, mapped at f0060000
ftgmac100 1e680000.ethernet: Generated random MAC address 66:3e:2f:d3:46:f7
libphy: ftgmac100_mdio: probed
ftgmac100 1e680000.ethernet eth1: irq 19, mapped at f0080000
i2c /dev entries driver
i2c_aspeed 1e78a000.i2c: i2c controller registered, irq 23
at24 3-0050: 1024 byte 24c08 EEPROM, writable, 16 bytes/write
i2c_aspeed i2c-3: i2c bus 3 registered, irq 259
i2c_aspeed i2c-7: i2c bus 7 registered, irq 263
```

amboar, 2018-04-09:
Right, yes, that was a bit lax of me. I assumed that the JEDEC ID was the cause of the following output:

```
[ 0.843380] aspeed-smc 1e620000.flash-controller: Aspeed SMC probe failed -2
[ 0.855994] aspeed-smc: probe of 1e620000.flash-controller failed with error -2
```
But maybe it wasn't.

The flash controller driver has failed to probe, therefore you won't get MTD devices exposed in `/dev/`. Maybe there's another cause, though I don't know off-hand what it would be. Oddly:
```
$ errno 2
ENOENT 2 No such file or directory
```
The aspeed-smc driver doesn't return ENOENT itself, so it's just passing back the error from something that it has called into.

mdmillerii, 2018-04-09:
It appears @ouyangxx10 only has the primary `bmc` chip installed, hence the 0 ids on the `pnor` and `alt` spi-nor chips.
```
[ 0.745966] aspeed-smc 1e620000.flash-controller: w25q256 (32768 Kbytes)
[ 0.752704] aspeed-smc 1e620000.flash-controller: CE0 window [ 0x20000000 - 0x22000000 ] 32MB
[ 0.761236] aspeed-smc 1e620000.flash-controller: CE1 window [ 0x22000000 - 0x2a000000 ] 128M B
[ 0.769846] aspeed-smc 1e620000.flash-controller: read control register: 203b0641
[ 0.777354] 3 ofpart partitions found on MTD device bmc
[ 0.786757] Creating 3 MTD partitions on "bmc":
[ 0.791326] 0x000000000000-0x000000060000 : "u-boot"
[ 0.800271] 0x000000060000-0x000000080000 : "u-boot-env"
[ 0.809621] 0x000000080000-0x000002000000 : "obmc-ubi"
[ 0.818921] aspeed-smc 1e620000.flash-controller: unrecognized JEDEC id bytes: 00, 00, 00
[ 0.843380] aspeed-smc 1e620000.flash-controller: Aspeed SMC probe failed -2
[ 0.855994] aspeed-smc: probe of 1e620000.flash-controller failed with error -2
[ 0.864955] aspeed-smc 1e630000.flash-controller: unrecognized JEDEC id bytes: 00, 00, 00
[ 0.873161] aspeed-smc 1e630000.flash-controller: Aspeed SMC probe failed -2
[ 0.882525] aspeed-smc: probe of 1e630000.flash-controller failed with error -2
```
(@legoater it appears the per-device prints above are missing the affected chip select or flash chip instance and instead print only the `flash-controller` device these days.)

The prior release was April 2007 which was before `ubi` layout, so his environment was not updated, however this was an external programmer and hence the original environment should have had everything needed.

@ouyangxx10 you can look for the `ubi` volume number by dumping the `ubi` volume information from `u-boot`; 
```
ubi part obmc-ubi
ubi info
```
Find the volume number of the file system image (it will end with the same hex number as the kerne in the `kernelname` variable, which was `kernel-77a2fa21` above.  Replace `<N>` in the commands below with the volume number from above:
```
setenv ubiblock 0,<N>
setenv root /dev/ubiblock0_<N>
saveenv
boot
```

ouyangxx10, 2018-04-10:
@mdmillerii  the detail info as follows:
```
ast# ubi part obmc-ubi
ubi0: attaching mtd1
ubi0: scanning is finished
ubi0: attached mtd1 (name "mtd=2", size 31 MiB)
ubi0: PEB size: 65536 bytes (64 KiB), LEB size: 65408 bytes
ubi0: min./max. I/O unit sizes: 1/1, sub-page size 1
ubi0: VID header offset: 64 (aligned 64), data offset: 128
ubi0: good PEBs: 504, bad PEBs: 0, corrupted PEBs: 0
ubi0: user volume: 3, internal volumes: 1, max. volumes count: 128
ubi0: max/mean erase counter: 1/0, WL threshold: 4096, image sequence number: 218684547
ubi0: available PEBs: 100, total reserved PEBs: 404, PEBs reserved for bad PEB handling: 0
ast# ubi info
UBI: MTD device name:            "mtd=2"
UBI: MTD device size:            31 MiB
UBI: physical eraseblock size:   65536 bytes (64 KiB)
UBI: logical eraseblock size:    65408 bytes
UBI: number of good PEBs:        504
UBI: number of bad PEBs:         0
UBI: smallest flash I/O unit:    1
UBI: VID header offset:          64 (aligned 64)
UBI: data offset:                128
UBI: max. allowed volumes:       128
UBI: wear-leveling threshold:    4096
UBI: number of internal volumes: 1
UBI: number of user volumes:     3
UBI: available PEBs:             100
UBI: total number of reserved PEBs: 404
UBI: number of PEBs reserved for bad PEB handling: 0
UBI: max/mean erase counter: 1/0
ast#
```

legoater, 2018-04-10:
@mdmillerii  yes. the error message comes from the spi_nor_scan() call which only prints out the controller device.

mdmillerii, 2018-04-10:
@ouyangxx10 there must be another `ubi` subcommand that will list the volume details not just the volume summary.  Sorry I don't have a box to experiment with, but I see there are 3 user volumes.

closed 2019-04-27 by stale[bot]