openbmc/openbmc #3071: what is the default username and password to login terminal

FrankHome, 2018-04-09:
Hi, 
    what is the default username and password to login terminal
Thanks a lot
[image]

FrankHome, 2018-04-09:
Thank you very much~~

athkar123, 2020-01-10:
What is the password?

brianmeritech, 2024-08-29:
> root / 0penBmc Where the 0 is a numeric zero.

I tried. but it doesn't work on mine 
E/TC:0 0 plat_rng_init:188 Warning: seeding RNG with zeroes
[    0.000000] Booting Linux on physical CPU 0xf00
[    0.000000] Linux version 6.1.61-de59419-dirty-1878a34 (oe-user@oe-host) (arm-openbmc-linux-gnueabi-gcc (GCC) 13.1.1 20230520, GNU ld (GNU Binutils) 2.40.0.20230419) #1 SMP Thu Jan 25 17:01:45 UTC 2024

brianmeritech, 2024-08-29:
[  146.261698] i2c i2c-5: new_device: Instantiated device raa229126 at 0x61
[  146.269744] i2c i2c-5: delete_device: Deleting device raa229126 at 0x61

Password:

Login incorrect
bmc-maca4bf018d1bd4 login: root
Password:

Login incorrect
bmc-maca4bf018d1bd4 login:
Login timed out after 60 seconds.

Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) nodistro.0 bmc-maca4bf018d1bd4 ttyS4

bmc-maca4bf018d1bd4 login: root
Password:

Login incorrect
bmc-maca4bf018d1bd4 login:
Login timed out after 60 seconds.

Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) nodistro.0 bmc-maca4bf018d1bd4 ttyS4

bmc-maca4bf018d1bd4 login:

brianmeritech, 2024-08-29:
> [ 146.261698] i2c i2c-5: new_device: Instantiated device raa229126 at 0x61 [ 146.269744] i2c i2c-5: delete_device: Deleting device raa229126 at 0x61
>
> Password:
>
> Login incorrect bmc-maca4bf018d1bd4 [...]

still can't login with root/0penBmc

williamspatrick, 2024-08-29:
kernel 6.1.1 doesn't seem like the latest code.  What OpenBMC commit and machine are you targeting?

brianmeritech, 2024-09-02:
> kernel 6.1.1 doesn't seem like the latest code. What OpenBMC commit and machine are you targeting?

Dear Mr. William, 
My test is on Intel OpenBmc of avenue city platform. 
This is my binary :OBMC-bhs-24.09-0-gdb04e9-6fddef2-pfr-full

Welcome to Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) bhs-24.09!

brianmeritech, 2024-09-10:
> > This is my binary :OBMC-bhs-24.09-0-gdb04e9-6fddef2-pfr-full
> 
> I don't recognize any of this tag information, but [6fddef2](https://github.com/openbmc/openbmc/commit/6fddef299932b1270a799e78566e25daa911f742) is a commit from this repository from August 2023. Can you update to something newer and retest?

Yes, Mr. William, 
I'll try it and talk Intel via IPS, as well. 

Merci
Brian