openbmc/openbmc #3073: obmc-phosphor-image-1.0-r0: Build Error

rpking42, 2018-04-09:
I'm getting the following error when trying to build a config with
`TEMPLATECONF=meta-openbmc-machines/meta-evb/meta-evb-aspeed/meta-evb-ast2500/conf`

```
ERROR: obmc-phosphor-image-1.0-r0 do_generate_static: Function failed: do_generate_static
...
dd: failed to open '/home/rpking42/proj/openbmc/build/tmp/deploy/images/evb-ast2500/u-boot.bin': No such file or directory
```

I've attached the fail log as well (renamed to a .txt for Github to accept it).
[log.do_generate_static.4640.txt](https://github.com/openbmc/openbmc/files/1890459/log.do_generate_static.4640.txt)

I then reran bitbake with the `-k` flag to allow all the targets except this one to finish, and then rerunning again (where only this one target was left) the build succeeded, so my guess is that there is some sort of dependency in the config not being handled.

This issue appeared at some point over the weekend. The error is on 65b556c7, but I ran early Friday and it worked (sorry I don't have an exact commit).

amboar, 2018-04-13:
@rpking42 any change in this issue? Is the problem still occurring for you?

rpking42, 2018-04-13:
I pulled an reran in a fresh directory. I'm actually seeing this now:

```
ERROR: u-boot-1_v2016.07+gitAUTOINC+e4c85cd4de-r0 do_patch: Command Error: 'quilt --quiltrc /home/rpking42/proj/openbmc/build/tmp/work/evb_ast2500-openbmc-linux-gnueabi/u-boot/1_v2016.07+gitAUTOINC+e4c85cd4de-r0/recipe-sysroot-native/etc/quiltrc push' exited with 0  Output:
/bin/sh: quilt: command not found
ERROR: u-boot-1_v2016.07+gitAUTOINC+e4c85cd4de-r0 do_patch: Function failed: patch_do_patch
```

Checking from the command line, I do have quilt installed. Although I'm sure if it means my host system doesn't have the command, or the system I'm building doesn't. Or possibly I'm overlooking something simple.

nest1ing, 2018-04-13:
We have this error too.
I push fix to [gerrit](https://gerrit.openbmc-project.xyz/#/c/10117/)

rpking42, 2018-04-13:
I applied the fix from gerrit and I can complete with no errors. Thanks.

rfrandse, 2018-04-13:
<https://gerrit.openbmc-project.xyz/10117> u-boot-aspeed: fix building
Resolves:  #3073 obmc-phosphor-image-1.0-r0: Build Error

akashgj, 2018-05-10:
The same problem occurs again when I tried to build the system using the latest checkout https://github.com/openbmc/openbmc/commit/b4b1b1c54b026cf6932cfe2e5e2486fc0342545b.

ERROR: obmc-phosphor-image-1.0-r0 do_generate_static: Function failed: do_generate_static (log file is located at /home/akash/Music/BMC/openbmc/build/tmp/work/zaius-openbmc-linux-gnueabi/obmc-phosphor-image/1.0-r0/temp/log.do_generate_static.13152) 

Please someone help to resolve this.

amboar, 2018-05-11:
@akashgj maybe try a `bitbake -c cleanall u-boot && bitbake obmc-phosphor-image`

akashgj, 2018-05-11:
@amboar  I cleaned u-boot and tried to build again. Still it is showing same error.

amboar, 2018-05-11:
@akashgj can you please attach the log file?

akashgj, 2018-05-11:
Please find attached log file.
[log.txt](https://github.com/openbmc/openbmc/files/1993997/log.txt)

amboar, 2018-05-11:
That log file wasn't too helpful in the end. Maybe @bradbishop has ideas. You could always try hitting it with a hammer and run `git clean -dxf` in the root of your openbmc clone to start from scratch.

lakesh-sharma, 2018-06-19:
i had pulled work-space yesterday (June 18) and tried to build evb-ast2500 , but got same u-boot err as listed in this issue.
Then i tried following changes, which are actually same as listed in one of patch above. Re-build the image and it went ok. So do we need this fix ?

diff --git a/meta-phosphor/aspeed-layer/recipes-bsp/u-boot/u-boot-aspeed.inc b/meta-phosphor/aspeed-layer/recipes-bsp/u-boot/u-boot-aspeed.inc
index 5286c1b0..7e943709 100644
--- a/meta-phosphor/aspeed-layer/recipes-bsp/u-boot/u-boot-aspeed.inc
+++ b/meta-phosphor/aspeed-layer/recipes-bsp/u-boot/u-boot-aspeed.inc
@@ -9,7 +9,7 @@ SRC_URI_append_aspeed_df-obmc-ubi-fs = " \
     file://0005-config-ast-common-Fall-back-to-secondary-flash-on-fa.patch \
     "
 
-do_patch[depends] += "os-release:do_populate_sysroot"
+do_patch[depends] += " os-release:do_populate_sysroot"
 
 python do_patch_append_aspeed_df-obmc-ubi-fs () {
     version_id=do_get_versionID(d)

amboar, 2018-06-20:
@lakesh-sharma can you confirm what commit hash you have checked out? This has been fixed since April with https://github.com/openbmc/openbmc/commit/32a2f0b44ebfc53f8a855f22b5879841da5bdd51 .

lakesh-sharma, 2018-06-20:
i cloned about 2 days back with following steps.
Seems issue is still seen from May 10 onwards, as akashj re-opened this issue on May 10.

git clone https://github.com/openbmc/openbmc.git
cd openbmc/
export TEMPLATECONF=meta-openbmc-machines/meta-evb/meta-evb-aspeed/meta-evb-ast2500/conf

mdmillerii, 2018-06-20:
@amboar note that merge was a recreated fix, and appears to miss the initial space. Maybe the dependency is being merged with the prior string?

amboar, 2018-06-25:
@mdmillerii good catch, thanks. Apologies to @lakesh-sharma and everyone else! I've pushed a change:

https://gerrit.openbmc-project.xyz/11260

Please test and score :)

mdmillerii, 2018-08-01:
@bradbishop rejected and @amboar abandoned the above proposed fix, but we still have observations that the products are not available in time.

> This is not the cause of whatever issue we were chasing.
> 
> 1 - grep -R \[depends\] from oe-core shows the pre-change syntax
> 2 - I did a build and I can see quilt-native:do_populate_sysroot prior to os-release:do_populate_sysroot, with a space
> 3 - The += operator adds a space for you

Further analysis of a failed state will be required.

closed 2018-08-06 by bradbishop

awxhw, 2018-09-07:
@amboar  I still met the same problem with commit 3e4da38c127bb7e7641adc2fc41f4c33744cb918, and the change you pushed is not merged. But I applied it by hand, the error still came, 

ERROR: obmc-phosphor-image-1.0-r0 do_generate_static: Error executing a python function in exec_python_func() autogenerated:

The stack trace of python calls that resulted in this exception/failure was:
File: 'exec_python_func() autogenerated', lineno: 2, function: <module>
     0001:
 *** 0002:do_generate_static(d)
     0003:
File: '/home/vagrant/source/tmp/openbmc/meta-phosphor/classes/image_types_phosphor.bbclass', lineno: 212, function: do_generate_static
     0208:
     0209:    _append_image(os.path.join(d.getVar('DEPLOY_DIR_IMAGE', True),
     0210:                               'u-boot.%s' % d.getVar('UBOOT_SUFFIX',True)),
     0211:                  int(d.getVar('FLASH_UBOOT_OFFSET', True)),
 *** 0212:                  int(d.getVar('FLASH_KERNEL_OFFSET', True)))
     0213:
     0214:    _append_image(os.path.join(d.getVar('DEPLOY_DIR_IMAGE', True),
     0215:                               d.getVar('FLASH_KERNEL_IMAGE', True)),
     0216:                  int(d.getVar('FLASH_KERNEL_OFFSET', True)),
File: '/home/vagrant/source/tmp/openbmc/meta-phosphor/classes/image_types_phosphor.bbclass', lineno: 200, function: _append_image
     0196:    nor_image = os.path.join(d.getVar('IMGDEPLOYDIR', True),
     0197:                             '%s.static.mtd' % d.getVar('IMAGE_NAME', True))
     0198:
     0199:    def _append_image(imgpath, start_kb, finish_kb):
 *** 0200:        imgsize = os.path.getsize(imgpath)
     0201:        if imgsize > (finish_kb - start_kb) * 1024:
     0202:            bb.fatal("Image '%s' is too large!" % imgpath)
     0203:
     0204:        subprocess.check_call(['dd', 'bs=1k', 'conv=notrunc',
File: '/usr/lib/python3.4/genericpath.py', lineno: 50, function: getsize
     0046:
     0047:
     0048:def getsize(filename):
     0049:    """Return the size of a file, reported by os.stat()."""
 *** 0050:    return os.stat(filename).st_size
     0051:
     0052:
     0053:def getmtime(filename):
     0054:    """Return the last modification time of a file, reported by os.stat()."""
Exception: FileNotFoundError: [Errno 2] No such file or directory: '/home/vagrant/source/tmp/openbmc/build/tmp/deploy/images/palmetto/u-boot.bin'

ERROR: obmc-phosphor-image-1.0-r0 do_generate_static: Function failed: do_generate_static
ERROR: Logfile of failure stored in: /home/vagrant/source/tmp/openbmc/build/tmp/work/palmetto-openbmc-linux-gnueabi/obmc-phosphor-image/1.0-r0/temp/log.do_generate_static.997
ERROR: Task (/home/vagrant/source/tmp/openbmc/meta-phosphor/common/recipes-phosphor/images/obmc-phosphor-image.bb:do_generate_static) failed with exit code '1'