openbmc/openbmc #3074: Config hwmon for new platform

ThaiThu, 2018-04-10:
Hi all, 

We are using openbmc for our hardware.
We created our own new layer under "/meta-openbmc-machines/meta-evb/", and after that, we config new sensor into "./meta-openbmc-machines/meta-evb/meta-mct/recipes-phosphor/sensors/phosphor-hwmon%.bbappend" as below:

`FILESEXTRAPATHS_prepend := "${THISDIR}/${PN}:"

NAMES = " i2c@1e78a000/i2c-bus@40/lm75@4a i2c@1e78a000/i2c-bus@80/lm75@4a i2c@1e78a000/i2c-bus@c0/mct_pfe850@59 i2c@1e78a000/i2c-bus@c0/mct_pfe850@58"
ITEMSFMT = "ahb/apb/{0}.conf"

ITEMS = "${@compose_list(d, 'ITEMSFMT', 'NAMES')}"

ENVS = "obmc/hwmon/{0}"
SYSTEMD_ENVIRONMENT_FILE_${PN} += "${@compose_list(d, 'ENVS', 'ITEMS')}"`

We also have conf file install into openbmc.

But when start up the image, the daemon phosphor-hwmon-readd is not started.
We noticed that the conf file were already installed into  "/etc/default/obmc/hwmon/ahb/apb/i2c@1e78a000/i2c-bus@..."

And the "start_hwmon.sh" is also started, but I don't know why the "xyz.openbmc_project.Hwmon@.service" service doesn't start the phosphor-hwmon-readd daemon.

Thank you !

mine260309, 2018-04-10:
You need to check:
1. If the devices are created in sysfs? (e.g. check if `/sys/firmware/devicetree/base/ahb/apb/i2c@1e78a000/i2c-bus@40/lm75@4a` exists or not)
2. Check the journal log to see if `phosphor-hwmon` ever started or not. E.g. if your config file is incorrect, `phosphor-hwmon` may start and exit with crash.

ThaiThu, 2018-04-10:
Dear @mine260309 

1. Yes, the device is created
[image]

2. When using journal to see log, we didn't see any info involve Hwmon

[image]

Thank you !

ThaiThu, 2018-04-10:
Hi @mine260309 

I think our "phosphor-hwmon%.bbappend"  and .conf files are correct, because we use them im evb platform, it is able to start "phosphor-hwmon-readd" and expose interface to dbus.

Thanks !

ThaiThu, 2018-04-10:
Hi @mine260309 

Sorry, I use different device tree, so the device name mismatch with the conf file, I have fix it. They now match to each other.

[image]

The .bbappend is also changed

```
FILESEXTRAPATHS_prepend := "${THISDIR}/${PN}:"

NAMES = " **i2c@1e78a000/i2c-bus@40/lm75_0@4a** i2c@1e78a000/i2c-bus@80/lm75_1@4a i2c@1e78a000/i2c-bus@c0/mct_pfe850@59 i2c@1e78a000/i2c-bus@c0/mct_pfe850@58"
ITEMSFMT = "ahb/apb/{0}.conf"

ITEMS = "${@compose_list(d, 'ITEMSFMT', 'NAMES')}"

ENVS = "obmc/hwmon/{0}"
SYSTEMD_ENVIRONMENT_FILE_${PN} += "${@compose_list(d, 'ENVS', 'ITEMS')}"
```

But the "phosphor-hwmon-readd" daemon is still not started .

Thank you

mine260309, 2018-04-10:
Then you need to debug what's wrong.
E.g. check if the udev runs correctly:
```
udevadm test /sys/class/hwmon/hwmon?
```
If it's OK, then you can manually run the command of the rule, and check the journal log to see why it fails to run.

ThaiThu, 2018-04-11:
Hi @mine260309 

Thank you.
Do you know how to start "phosphor-hwmon-readd" manually with -p option ?

My i2c device is not described in the devicetree , I instantiate it manually in userspace using 
"echo jc42 0x18 > /sys/class...".
And we only have it in "/sys/class/hwmon/hwmon10/", but we don't have it in "/sys/firmware/devicetree/base/ahb/apb/i2c\@1e78a000/"

mine260309, 2018-04-11:
Since the device is not in device tree, it will not work because the current code uses `dev-path` (`-o`) in `xyz.openbmc_project.Hwmon@.service`.

But if you really want to test the `-p` option, you need to provide environment for it, otherwise there is nothing to monitor.

So ideally you can:
```
export LABEL_temp1=dimm0
phosphor-hwmon-readd -p /sys/class/hwmon/hwmon10
```
Then you should have `/xyz/openbmc_project/sensors/temperature/dimm0` on dbus.

But please be noted that this is not suggested, please describe the device in dts, and `xyz.openbmc_project.Hwmon@.service` will be automatically started by udev rule.

spinler, 2018-04-11:
FYI, with https://gerrit.openbmc-project.xyz/#/c/9412/ which shows up along with the  4.13 kernel,  I added support to phosphor-hwmon to support the actual device paths if the device tree name isn't present.

ThaiThu, 2018-04-23:
Hi @mine260309 
do you know where is the bmc store out stream from phosphor-hwmon (std::cerr << "ERROR: " << err << std::endl;)

I didn't see them on the console terminal.

Thank you !!

ThaiThu, 2018-04-23:
Hi @spinler 

Thank you.

ThaiThu, 2018-04-23:
hi @mine260309 

I found out the problem, I forget to flush it.

Thank you.

closed 2019-04-27 by stale[bot]

Tommy-git2033, 2021-03-23:
Hi,@ThaiThu , i meet the same issue as yours, please could you tell me how you fix it ? what is "I forget to flush it." mean?

AloneWarrior11011, 2025-04-07:
> You need to check:
>
>     1. If the devices are created in sysfs? (e.g. check if `/sys/firmware/devicetree/base/ahb/apb/i2c@1e78a000/i2c-bus@40/lm75@4a` exists or not)
>
>     2. Check the journal log to [...]

I was facing the same issue but I my case device isn't created in sysfs, despite begin configs are installed into /etc/default/obmc/hwmon/i2c-1/tmp75@4d.conf 

what could be the reason : 

```
meta-evb/meta-evb-raspberrypi/recipes-kernel/linux/linux-raspberrypi/tmp75.dts file : 

 /dts-v1/;
 /plugin/;

/ {
    compatible = "brcm,bcm2835";

    fragment@0 {
            target = <&i2c1>;
            __overlay__ {
                    #address-cells = <1>;
                    #size-cells = <0>;

                    tmp75@4d {
                            status = "okay";
                            compatible = "ti,tmp75", "national,lm75";
                            reg = <0x4d>;
                    };
            };
    };

Below are some logs which'll help to debug the issue : 
**Configuration file been installed :** 

root@raspberrypi3:~# ls -la /etc/default/obmc/hwmon/i2c-1
drwxr-xr-x    2 root     root          1024 Apr  5  2011 .
drwxr-xr-x    3 root     root          1024 Apr  5  2011 ..
-rw-r--r--    1 root     root           211 Apr  5  2011 tmp75@4d.conf

**journalctl shows skipped logs :** 

root@raspberrypi3:~# journalctl | grep Hwmon
Dec 19 21:25:51 raspberrypi3 systemd[1]: Created slice Slice /system/xyz.openbmc_project.Hwmon.
Dec 19 21:25:54 raspberrypi3 systemd[1]: Phosphor Hwmon Poller was skipped because of an unmet condition check (ConditionFileNotEmpty=/etc/default/obmc/hwmon/devices/virtual/thermal/thermal_zone0/hwmon0.conf).

**Conf file been installed to the right directory :** 

root@raspberrypi3:~# ls -l /etc/default/obmc
drwxr-xr-x    3 root     root          1024 Apr  5  2011 hwmon
drwxr-xr-x    2 root     root          1024 Apr  5  2011 watchdog

**No entry here for tmp75 sensor (device tree entries) :** 

root@raspberrypi3:~# ls -la /sys/firmware/devicetree/base
-r--r--r--    1 root     root             4 Apr  4 10:53 #address-cells
-r--r--r--    1 root     root             4 Apr  4 10:53 #size-cells
drwxr-xr-x   25 root     root             0 Dec 19 21:25 .
drwxr-xr-x    3 root     root             0 Dec 19 21:25 ..
drwxr-xr-x    2 root     root             0 Apr  4 10:53 __overrides__
drwxr-xr-x    2 root     root             0 Apr  4 10:53 __symbols__
drwxr-xr-x    2 root     root             0 Apr  4 10:53 aliases
drwxr-xr-x    2 root     root             0 Apr  4 10:53 arm-pmu
drwxr-xr-x    3 root     root             0 Apr  4 10:53 axi
drwxr-xr-x    2 root     root             0 Apr  4 10:53 cam0_clk
drwxr-xr-x    2 root     root             0 Apr  4 10:53 cam0_regulator
drwxr-xr-x    2 root     root             0 Apr  4 10:53 cam1_clk
drwxr-xr-x    2 root     root             0 Apr  4 10:53 cam1_regulator
drwxr-xr-x    2 root     root             0 Apr  4 10:53 cam_dummy_reg
[... 72 lines cut ...]
Reading rules file: /usr/lib/udev/rules.d/80-net-setup-link.rules
Reading rules file: /usr/lib/udev/rules.d/80-obmc-console-uart.rules
Reading rules file: /usr/lib/udev/rules.d/81-net-dhcp.rules
Reading rules file: /etc/udev/rules.d/99-com.rules
/etc/udev/rules.d/99-com.rules:2 Unknown group 'i2c', ignoring.
/etc/udev/rules.d/99-com.rules:3 Unknown group 'spi', ignoring.
/etc/udev/rules.d/99-com.rules:4 Unknown group 'gpio', ignoring.
/etc/udev/rules.d/99-com.rules:10 Unknown group 'gpio', ignoring.
Reading rules file: /usr/lib/udev/rules.d/99-systemd.rules
Reading rules file: /etc/udev/rules.d/can.rules
Reading rules file: /etc/udev/rules.d/fb.rules
Reading rules file: /etc/udev/rules.d/touchscreen.rules
hwmon0: /usr/lib/udev/rules.d/70-hwmon.rules:1 RUN '/usr/bin/start_hwmon.sh start %p %E{OF_FULLNAME}'
Properties:
  DEVPATH=/devices/virtual/thermal/thermal_zone0/hwmon0
  ACTION=add
  SUBSYSTEM=hwmon
  USEC_INITIALIZED=307964567
Queued commands:
  RUN{program} : /usr/bin/start_hwmon.sh start /devices/virtual/thermal/thermal_zone0/hwmon0 
Unloaded link configuration context.

};

```

AloneWarrior11011, 2025-04-07:
> Hi [@mine260309](https://github.com/mine260309)
>
> Sorry, I use different device tree, so the device name mismatch with the conf file, I have fix it. They now match to each other.
>
> [image]
>
> The [...]

I am a newbie to openbmc and don't have the idea to perfectly customize the recipe that's why I am asking you to provide complete recipes and their structure inside of meta-evb/meta-evb-raspberrypi/ this will help me to resolve the problem.

And tmp75.dts as below : 

```
/dts-v1/;
/plugin/;

/ {
    compatible = "brcm,bcm2835";

    fragment@0 {
        target = <&i2c1>;
        __overlay__ {
            #address-cells = <1>;
            #size-cells = <0>;

            tmp75@4d {
                compatible = "ti,tmp75", "national, lm75" ;
                reg = <0x4d>;
                status = "okay";
            };
        };
    };

```
};

In my case the configuration file of tmp75@4d perfectly installed into /etc/default/obmc/hwmon/i2c-1/tmp75@4d.conf, but devide node not created, tmp75.dts file content looks
fine, but I think there might be some issue in linux-raspberrypi_%.bbappend recipe that don't place dts file in the right location while building openbmc image for raspberrypi. 

If I get your complete recipe structure, it would really helpful for me to figure out the issue. 

if you need complete recipe structure to get the actual issue I will post here as well. 

```
**Configuration file been installed :** 

root@raspberrypi3:~# ls -la /etc/default/obmc/hwmon/i2c-1
drwxr-xr-x    2 root     root          1024 Apr  5  2011 .
drwxr-xr-x    3 root     root          1024 Apr  5  2011 ..
-rw-r--r--    1 root     root           211 Apr  5  2011 tmp75@4d.conf
```

```
**journalctl shows skipped logs :** 

root@raspberrypi3:~# journalctl | grep Hwmon
Dec 19 21:25:51 raspberrypi3 systemd[1]: Created slice Slice /system/xyz.openbmc_project.Hwmon.
Dec 19 21:25:54 raspberrypi3 systemd[1]: Phosphor Hwmon Poller was skipped because of an unmet condition check (ConditionFileNotEmpty=/etc/default/obmc/hwmon/devices/virtual/thermal/thermal_zone0/hwmon0.conf).

```

```
**Conf file been installed to the right directory :** 

root@raspberrypi3:~# ls -l /etc/default/obmc
drwxr-xr-x    3 root     root          1024 Apr  5  2011 hwmon
drwxr-xr-x    2 root     root          1024 Apr  5  2011 watchdog
```

```
[... 94 lines cut ...]
Reading rules file: /usr/lib/udev/rules.d/80-net-setup-link.rules
Reading rules file: /usr/lib/udev/rules.d/80-obmc-console-uart.rules
Reading rules file: /usr/lib/udev/rules.d/81-net-dhcp.rules
Reading rules file: /etc/udev/rules.d/99-com.rules
/etc/udev/rules.d/99-com.rules:2 Unknown group 'i2c', ignoring.
/etc/udev/rules.d/99-com.rules:3 Unknown group 'spi', ignoring.
/etc/udev/rules.d/99-com.rules:4 Unknown group 'gpio', ignoring.
/etc/udev/rules.d/99-com.rules:10 Unknown group 'gpio', ignoring.
Reading rules file: /usr/lib/udev/rules.d/99-systemd.rules
Reading rules file: /etc/udev/rules.d/can.rules
Reading rules file: /etc/udev/rules.d/fb.rules
Reading rules file: /etc/udev/rules.d/touchscreen.rules
hwmon0: /usr/lib/udev/rules.d/70-hwmon.rules:1 RUN '/usr/bin/start_hwmon.sh start %p %E{OF_FULLNAME}'
Properties:
  DEVPATH=/devices/virtual/thermal/thermal_zone0/hwmon0
  ACTION=add
  SUBSYSTEM=hwmon
  USEC_INITIALIZED=307964567
Queued commands:
  RUN{program} : /usr/bin/start_hwmon.sh start /devices/virtual/thermal/thermal_zone0/hwmon0 
Unloaded link configuration context.

```
Do help me with that, device not created in sysfs, despite being config are installed into 
/etc/default/obmc/hwmon/i2c-1/tmp75@4d.conf

williamspatrick, 2025-04-07:
Device not showing up in sysfs is a kernel config and/or device tree problem. Phosphor-hwmon can’t fix a device that isn’t there.

AloneWarrior11011, 2025-04-07:
> Device not showing up in sysfs is a kernel config and/or device tree problem. Phosphor-hwmon can’t fix a device that isn’t there.

Then how could I fix that problem

AloneWarrior11011, 2025-04-07:
After compiling tmp75.dts file it reflecting in the following directories : 
```
/tmp/work/raspberrypi3-openbmc-linux-gnueabi/linux-raspberrypi/6.6.63+git/image/boot/overlays
./tmp/work/raspberrypi3-openbmc-linux-gnueabi/linux-raspberrypi/6.6.63+git/package/boot/overlays
./tmp/work/raspberrypi3-openbmc-linux-gnueabi/linux-raspberrypi/6.6.63+git/packages-split/kernel-devicetree/boot/overlays
``` 

But not in the final image : 
```
[user@linuxpc117 rootfs]$ ls
bin  boot  dev  etc  home  lib  media  mnt  proc  run  sbin  srv  sys  tmp  usr  var
[user@linuxpc117 rootfs]$ cd boot 
[user@linuxpc117 boot]$ ls
fitImage                                     vmlinux  zImage-6.6.63-dirty-e442e5c-ge442e5c1ab6b
fitImage-6.6.63-dirty-e442e5c-ge442e5c1ab6b  zImage
[user@linuxpc117 boot]$ pwd
/home/user/obmc-1/openbmc/build/tmp/work/raspberrypi3-openbmc-linux-gnueabi/obmc-phosphor-image/1.0/rootfs/boot
[user@linuxpc117 boot]$ 
```
/boot/overlays/tmp75.dtbo isn't there.