openbmc/openbmc #3083: Memory leak in phosphor-webui

gtmills, 2018-04-10:
Used Chrome DevTools Performance.
Steps to recreate:
Fresh browser, no other tabs open. 
Went to the Web UI running on a Witherspoon. 
After DevTools are up, click Performance, select memory, and then record.
Logged into the Web UI. 
Then for ~1 minute navigated the different tabs. I did not change network settings, change the password, boot the server, or upload an image or any other firmware actions.
Then Logged out. 
Stopped the recording. 
As the graphs below show we are leaking memory. 

The first image was v2.1-366-g26fcc80 but saw this on multiple systems/versions.  

[image]

[image]

gtmills, 2018-04-10:
This might be a better test.
You can manually force the garbage collection.. 
So start recording, force garbage collection, log in, ~1 minute navigated the different tabs, log out, force garbage collection, end recording. 

[image]

After 1st garbage collection ~19MB. After 2nd garbage collection ~24MB.

edtanous, 2018-05-02:
If the objects are attached to the scope, they should be destroyed when the scope is destroyed automatically.  I might suggest a slight change to your test procedure.

1. log in.
2. walk through all the UI windows.
3. Log out
4. force a garbage collection, and record the memory used.
Repeat steps 1-4 a few time and see if you get increasing numbers on step 4.

If the memory is actually increasing on step 4, you can dump a heap snapshot in chrome, and compare the two.  https://developers.google.com/web/tools/chrome-devtools/memory-problems/heap-snapshots

Just because we're using memory, doesn't necessarily mean that it was "leaked".  Angular keeps plenty of objects lying around and cached so it can start modules up faster a second time.

gtmills, 2018-08-15:
1. Forced garbage collection. 
2. Took heap-start snapshot and started recording. 
3. Logged in
4. Navigated to every page.
5. Logged out.
6. Forced a garbage collection. 
7. Repeated steps 3-6 three more times (four total).
8. After last time through and after garbage collection ended recording and took heap end snapshot.

[image]
Heap grew from ~19MB to ~27MB.

gtmills, 2018-08-15:
.heapsnapshot are not supported by GitHub issues. 
Here are screenshots:
[image]
[image]

closed 2019-04-26 by stale[bot]