openbmc/openbmc #3088: Build Error for meta-q71l

reju85, 2018-04-12:
I am trying to build IBM-Open BMC for quanta but getting build issues
**Steps Done:**
1.git clone git@github.com:openbmc/openbmc.git
2.cd openbmc
3.export TEMPLATECONF=meta-openbmc-machines/meta-x86/meta-quanta/meta-q71l/conf
4.. openbmc-env
5.bitbake obmc-phosphor-image

while building getting the following error.Kindly help to fix the issue
**Error Log:-**
WARNING: phosphor-ipmi-fru-1.0-r1 do_prepare_recipe_sysroot: Manifest /home/user/Aspped_BMC/openbmc/build/tmp/sstate-control/manifest-allarch-q71l-ipmi-fru-properties.populate_sysroot not found?
ERROR: phosphor-ipmi-fru-1.0-r1 do_compile: oe_runmake failed
ERROR: phosphor-ipmi-fru-1.0-r1 do_compile: Function failed: do_compile (log file is located at /home/user/Aspped_BMC/openbmc/build/tmp/work/armv5e-openbmc-linux-gnueabi/phosphor-ipmi-fru/1.0-r1/temp/log.do_compile.7519)
ERROR: Logfile of failure stored in: /home/user/Aspped_BMC/openbmc/build/tmp/work/armv5e-openbmc-linux-gnueabi/phosphor-ipmi-fru/1.0-r1/temp/log.do_compile.7519
Log data follows:
| DEBUG: SITE files ['endian-little', 'bit-32', 'arm-common', 'arm-32', 'common-linux', 'common-glibc', 'arm-linux', 'arm-linux-gnueabi', 'common']
| DEBUG: Executing shell function do_compile
| NOTE: make -j 4
| /home/user/Aspped_BMC/openbmc/build/tmp/work/armv5e-openbmc-linux-gnueabi/phosphor-ipmi-fru/1.0-r1/recipe-sysroot-native/usr/bin/python-native/python ../git/scripts/fru_gen.py -i /home/user/Aspped_BMC/openbmc/build/tmp/work/armv5e-openbmc-linux-gnueabi/phosphor-ipmi-fru/1.0-r1/recipe-sysroot-native/usr/share/phosphor-ipmi-fru/config/config.yaml -o . generate-cpp
| /home/user/Aspped_BMC/openbmc/build/tmp/work/armv5e-openbmc-linux-gnueabi/phosphor-ipmi-fru/1.0-r1/recipe-sysroot-native/usr/bin/python-native/python ../git/scripts/extra-properties.py -e /home/user/Aspped_BMC/openbmc/build/tmp/work/armv5e-openbmc-linux-gnueabi/phosphor-ipmi-fru/1.0-r1/recipe-sysroot-native/usr/share/phosphor-ipmi-fru/properties/extra-properties.yaml
| Traceback (most recent call last):
|   File "../git/scripts/extra-properties.py", line 34, in <module>
|     main()
|   File "../git/scripts/extra-properties.py", line 20, in main
|     with open(os.path.join(script_dir, args.extra_props_yaml), 'r') as fd:
| IOError: [Errno 2] No such file or directory: '/home/user/Aspped_BMC/openbmc/build/tmp/work/armv5e-openbmc-linux-gnueabi/phosphor-ipmi-fru/1.0-r1/recipe-sysroot-native/usr/share/phosphor-ipmi-fru/properties/extra-properties.yaml'
| make: *** [extra-properties-gen.cpp] Error 1
| make: *** Waiting for unfinished jobs....
| Traceback (most recent call last):
|   File "../git/scripts/fru_gen.py", line 59, in <module>
|     main()
|   File "../git/scripts/fru_gen.py", line 55, in main
|     function(args.inventory_yaml, args.outputdir)
|   File "../git/scripts/fru_gen.py", line 24, in generate_cpp
|     fd.write(t.render(fruDict=ifile))
|   File "/home/user/Aspped_BMC/openbmc/build/tmp/work/armv5e-openbmc-linux-gnueabi/phosphor-ipmi-fru/1.0-r1/recipe-sysroot-native/usr/lib/python2.7/site-packages/mako/template.py", line 462, in render
|     return runtime._render(self, self.callable_, args, data)
|   File "/home/user/Aspped_BMC/openbmc/build/tmp/work/armv5e-openbmc-linux-gnueabi/phosphor-ipmi-fru/1.0-r1/recipe-sysroot-native/usr/lib/python2.7/site-packages/mako/runtime.py", line 838, in _render
|     **_kwargs_for_callable(callable_, data))
|   File "/home/user/Aspped_BMC/openbmc/build/tmp/work/armv5e-openbmc-linux-gnueabi/phosphor-ipmi-fru/1.0-r1/recipe-sysroot-native/usr/lib/python2.7/site-packages/mako/runtime.py", line 873, in _render_context
|     _exec_template(inherit, lclcontext, args=args, kwargs=kwargs)
|   File "/home/user/Aspped_BMC/openbmc/build/tmp/work/armv5e-openbmc-linux-gnueabi/phosphor-ipmi-fru/1.0-r1/recipe-sysroot-native/usr/lib/python2.7/site-packages/mako/runtime.py", line 899, in _exec_template
|     callable_(context, *args, **kwargs)
|   File "_home_user_Aspped_BMC_openbmc_build_tmp_work_armv5e_openbmc_linux_gnueabi_phosphor_ipmi_fru_1_0_r1_git_scripts_writefru_mako_cpp", line 36, in render_body
| KeyError: 'entityID'
| make: *** [fru-gen.cpp] Error 1
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /home/user/Aspped_BMC/openbmc/build/tmp/work/armv5e-openbmc-linux-gnueabi/phosphor-ipmi-fru/1.0-r1/temp/log.do_compile.7519)
ERROR: Task (/home/user/Aspped_BMC/openbmc/meta-phosphor/common/recipes-phosphor/ipmi/phosphor-ipmi-fru.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 3730 tasks of which 3176 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/user/Aspped_BMC/openbmc/meta-phosphor/common/recipes-phosphor/ipmi/phosphor-ipmi-fru.bb:do_compile
Summary: There was 1 WARNING message shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.

bradbishop, 2018-04-12:
@pstrinkle can you help out here?  @yuennancy fyi...

bradbishop, 2018-04-12:
> I am trying to build IBM-Open BMC 
Just FYI there is no more "IBM" OpenBMC - OpenBMC (this project) is a Linux Foundation project:

https://www.linuxfoundation.org/projects/

reju85, 2018-04-13:
Hi,
Thank you for the info.But still how will resolve the build issue?
Kindly help

amboar, 2018-04-13:
> IOError: [Errno 2] No such file or directory: '/home/user/Aspped_BMC/openbmc/build/tmp/work/armv5e-openbmc-linux-gnueabi/phosphor-ipmi-fru/1.0-r1/recipe-sysroot-native/usr/share/phosphor-ipmi-fru/properties/extra-properties.yaml'

Might be a race condition. @reju85 maybe try building again a few times?

reju85, 2018-04-13:
Hi,
I have been trying this for the past 1 week and tried out some modifications ,but nothing worked out.
Kindly help

pstrinkle, 2018-04-16:
This is a duplicate, I think, but either way, I have two patches queued that should address this: 
https://gerrit.openbmc-project.xyz/#/c/10126/
https://gerrit.openbmc-project.xyz/#/c/10127/

rfrandse, 2018-04-17:
<https://gerrit.openbmc-project.xyz/10126> meta-quanta-q71l: phosphor: sensors: add missing inherit
Fixes:  #3088 Build Error for meta-q71l

closed 2018-04-18 by geissonator

bluecmd, 2018-06-26:
FWIW, I still have this (I think) issue:

```
Initialising tasks: 100% |######################################################################################################| Time: 0:00:04
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: phosphor-ipmi-fru-1.0-r1 do_compile: oe_runmake failed
ERROR: phosphor-ipmi-fru-1.0-r1 do_compile: Function failed: do_compile (log file is located at /home/dhtech/openbmc/build/tmp/work/armv5e-openbmc-linux-gnueabi/phosphor-ipmi-fru/1.0-r1/temp/log.do_compile.11964)
ERROR: Logfile of failure stored in: /home/dhtech/openbmc/build/tmp/work/armv5e-openbmc-linux-gnueabi/phosphor-ipmi-fru/1.0-r1/temp/log.do_compile.11964
Log data follows:
| DEBUG: SITE files ['endian-little', 'bit-32', 'arm-common', 'arm-32', 'common-linux', 'common-glibc', 'arm-linux', 'arm-linux-gnueabi', 'common']
| DEBUG: Executing shell function do_compile
| NOTE: make -j 4
| /home/dhtech/openbmc/build/tmp/work/armv5e-openbmc-linux-gnueabi/phosphor-ipmi-fru/1.0-r1/recipe-sysroot-native/usr/bin/python-native/python ../git/scripts/extra-properties.py -e /home/dhtech/openbmc/build/tmp/work/armv5e-openbmc-linux-gnueabi/phosphor-ipmi-fru/1.0-r1/recipe-sysroot-native/usr/share/phosphor-ipmi-fru/properties/extra-properties.yaml
| Traceback (most recent call last):
|   File "../git/scripts/extra-properties.py", line 34, in <module>
|     main()
|   File "../git/scripts/extra-properties.py", line 20, in main
|     with open(os.path.join(script_dir, args.extra_props_yaml), 'r') as fd:
| IOError: [Errno 2] No such file or directory: '/home/dhtech/openbmc/build/tmp/work/armv5e-openbmc-linux-gnueabi/phosphor-ipmi-fru/1.0-r1/recipe-sysroot-native/usr/share/phosphor-ipmi-fru/properties/extra-properties.yaml'
| Makefile:1068: recipe for target 'extra-properties-gen.cpp' failed
| make: *** [extra-properties-gen.cpp] Error 1
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /home/dhtech/openbmc/build/tmp/work/armv5e-openbmc-linux-gnueabi/phosphor-ipmi-fru/1.0-r1/temp/log.do_compile.11964)
ERROR: Task (/home/dhtech/openbmc/meta-phosphor/common/recipes-phosphor/ipmi/phosphor-ipmi-fru.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 2594 tasks of which 2589 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/dhtech/openbmc/meta-phosphor/common/recipes-phosphor/ipmi/phosphor-ipmi-fru.bb:do_compile
Summary: There was 1 WARNING message shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
dhtech@ocp-bluecmd:~/openbmc$ git rev-parse HEAD
d4250f4a8cd9f4cccf46774e62cdf6cd5f0333b0
```

pstrinkle, 2018-06-26:
I'm checking now -- the extra-properties we have set for the IPMI FRU can likely just be dropped.  I'll test that first.

pstrinkle, 2018-06-26:
https://gerrit.openbmc-project.xyz/#/c/11292/