openbmc/openbmc #3090: OP920: FVT : org.open_power.OCC.Device.Error.OpenFailure: Opening OCC device failed.

cdeadmin, 2018-04-12:


amboar, 2018-04-13:
Please populate this issue with a proper description covering a publicly-relevant BMC version and a description of the steps taken or steps to reproduce the problem.

rfrandse, 2018-04-13:
@gkeishin said.
Test:

1. Boot Host
2. reboot BMC
3. OCC active and Host still up and running

Except the OCC crashed but restarted clean.
Something which we definitely want to handle it cleaner without crashing out.

eddiejames, 2018-04-13:
This is a bug for an OCC control application (on any BMC version) crash during BMC boot. This happens because the host sends the "OCC active" signal over IPMI before FSI is scanned during BMC boot. Since FSI is not scanned, the OCC hwmon devices cannot be bound.

Once FSI is scanned, the OCC control application (which is restarted by systemd) successfully binds the OCC hwmon devices. So, there's no impact to this bug except for some journal entries about the OCC app crashing.

closed 2019-04-27 by stale[bot]